mu-editor/mu #458: Plotter feature creates / does not delete CSV file in data_capture folder.

caternuson, 2018-05-01:
* What you were trying to do,
Plot sensor data.

* What steps you took to make this happen,
On an Adafruit Circuit Playground Express running CircuitPython 2.2.4, run the following code:
```python
import analogio
import board
import time
 
light = analogio.AnalogIn(board.LIGHT)
 
while True:
    print((light.value,))
    time.sleep(0.1)
```
open the REPL and the plotter:
[image]
and let that run for a while. Then close the plotter window by pressing the toolbar button.

* What you expected to happen,
Data gets plotted with no side effects.

* What actually happened,
A CSV file gets created in a folder named **data_capture** on the CircuitPython file system.

* Why this difference is problematic (it may not be a bug!),
If the plotter is left running for a long period of time, the CSV file will take up all the available space on the file system.

* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 
Mu 1.0.0 beta15
Ubuntu 16.04LTS
Adafruit Circuit Playground Express running CircuitPython 2.2.4

caternuson, 2018-05-02:
Ah, thought it might be a feature like that. Thanks. 

A toggle would be a good idea, and I think defaulting to OFF would make most sense. Another idea - would it be possible to add some kind of SAVE button to the plot window? So plot data is only stored in RAM and then the SAVE button would write contents to file.

ladyada, 2018-05-06:
hi @ntoll i just pulled master and running manually with python run.py

i still get a `data_capture` folder & also 'fonts' 'images' 'music' and 'sounds' created on the CPX

ntoll, 2018-05-06:
Ugh... wat? Something's not right. :-/

I've just (quickly) tried to recreate this but I'm failing. What steps do I need to recreate the problem? No matter how hard I try, I can't get a `data_capture` directory nor fonts, images music and sound directories to be incorrectly created on the CPX.

Help!?!

ladyada, 2018-05-07:
hmm the version still says beta15 - but im running it from the command line, so i think its the right version? 

> Starting Mu 1.0.0.beta.15
> 2018-05-06 18:57:02,523 - root:112(run) INFO: uname_result(system='Windows', node='ladyada301-PC', release='7', version='6.1.7601', machine='AMD64', processor='Intel64 Family 6 Model 58 Stepping 9, GenuineIntel')
> 2018-05-06 18:57:02,523 - root:113(run) INFO: Python path: ['C:\\Users\\ladyada\\Dropbox (Personal)\\micropython\\mu', 'C:\\Python36-32\\python36.zip', 'C:\\Python36-32\\DLLs', 'C:\\Python36-32\\lib', 'C:\\Python36-32', 'C:\\Python36-32\\lib\\site-packages', 'C:\\Python36-32\\lib\\site-packages\\win32', 'C:\\Python36-32\\lib\\site-packages\\win32\\lib', 'C:\\Python36-32\\lib\\site-packages\\Pythonwin', 'C:\\Python36-32\\lib\\site-packages\\IPython\\extensions']
> 2018-05-06 18:57:02,635 - mu.logic:546(__init__) INFO: Setting up editor.
> 2018-05-06 18:57:02,635 - mu.logic:560(__init__) INFO: Settings path: C:\Users\ladyada\AppData\Local\python\mu\settings.json
> 2018-05-06 18:57:02,635 - mu.logic:561(__init__) INFO: Session path: C:\Users\ladyada\AppData\Local\python\mu\session.json
> 2018-05-06 18:57:02,635 - mu.logic:562(__init__) INFO: Log directory: C:\Users\ladyada\AppData\Local\python\mu\Logs
> 2018-05-06 18:57:02,635 - mu.logic:563(__init__) INFO: Data directory: C:\Users\ladyada\AppData\Local\python\mu
> 2018-05-06 18:57:02,714 - mu.logic:576(setup) INFO: Available modes: python, adafruit, microbit, debugger
> 2018-05-06 18:57:02,715 - mu.logic:588(setup) DEBUG: Creating directory: E:\images
> 2018-05-06 18:57:04,360 - mu.logic:595(setup) DEBUG: Creating directory: E:\fonts
> 2018-05-06 18:57:04,752 - mu.logic:598(setup) DEBUG: Creating directory: E:\sounds
> 2018-05-06 18:57:07,584 - mu.logic:603(setup) DEBUG: Creating directory: E:\music
> 2018-05-06 18:57:08,175 - mu.logic:1054(change_mode) INFO: Workspace directory: E:\
> 2018-05-06 18:57:08,175 - mu.logic:625(restore_session) INFO: Restoring session from: C:\Users\ladyada\AppData\Local\python\mu\session.json
> 2018-05-06 18:57:08,175 - mu.logic:626(restore_session) DEBUG: {'theme': 'day', 'mode': 'adafruit', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': ''}
> 2018-05-06 18:57:08,175 - mu.logic:647(restore_session) INFO: Loaded files.
> 2018-05-06 18:57:08,175 - mu.logic:651(restore_session) INFO: User defined environment variables: []
> 2018-05-06 18:57:08,175 - mu.logic:655(restore_session) INFO: Minify scripts on micro:bit? False
> 2018-05-06 18:57:08,339 - mu.logic:673(restore_session) INFO: Starting with blank file.
> 2018-05-06 18:57:08,356 - mu.logic:1054(change_mode) INFO: Workspace directory: E:\
> 2018-05-06 18:57:09,833 - mu.logic:988(show_admin) INFO: Showing logs from C:\Users\ladyada\AppData\Local\python\mu\Logs\mu.log
> 2018-05-06 18:58:54,837 - mu.modes.base:199(find_device) INFO: Found device on port: COM20
> 2018-05-06 18:58:54,849 - mu.modes.base:287(add_plotter) INFO: Started plotter
> 2018-05-06 18:58:54,849 - mu.modes.base:277(toggle_plotter) INFO: Toggle plotter on.
> 2018-05-06 18:58:57,697 - mu.modes.base:145(remove_plotter) DEBUG: Creating directory: E:\data_capture
> 2018-05-06 18:58:58,386 - mu.modes.base:155(remove_plotter) INFO: Removing plotter
> 2018-05-06 18:58:58,386 - mu.modes.base:274(toggle_plotter) INFO: Toggle plotter off.
> 2018-05-06 18:59:09,669 - mu.logic:988(show_admin) INFO: Showing logs from C:\Users\ladyada\AppData\Local\python\mu\Logs\mu.log

ntoll, 2018-05-07:
I'm not going to bump master's version to beta.16 until it's out the door (i.e. you should still see beta.15).

The Git hash for the latest commit is 62676b1259de42c56652ff5433dd0fbb168e8607 and at that point (HEAD of master) it works for me.

Just a thought... have you overridden the `workspace_dir` in your settings file..? (From the logs it looks like you need to check `C:\Users\ladyada\AppData\Local\python\mu\settings.json`).

Check out how the method called by the base mode for getting the default workspace directory (https://github.com/mu-editor/mu/blob/master/mu/modes/base.py#L46). I wonder if you edited this for an older version of Mu before Adafruit mode existed or had settled down..?

Please do let me know how you get on.

ntoll, 2018-05-07:
Hmm.... OK... if you remove the "workspace" entry and restart Mu "it should just work". ;-)

dhalbert, 2018-05-09:
Did I have the same issue with master, trying not to have the `images`, `sounds`, and `fonts` directories? I had to remove `~/.local/share/mu/settings.json`.

dhalbert, 2018-05-09:
Maybe there should be some repair code to clean up old settings.json values (if it's possible to figure out that they are wrong).

ntoll, 2018-05-09:
Hi folks, I've opened #466 to track adding a simple feature to help people reset Mu. I'll make sure it's in beta.16 since I don't think it'll take long to do (maybe even this evening).

caternuson, 2018-05-10:
Back to suggestions about the CSV save feature:

- If the plotter window is not open, this feature is OFF.
- In the plotter window, add a UI element to toggle turning this feature ON/OFF with default setting of OFF.
- Toggling from OFF to ON opens a new unique file (serial number? time stamp?) to which plot data is appended until turned OFF.
- Toggling from ON to OFF closes the file. If data was never written, don't actually create a file.
- When ON, show the current file name in the plot window (title bar?).
- For KISS, folder location of file can be hardcoded / buried in json config / not really meant to be changed (unless you edit config file).

caternuson, 2019-10-24:
Doing some housekeeping and came across this. Closing as the original issue seems to be fixed now. There is some other discussion here about plot feature enhancements, but probably best to cover all that in a separate issue.

thanks for all the work on mu!

closed 2019-10-24 by caternuson