mu-editor/mu #459: Files panel: micro:bit right click menu not wide enough for text

carlosperate, 2018-05-02:
Tested on macOS High Sierra, with the latest master at the time of writting.

[image]

ZanderBrown, 2018-05-02:
It should expand?
https://github.com/mu-editor/mu/blob/48821fdf6566c5a36a1e259b2b997df09072ace6/mu/resources/css/day.css#L102-L111

ZanderBrown, 2018-05-02:
[image]
Okay so I can reproduce this on Fedora 28

ZanderBrown, 2018-05-02:
Commenting out
https://github.com/mu-editor/mu/blob/48821fdf6566c5a36a1e259b2b997df09072ace6/mu/resources/css/day.css#L102-L111
Seems to be an initial workaround

PR #461 (2018-05-02): Don't try to size menus
Qt doesn't seem to handle min-width very well for QMenu, doesn't make much difference to us

Fix #459
commit 19bfdce0 (2018-05-02): Don't try to size menus

Qt doesn't seem to handle min-width very well for QMenu, doesn't make much difference to us

Fix #459
closed 2018-05-02 by ntoll