mu-editor/mu #460: Files Panel: Clicking on the widget list resets the panel height

carlosperate, 2018-05-02:
Tested on macOS High Sierra with the latest master at the time of writing:

[image]

ntoll, 2018-05-02:
I too cannot reproduce on Linux. Could this be a Qt5 bug in OSX..?

ZanderBrown, 2018-05-02:
@ntoll that would seem likely, @carlosperate are all panels affected?

carlosperate, 2018-05-02:
Yeah, I could be a PyQt bug, I'm using PyQt5==5.10.1, I assume we are all on the same version?
There are no other panes with similar configuration, so it doesn't happen with the others.

ntoll, 2018-05-03:
Folks, the problem with the dangling file system panels is fixed in ff7bb826a8304a721f317ec9785afc4d65232d26. I'll leave this open until we can be certain this is a Qt bug.

ntoll, 2018-06-04:
Since I've not heard anything to suggest otherwise, I guess this IS a Qt bug.

closed 2018-06-04 by ntoll

carlosperate, 2018-07-06:
Just a note to say that this is still present on PyQt5 5.11.