mu-editor/mu #462: Plotter inaccurate at lower numbers

kattni, 2018-05-02:
I have a potentiometer hooked up to a CPX, min 0, max 3.3. At potentiometer max, the plotter does two odd things:
1. It marks the equal grid lines as 1 and 3, so the first grid space would be representing 1 unit and the equally sized second grid space above it would be representing 2 units.
[image]

2. The max of 3.3 appears on the plotter to be less than 3, but difficult to determine exactly as the gridlines are odd per number 1.
[image]

If I turn the potentiometer until it equals 1.9, it plots along the 1-labeled gridline. 
[image]

I tested it with a simple `print((1.9,))` to eliminate some unlikely but possible issue with the potentiometer and it plots just below the 1 line as well.
[image]

It still prints slightly below the line for 7 and 15.
[image]

31-32 seem to start getting closer to plotting accurately, 61-62 even closer, and 125 seems entirely accurate. It's easiest to tell with these specific numbers because that's where the grid lines on the plotter show up when you start to get into these ranges. It seems the issue evens itself out as you get into higher numbers, but it could be the fact that the plotter is scaling into higher ranges in the same visual space.

jepler, 2018-05-03:
It's like the numbers on the legend are all 1 value smaller than they ought to be.  If the Y axis values in the first two screenshots were 4, 2, 0, -2, -4 the graph would seem to match the values.

ntoll, 2018-05-03:
Ouch... that's not good. Thanks for reporting the problem. I'll take a look... I suspect an OBOE (I don't mean the reed instrument either). ;-)

ntoll, 2018-05-03:
Hi @kattni, what version of Mu are you using? Is it, perhaps, `beta.15`..? 

I've just started to look at the problem and noticed I'd already refactored the way the plotter displays the Y axis to fix an earlier problem. The screenie below shows how it looks now... I hope you agree, it's a more accurate reflection of what should be displayed. This'll be in the upcoming `beta.16` and is currently in `master`.

[image]

Please feel free to re-open if `beta.16` doesn't work as you expect (when released).

Once again, thanks to you and @jepler for pointing this out. :-)

closed 2018-05-03 by ntoll