mu-editor/mu #470: Using MU with an Adafruit Pyboard

KarlD1239, 2018-05-19:
I'm using MU 1.0.0.beta15 on a Microsoft Surface Pro 4 and Windows 10
.
It works OK with a Featherboard MO Express.  Upload, edit, download and REPL; all there.

When I use it with a Pyboard it allows me to upload a program from the Pyboard, edit it, and download it OK.  I can reset the Pyboard and the program works OK.  But no REPL  The response when clicking on the REPL is that there isn't a board connected.

ntoll, 2018-05-19:
The PyBoard is not a board that's supported by Mu. While Mu works with MicroPython/CircuitPython on the micro:bit and Adafruit boards, the PyBoard is a "best effort" affair.

closed 2018-05-19 by ntoll

KarlD1239, 2018-05-19:
It looks like all it takes is getting the REPL function operational. What would that involve?
Best,,

ntoll, 2018-05-19:
It wouldn't take a lot to make the REPL work, but I believe Damien (who makes the PyBoard) wants to create his own editor. cc/@dpgeorge

KarlD1239, 2018-05-19:
In the meantime the Pyboard is more difficult to use than necessary.  Other options include Rasperry's and Featherboards with good IDE support.

Is the MU editor open source and if so, how would one go about making the REPL work for a Pyboard?

ZanderBrown, 2018-05-20:
@KarlD1239  I'm a little busy & don't have the board. Sorry.

ntoll, 2018-05-20:
Before doing this, I'd want to know we were not stepping on the toes of people from MicroPython (i.e. @dpgeorge). As I said, I believe he has plans for some sort of MicroPython editor and so I'd rather not re-invent the wheel. 

Having said that, adding a MicroPython/PyBoard mode to Mu would be relatively trivial to do.

Nevertheless, if all you need is an editor for the files on the SD card, I'd suggest you use the editor you're most familiar with. If you want to connect to the REPL on the PyBoard, then I suggest the simple process described here is worth a look: https://docs.micropython.org/en/latest/pyboard/pyboard/tutorial/repl.html

I hope this makes sense / helps.

KarlD1239, 2018-05-20:
Thanks for the reply and suggestion of using PuTTY.  I have and while it works OK, the ability to use MU with it’s convenience makes one yearn for more.

If MU is open source and I understand the Pyboard is, I wonder if a students at a local University might be able to get the job done without upsetting anybody?

Just a thought,
Best regards

From: Nicholas Tollervey
Sent: Sunday, May 20, 2018 10:42 AM
To: mu-editor/mu
Cc: KarlD1239; Mention
Subject: Re: [mu-editor/mu] Using MU with an Adafruit Pyboard (#470)

Before doing this, I'd want to know we were not stepping on the toes of people from MicroPython (i.e. @dpgeorge). As I said, I believe he has plans for some sort of MicroPython editor and so I'd rather not re-invent the wheel.
Having said that, adding a MicroPython/PyBoard mode to Mu would be relatively trivial to do.
Nevertheless, if all you need is an editor for the files on the SD card, I'd suggest you use the editor you're most familiar with. If you want to connect to the REPL on the PyBoard, then I suggest the simple process described here is worth a look: https://docs.micropython.org/en/latest/pyboard/pyboard/tutorial/repl.html
I hope this makes sense / helps.

dpgeorge, 2018-05-22:
@ntoll I have no problem supporting pyboard with Mu.  The more editor options the better, so users can pick and choose what suits them best.  Can I help with this?

ntoll, 2018-05-22:
Ahhh... OK... then it should be relatively easy to add a PyBoard mode to Mu. I'm currently in Ireland training developers for the next week, but should be able to put something together upon my return. :-)

KarlD1239, 2018-05-22:
Thanks for keeping me in the loop on this!  It’s good to hear that there’s interest in getting it done.

I think this is a great idea!

Thanks again.
Denny

Dennis Siemer
Mankato, MN USA

From: Nicholas Tollervey
Sent: Tuesday, May 22, 2018 2:38 AM
To: mu-editor/mu
Cc: KarlD1239; Mention
Subject: Re: [mu-editor/mu] Using MU with an Adafruit Pyboard (#470)

Ahhh... OK... then it should be relatively easy to add a PyBoard mode to Mu. I'm currently in Ireland training developers for the next week, but should be able to put something together upon my return. :-)

KarlD1239, 2018-07-13:
Any progress on this idea?
Denny

From: Nicholas Tollervey
Sent: Tuesday, May 22, 2018 2:38 AM
To: mu-editor/mu
Cc: KarlD1239; Mention
Subject: Re: [mu-editor/mu] Using MU with an Adafruit Pyboard (#470)

Ahhh... OK... then it should be relatively easy to add a PyBoard mode to Mu. I'm currently in Ireland training developers for the next week, but should be able to put something together upon my return. :-)

dpgeorge, 2018-07-16:
@ntoll to move forward with this I guess we first need to decide what the pyboard mode would be: use the existing microbit or Adafruti mode, or add a new one?  Path of least resistance might be to use an existing one, but conceptually a pyboard is neither an Adafruit nor a microbit device, so probably needs its own mode.

ntoll, 2018-07-16:
There's a base `MicroPythonMode` class that implements most of the features that would probably be needed. The child mode simply needs to call the buttons the right sort of things and specify where to save files (and other mode specific settings). It's not hard to do.

dpgeorge, 2018-07-16:
Thanks, I'll take a look at adding a new mode then.

ntoll, 2018-07-16:
TBH... it's probably quicker if I do it... ;-) Give me a kick when you know what you need.