mu-editor/mu #473: Mu doesn't detect an ESP8266 board with Micropython on Windows 10

cefn, 2018-05-27:
If you are reporting a bug, we would like to know:

* What you were trying to do,

Clicked on REPL button to see the REPL interface. 

* What steps you took to make this happen,

* Installed Mu with `pip install mu-editor`
* Launched Mu from cmd.exe with `mu`
* Flashed [Vanguard board](https://vgkits.org/blog/) with Micropython 1.9.3, ensuring there was no main.py to block REPL connection. 
  - Vanguard board is practically equivalent to Feather Huzzah but with a CH340 USB to UART.
* Plugged in Vanguard board over USB.
  - running in virtualbox, so bound the usb signature to the Windows10 VM
* Proved connectivity over serial 
  - successfully connected over miniterm like `python -m serial.tools.miniterm --raw --eol CRLF --encoding ascii COM3 115200`

* What you expected to happen,

Not sure what the REPL interface is, hence I was running it

* What actually happened,

Mu reported 'could not find an attached device'

* Why this difference is problematic (it may not be a bug!),

Other tests in python3 installed on the machine (connecting over miniterm) demonstrate that the device is responsive over serial REPL on COM3 and is running Micropython, but Mu doesn't identify it.

* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Windows 10, Mu 1.0.0.beta.15

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

See below for the log...

```
2018-05-27 15:00:50,587 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.0.beta.15
2018-05-27 15:00:50,587 - root:113(run) INFO: uname_result(system='Windows', node='MSEDGEWIN10', release='10', version='10.0.17134', machine='AMD64', processor='Intel64 Family 6 Model 58 Stepping 9, GenuineIntel')
2018-05-27 15:00:50,587 - root:114(run) INFO: Python path: ['C:\\Users\\cefn\\AppData\\Local\\Programs\\Python\\Python36\\Scripts\\mu.exe', 'c:\\users\\cefn\\appdata\\local\\programs\\python\\python36\\python36.zip', 'c:\\users\\cefn\\appdata\\local\\programs\\python\\python36\\DLLs', 'c:\\users\\cefn\\appdata\\local\\programs\\python\\python36\\lib', 'c:\\users\\cefn\\appdata\\local\\programs\\python\\python36', 'c:\\users\\cefn\\appdata\\local\\programs\\python\\python36\\lib\\site-packages', 'c:\\users\\cefn\\appdata\\local\\programs\\python\\python36\\lib\\site-packages\\IPython\\extensions']
2018-05-27 15:00:50,680 - mu.logic:355(__init__) INFO: Setting up editor.
2018-05-27 15:00:50,680 - mu.logic:169(get_settings_path) DEBUG: Creating settings file: C:\Users\cefn\AppData\Local\python\mu\settings.json
2018-05-27 15:00:50,680 - mu.logic:366(__init__) INFO: Settings path: C:\Users\cefn\AppData\Local\python\mu\settings.json
2018-05-27 15:00:50,680 - mu.logic:367(__init__) INFO: Log directory: C:\Users\cefn\AppData\Local\python\mu\Logs
2018-05-27 15:00:50,680 - mu.logic:368(__init__) INFO: Data directory: C:\Users\cefn\AppData\Local\python\mu
2018-05-27 15:00:50,743 - mu.logic:376(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2018-05-27 15:00:50,743 - mu.logic:380(setup) DEBUG: Creating directory: C:\Users\cefn\mu_code
2018-05-27 15:00:50,743 - mu.logic:386(setup) DEBUG: Creating directory: C:\Users\cefn\mu_code\images
2018-05-27 15:00:50,743 - mu.logic:393(setup) DEBUG: Creating directory: C:\Users\cefn\mu_code\sounds
2018-05-27 15:00:52,510 - mu.logic:712(change_mode) INFO: Workspace directory: C:\Users\cefn\mu_code
2018-05-27 15:00:52,510 - mu.logic:415(restore_session) INFO: Restoring session from: C:\Users\cefn\AppData\Local\python\mu\settings.json
2018-05-27 15:00:52,510 - mu.logic:416(restore_session) DEBUG: {}
2018-05-27 15:00:52,510 - mu.logic:679(select_mode) INFO: Showing available modes: ['python', 'adafruit', 'microbit', 'debugger', 'pygamezero']
2018-05-27 15:01:07,000 - mu.interface.main:650(show_message) DEBUG: Could not find an attached Adafruit CircuitPython device.
2018-05-27 15:01:07,000 - mu.interface.main:651(show_message) DEBUG: Python files for Adafruit CircuitPython devices are stored on the device. Therefore, to edit these files you need to have the device plugged in. Until you plug in a device, Mu will use the directory found here:

 C:\Users\cefn\mu_code

...to store your code.
[... 132 lines cut ...]
2018-05-27 15:50:47,881 - mu.logic:712(change_mode) INFO: Workspace directory: C:\Users\cefn\mu_code
2018-05-27 15:50:47,881 - mu.logic:415(restore_session) INFO: Restoring session from: C:\Users\cefn\AppData\Local\python\mu\settings.json
2018-05-27 15:50:47,881 - mu.logic:416(restore_session) DEBUG: {'theme': 'day', 'mode': 'adafruit', 'paths': [], 'workspace': 'C:\\Users\\cefn\\mu_code', 'microbit_runtime_hex': None}
2018-05-27 15:50:47,881 - mu.logic:435(restore_session) INFO: Loaded files.
2018-05-27 15:50:48,553 - mu.logic:444(restore_session) INFO: Starting with blank file.
2018-05-27 15:50:48,615 - mu.interface.main:650(show_message) DEBUG: Could not find an attached Adafruit CircuitPython device.
2018-05-27 15:50:48,615 - mu.interface.main:651(show_message) DEBUG: Python files for Adafruit CircuitPython devices are stored on the device. Therefore, to edit these files you need to have the device plugged in. Until you plug in a device, Mu will use the directory found here:

 C:\Users\cefn\mu_code

...to store your code.
2018-05-27 15:50:50,772 - mu.logic:712(change_mode) INFO: Workspace directory: C:\Users\cefn\mu_code
2018-05-27 15:50:52,648 - mu.modes.base:130(find_device) WARNING: Could not find device.
2018-05-27 15:50:52,648 - mu.modes.base:131(find_device) DEBUG: Available ports:
2018-05-27 15:50:52,648 - mu.modes.base:135(find_device) DEBUG: ['PID:29987 VID:6790 PORT:COM3']
2018-05-27 15:50:52,664 - mu.interface.main:650(show_message) DEBUG: Could not find an attached device.
2018-05-27 15:50:52,664 - mu.interface.main:651(show_message) DEBUG: Please make sure the device is plugged into this computer.

It must have a version of MicroPython (or CircuitPython) flashed onto it before the REPL will work.

Finally, press the device's reset button and wait a few seconds before trying again.
2018-05-27 15:50:57,822 - mu.modes.base:158(toggle_repl) INFO: Toggle REPL on.
2018-05-27 15:53:00,716 - mu.logic:668(show_logs) INFO: Showing logs from C:\Users\cefn\AppData\Local\python\mu\Logs\mu.log

```

ntoll, 2018-06-04:
I'm closing this since it's a feature request and folks like @eduvik have already been pinged about the new modal architecture which would allow an ESP8266 mode.

closed 2018-06-04 by ntoll

tcpipchip, 2019-06-24:
I have the same problem!

erniebose, 2019-09-18:
I have the same problem on a Pyboard. fortunately I purchased an Adafruit Pyruler and its works just fine but all this makes me think that Mu may be a wase of time as both products bought from Adafruit behave differently. Before I get a lot of shade about Micropython and Circuitpyhthon I do know the difference. I have now wasted several days trying to solve this.

ntoll, 2019-09-19:
@erniebose try the latest alpha found on the downloads page: https://codewith.mu/en/download

erniebose, 2019-09-19:
I tried and it did not work. I am on an apple not on Windows. To add a bit more info, the CircuitPython device works great. I can read and write files and REPL works. 
 On the MicroPython pyBoard I can access files and I can connect using screen.  I then get “print” messages. ctrl c and d work with a little frustration. REPL does not work and does not recognize the detect the serial port on the board.