mu-editor/mu #475: Unicode error Python-mode input("What? ÅÄÖ") MU Crash

smeand, 2018-06-03:
It is working if I use Python 3.6 but not with Python-mode in MU. Crashes every time. Latest build.
Probably swedish teachers not happy if MU-Python-Mode not supporting Unicode-8/Latin

2018-06-03 16:47:59,568 - mu.logic:1082(autosave) INFO: Autosave has detected changes.
2018-06-03 16:47:59,569 - mu.logic:824(save_tab_to_file) INFO: Saving script to: C:\Users\Anders\mu_code\testpython.py
2018-06-03 16:47:59,569 - mu.logic:825(save_tab_to_file) DEBUG: input("What? ÅÄÖ")

2018-06-03 16:48:04,566 - mu.logic:1082(autosave) INFO: Autosave has detected changes.
2018-06-03 16:48:04,566 - mu.logic:824(save_tab_to_file) INFO: Saving script to: C:\Users\Anders\mu_code\testpython.py
2018-06-03 16:48:04,566 - mu.logic:825(save_tab_to_file) DEBUG: input("What? ÅÄÖ")
2018-06-03 16:48:06,032 - mu.logic:824(save_tab_to_file) INFO: Saving script to: C:\Users\Anders\mu_code\testpython.py
2018-06-03 16:48:06,032 - mu.logic:825(save_tab_to_file) DEBUG: input("What? ÅÄÖ")
2018-06-03 16:48:07,925 - mu.logic:824(save_tab_to_file) INFO: Saving script to: C:\Users\Anders\mu_code\testpython.py
2018-06-03 16:48:07,925 - mu.logic:825(save_tab_to_file) DEBUG: input("What? ÅÄÖ")
2018-06-03 16:48:26,352 - mu.logic:907(check_code) INFO: Checking code.
2018-06-03 16:48:37,955 - mu.modes.python3:184(run_script) DEBUG: input("What? ÅÄÖ")
2018-06-03 16:48:37,958 - mu.interface.panes:617(start_process) INFO: Running script: c:\users\anders\mu_code\testpython.py
2018-06-03 16:48:37,958 - mu.interface.panes:619(start_process) INFO: Running with interactive mode.
2018-06-03 16:48:37,958 - mu.interface.panes:622(start_process) INFO: Command args: []
2018-06-03 16:48:37,959 - mu.interface.panes:633(start_process) INFO: Working directory: C:\Users\Anders\mu_code
2018-06-03 16:48:37,959 - mu.interface.panes:638(start_process) INFO: Python path: ['C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\Python\\python36.zip', 'C:\\Program Files\\Mu\\Python', 'C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']
2018-06-03 16:48:38,026 - root:94(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "C:\Program Files\Mu\pkgs\mu\interface\panes.py", line 846, in read_from_stdout
    self.append(data)
  File "C:\Program Files\Mu\pkgs\mu\interface\panes.py", line 864, in append
    cursor.insertText(msg.decode('utf-8'))
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc5 in position 6: invalid continuation byte
2018-06-03 16:48:44,489 - root:111(run) INFO:

tjguk, 2018-06-03:
Thanks for the report, Anders. We've worked quite hard to get international / unicode support in place, but clearly we've missed something here. I'll try to reproduce the error to pinpoint the problem.

tjguk, 2018-06-03:
[might come through twice]

This is frustrating: I can't reproduce this on my current dev build. I 
did briefly get a slightly different error (to do with a missing 
source-encoding cookie) but I can't now reproduce that.

@smeand would you mind uploading the entire log file as an attachment 
file, rather than embedded in the github message, please?

@ntoll do you know what commit the most recent Windows build is based off?

ntoll, 2018-06-03:
The latest build will be the latest commit to master. 

@smeand do you really mean latest (development) build or do you mean the 15.beta build released a few months ago..? +1 to @tjguk's request for the WHOLE log. Thanks

ZanderBrown, 2018-06-03:
@smeand your .txt is everything we wanted, for reference the source .log is at `C:\Users\Anders\AppData\Local\python\mu\Logs\mu.log` as demonstrated in the last line

smeand, 2018-06-04:
[mu.txt](https://github.com/mu-editor/mu/files/2067094/mu.txt)
Tested on win 7 at work. Same result. Crashes. Same version MU

ntoll, 2018-06-04:
My guess is this is likely to be fixed when #471 lands in master (coming soon!).

ntoll, 2018-06-04:
@smeand thanks for sending over the log file. All help and information is most appreciated.

ntoll, 2018-06-04:
@smeand OK... I just merged @tjguk's potential fix. When you have time, can you try the development build based off ea61159fac2aa14631af493eb586d4b1e812a9db and let me know how you get on?

tjguk, 2018-06-04:
Thanks, @ntoll; I'd forgotten that I'd left that in a halfway state. I'm not 100% sure, though, because I couldn't reproduce the problem running off master. However, if @smeand can try the newer build when it arrives, then we'll know where we're at.

smeand, 2018-06-04:
[image]
Ooops... I try again.

ntoll, 2018-06-04:
Hmmm... that very installer just worked perfectly for me. Not quite sure what to suggest since it works for me but not for you... :-/

@tjguk / @carlosperate any ideas..?

tjguk, 2018-06-04:
On the surface it means that there's code still running which is holding a lock on sip.pyd -- if it's not Mu itself, it could be a virus checker or indexer (typically with a delete-share lock which means you can delete while it's locked, but not recreate it until it's unlocked). If it's one of those, you can just try again a few moments later and it'll probably be clear

ntoll, 2018-06-04:
Aha... the old "have you tried switching it off and on again?" trick... ;-) Thanks @tjguk

smeand, 2018-06-04:
Ok. Tried at home "instead. Much better. ÅÄÖ working!! In REPL I could input("ÅÄÖ") and answer with åäö, but in "RUN-mode" I couldn´t answer with åäö.

ntoll, 2018-06-04:
@smeand not quite sure what you mean by "RUN-mode". Here's what I think you mean:

* It works when you use the version of Mu you install with the automatically built package.
* It doesn't work when you try to run it from source -- i.e. you type `run.py` in the root of the repos (in which case, `git pull origin master` should update your source code to the latest version).

Please confirm..? Thanks..! :-)

tjguk, 2018-06-05:
I imagine he means "When I press the [Run] button" as opposed to "When I'm in the ad hoc REPL"

smeand, 2018-06-05:
Yes. I made a short video. I hope it explain the "problem". Maybe github don´t like video link?
https://youtu.be/ddqmJy4nU3A

ntoll, 2018-06-05:
Actually, now that I've seen your video, I've been able to recreate it and know exactly what the problem is. I'll fix it now.

ntoll, 2018-06-05:
OK... fixed in bf5502acb303ce67c248071761339ab5224cd5bd (includes tests). I'll close this now, but please feel free to re-open if it's not completely working right.

commit bf5502ac (2018-06-05): Fix non-ascii character problem with the PythonProcessPane. Fixes #475.
closed 2018-06-05 by ntoll