mu-editor/mu #476: Have Mu prefer code file names in the same order as CircuitPython

kattni, 2018-06-03:
Apologies if this is something you've already included or discussed and chosen not to include - I didn't see anything about it in any issues.

I received this error when starting Mu:
```
Logging to /Users/kattni/Library/Logs/mu/mu.log
objc[47355]: Class FIFinderSyncExtensionHost is implemented in both /System/Library/PrivateFrameworks/FinderKit.framework/Versions/A/FinderKit (0x7fffaf0fbc90) and /System/Library/PrivateFrameworks/FileProvider.framework/OverrideBundles/FinderSyncCollaborationFileProviderOverride.bundle/Contents/MacOS/FinderSyncCollaborationFileProviderOverride (0x112d58cd8). One of the two will be used. Which one is undefined.
```
I had both `main.py` and `code.py` on a CPX at the time. When I removed `main.py` the error went away and Mu automatically opened `code.py`. I assume, therefore, that it is related.

CircuitPython has an order in which it looks for file names: code.txt, code.py, main.txt, main.py. I don't know if Mu even reads text files, so that bit may be irrelevant. 

Consider having Mu prefer code.py over main.py. Don't throw and error when both are present, and simply open the one that CircuitPython is going to be running. 

At the moment, it appears to open an "untitled" and forces you to load the one you'd like to work with. I feel like making the assumption that the user is going to be working with the same file that CircuitPython is running is reasonable.

ZanderBrown, 2018-06-04:
As `py` is just `txt` by another name there is no reason Mu shouldn't be able to handle it but I believe the current logic rejects non `py` extensions (except `hex` which is opened in m:b mode)

ntoll, 2018-06-04:
Hi @kattni :wave: 

Hmm... this doesn't look like a Mu thing, but is more to do with OSX. Why do I think so? Because Mu works fine on Linux with a CPX containing both `main.py` and `code.py` files (I just checked). Also, the error message you included is not from Mu itself, rather, it appears to be something called `FinderKit` in the bowels of OSX. Not quite sure what to suggest, as a result. :-/

I've also just tried recreating the problem on Mary's MacBook. She's on the latest build of the latest version of OSX and it works fine for me when starting Mu and I don't see the error message you report.

Could this be something specific to your machine..?

If you can give me more information on your setup I may be able to recreate the problem.

kattni, 2018-06-04:
2017 MacBook Pro running 10.13.4 (17E199). 

That it is OSX is entirely believable. I didn't think to test it on a box running a different OS.  It's definitely possible that it's something with my machine. I have a lot of backend customisations, however, really, none of them should affect Mu and its functionality.

If the issue isn't reproducible, then brilliant! I can deal with it on my end, but I figured if it was something happening globally, it might be worth filing an issue.

ntoll, 2018-06-04:
@kattni I'm pretty certain this is isolated to your machine especially since you say you have "a lot of backend customisations". When you say, "none of them should affect Mu and its functionality" I quite understand where you're coming from... but if there's one thing working with computers has taught me, it's that "should" is an optimistic outlook. :-P

(Also, "could" is often a synonym for "will" -- especially if the situation being described is of a potentially catastrophic nature). :grimacing: 

I'll close for now, but you should feel free to re-open if it can be reproduced. (D'ya see what I did there...?)

closed 2018-06-04 by ntoll