mu-editor/mu #480: mu_code path issues

bennuttall, 2018-06-09:
Currently, when you click load, and want to load a few files from a path other than `mu_code`, you have to browse to the file location each time. It would be nice if Mu remembered the last path you browsed to, rather than defaulting to `mu_code`. Alternatively, if you could select multiple files to open together, that would be handy.

Similarly, it seems the path `~/mu_code/` is hard-coded as the pwd, so if your file is located in `~/Projects/project_bob/src/`, the following code:

```python
with open('file.txt') as f:
    print(f.readline())
```

will attempt to open `~/mu_code/file.txt`. But if you run the script from the terminal from within the `src` directory it would use `~/Projects/project_bob/src/file.txt`. I think the pwd should be the location of the file run, not always `mu_code`.

ZanderBrown, 2018-06-09:
I believe @ntoll decided not to implement this in favour of expecting people to use the workspace_dir setting (#87)

carlosperate, 2018-07-06:
While mu_code is by design always as the default location for loading/saving scripts (for the reasons stated in https://github.com/mu-editor/mu/issues/126#issuecomment-236389947), I'm not so sure if that's also applicable to the cwd.

In the example given by Ben, when executing just `open('file.txt')` most people and online tutorials/guides would usually assume that file ends up in the same directory than the script that is running.
Yes, if you know everything starts by default in the `mu_code` directory, then you'd always look there, but this is mostly a "hidden" feature, only visible when opening and saving files.

I think I'm with Ben on this one, and that when a file is run the cwd should be script directory.
@ntoll any thoughts?

ntoll, 2018-12-22:
So, ironically, I was told that all user code and file "activity" should be stored in a single location and Mu should never deviate from this location. @bennuttall, it's a "feature" requested by your colleagues.

However, I believe your suggestion is the correct behaviour and will take it as an excuse to change things as per your suggestions. ;-)

ntoll, 2018-12-22:
To be clear, Mu will continue to work with `mu_code` as the default location. But if the user's script is in a different location then the `cwd` should reflect this (and be remembered in the file-selection dialog).

bennuttall, 2018-12-23:
Thanks @ntoll I think it makes sense to have a default directory so people who don't (yet) care where their script is saved don't have to worry, but the issue was with change of behaviour (from a normal python prompt) when you save elsewhere and refer to relative files.

> But if the user's script is in a different location then the cwd should reflect this (and be remembered in the file-selection dialog).

Absolutely :+1:

PR #739 (2019-01-14): File dialog will remember path last used.
This PR fixes #480. @bennuttall please test this behaviour works as expected for you. I'll merge once you reply. This is the final blocker for 1.0.2.

Thanks.
commit a3adbe8a (2019-01-14): File dialog will remember path last used. Fixes #480.
closed 2019-01-14 by ntoll