mu-editor/mu #486: Output pane stays open after changing modes PyGame Zero -> Python 3

bennuttall, 2018-06-17:
1. Open Mu, create new file, save it (empty)
1. Choose PyGame Zero mode
1. Press Run button
1. PyGame window opens up, close it
1. Don't hit Stop button
1. Change to Python 3 mode
1. Run same file
1. Press Stop button

It's now impossible to get rid of the output pane.

The issue could actually be that it's possible to change modes before stopping a PyGame Zero script. I see that you can't change modes from Python 3 before stopping the script (the Modes button is greyed out). Perhaps that's what's missing from the PyGame Zero mode (and maybe others).

ZanderBrown, 2018-06-17:
This has cropped up a few times now with the various modes

ZanderBrown, 2018-06-17:
Reference: https://github.com/mu-editor/mu/issues/460#issuecomment-385931623 & https://github.com/mu-editor/mu/commit/ff7bb826a8304a721f317ec9785afc4d65232d26

ntoll, 2018-06-19:
Thanks for the heads up! Fixed and will be in tomorrow's beta.16 release. ;-)

commit 80dbc598 (2018-06-19): Fixes #486. Ensure it's not possible to transition to a new mode when the game is running.
closed 2018-06-19 by ntoll