mu-editor/mu #488: Mu requires an old version of pycodestyle

bennuttall, 2018-06-18:
I just ran `pip install pycodestyle --upgrade` and then Mu failed to run, because it depends on `pycodestyle==2.3.1` not `'pycodestyle>=2.3.1`.

As a compromise, if there's a concern future pycodestyle releases may break Mu, we could insist on `<3` as well: `pycodestyle>=2.3.1,<3`.

Perhaps this issue relates to other dependencies too.

ntoll, 2018-06-18:
Aha... good catch. Will try to look into this pre-1.0. Beta 16 should land tomorrow. :-)

tmontes, 2019-06-01:
While the particular details raised by @bennuttall nearly a year ago are no longer true (even though PR #489 was never merged, commit e57485632e75f6df4af58fdce6e854cf40ca656d updated `pycodestyle` to `==2.4.0`), the world out keeps moving and, as of today, many `setup.py` pinned dependencies have evolved.

I raised this generic topic as a follow up action in PR #779's post-merge notes [here](https://github.com/mu-editor/mu/pull/779#issuecomment-496009608).

With that in mind, I'm wondering if we could maybe close this particular issue, while creating a more widely scoped one with regards to pinning-/not-pinning runtime and dev-time dependences.

What do you say?

carlosperate, 2019-06-01:
> With that in mind, I'm wondering if we could maybe close this particular issue, while creating a more widely scoped one with regards to pinning-/not-pinning runtime and dev-time dependences.

We should have one of those.
One thing to take in consideration with pinned dependencies is that the deb installer from Debian buster and Raspbian will install Mu and it's dependencies in the global python, and we have a lot of pinned dependencies that don't need to be. So, If the user or a different package decides to install a slightly different version that will break the Mu launcher.

tmontes, 2019-07-13:
I propose this issue is closed for two motives:
* The original issue is no longer true, AFAICT.
* A new, broader scoped issue, following up on this one and on #779 has just been logged. See #893.

carlosperate, 2019-07-18:
Closing for the reasons listed by @tmontes in https://github.com/mu-editor/mu/issues/488#issuecomment-511150169.

closed 2019-07-18 by carlosperate