mu-editor/mu #490: Debug Inspector pane: Changing "selected" state resets its size

carlosperate, 2018-06-19:
This might be an bug with the macOS version of PyQt5, similar to https://github.com/mu-editor/mu/issues/460, could somebody in other OS try to replicate?

The issue is that when the debugger pane changes from "selected" to "not selected" it also resets its size to the original size. The GIF below show show it better:

[image]

ntoll, 2018-06-19:
I can confirm this happens on Linux. Taking a quick look at the code (see, https://github.com/mu-editor/mu/blob/master/mu/interface/editor.py#L279 -- which is where the interface is updated when a button-click in the debugger causes a change of state) this could be a bug in Qt. :-/

carlosperate, 2018-06-19:
But that bit of code is about the editor itself, where is the debug inspect pane size set in the first place? I haven't been able to find it on a quick pass at the code.

ntoll, 2018-06-19:
That's because I think (IIRC) we just let Qt do the "default" settings.

carlosperate, 2018-06-19:
Right, it's only a small annoyance at the moment. Should we leave this open until the next release of qt, or until we find a bug report in their tracker?
This is low priority, so could be left in the backburner.

carlosperate, 2018-07-05:
Just to note that the same thing happens with the micro:bit (and I assume other modes) REPL pane when the editor content is modified:

[image]

In this case I am still using PyQt5 5.10, but I think 5.11 is out in PyPI already, so we could re-test all this in that version whenever we find a chance.

carlosperate, 2018-07-06:
I can confirm this still happens in PyQt5 5.11.

ZanderBrown, 2018-07-06:
This only seems to happen when you interact with QScientilla everything else is fine so i would suggest any upstream bug would be with that not QT

ZanderBrown, 2018-07-06:
Not running against 5.11 but I'm seeing this on Fedora

ntoll, 2018-07-07:
I can reproduce this with Qt5.11 on Linux. Since this appears to be an upstream bug, all I can suggest is:

* Is this already reported?
* If not, we should report it.
* If it is we should add comments to make it more visible.

bddicken, 2019-04-10:
Figured I'd chime in:

I teach intro CS and recommend Mu to my students. I recently did a section on debugging, and this issue is rather annoying when presenting the debugger to students. Perhaps it is a Qt issue, but it would be great to fix this sooner than later.

carlosperate, 2020-06-14:
I can confirm I could not replicate this in macOS Mojave with PyQt 5.12.1 nor with PyQt 5.13.2.

If somebody could confirm this is not an issue in any of the other platforms I think we can close this one.

ZanderBrown, 2020-06-15:
I don't see it (5.15)

carlosperate, 2020-06-15:
Thanks @ZanderBrown, I assume that was on Linux, which was the other platform identified here with the original issue, so we can this one 👍 
Thanks everyone!

closed 2020-06-15 by carlosperate