mu-editor/mu #491: Moving breakpoints that fall into a comment cannot be removed

carlosperate, 2018-06-19:
First of all, great progress with the debugger updates, it's fixed a lot of the issues, thanks Nicholas!

One thing I've found: when breakpoints are added and then the code is edited the breakpoints can move around. If a breakpoint then fall onto a comment then it cannot removed.

GIF for clarity:

[image]

carlosperate, 2018-06-19:
Also, I cannot find easy steps to replicate this other issue, but if has happened a couple of times already:
When a line with a breakpoint is removed and its breakpoint falls on top of a line that already had a breakpoint, then sometimes (and only sometimes) when the debugger is opened an extra breakpoint will magically appear in the line below (where it used to be before the line was deleted).

ntoll, 2018-06-19:
OK... I think I know how to fix both of these issues. I'll try to get a fix in momentarily.

closed 2018-06-19 by ntoll

carlosperate, 2018-06-19:
Found it, I've opened a new issue: #493

ntoll, 2018-06-19:
@carlosperate when did the breakpoint appear? After you clicked the "Debug" button? If that's the case, I'm not sure what's going on. Each line with a breakpoint marker next to it checked to see if it's a valid breakpoint line (i.e blank lines like line 4 are NOT valid). If it isn't the breakpoint is removed once the debugger has bootstrapped into existence.

ntoll, 2018-06-19:
Aha #493..! That should be easy to fix.

commit 01cac6bc (2018-06-19): Fixes #345. Fixes #491. Update debugger behaviour when handling breakpoints and breakpoint markers.