mu-editor/mu #494: Debugger: If we step through code, click run and there is an infinite loop the line highlight sticks

carlosperate, 2018-06-19:
If there is an infinite loop in the code (in this case let's say at the end of the script) and we step through previous lines of code, and then click run, the code runs as expected but the editor does not remove the line highlight.

GIF for clarity:
[image]

closed 2018-06-20 by ntoll

ntoll, 2018-06-20:
OK... I think this is what's happening:

* The debug runner child process is basically stuck in the infinite loop so never gets the opportunity to respond to the debug client to say that the debug state has changed (at which point Mu used to update the currently highlighted line).
* My fix ensures the currently highlighted line is switched off *before* sending any signal to the debug runner child process to change the state of the script under control of the debugger. This ensures the UI is updated before any potential blocking in the child process.
* At some future time we should attempt to detect that the script is blocking (for example, by adding a time-out for responses from the child process). If the response comes back, the timeout is cancelled. If the timeout is hit, all debug related buttons except for stop (which will kill the child process and clean things up) are disabled.

Thoughts..?

ntoll, 2018-06-20:
See #496

carlosperate, 2018-06-20:
Sounds good!

commit 6c281038 (2018-06-20): Fixes #494. Ensure removal of line highlighting happens before we know the state of the next highlighted line.