mu-editor/mu #497: Mu-moot 1.0

ntoll, 2018-06-21:
Getting Mu to the imminent 1.0 release has been a team effort (see: https://codewith.mu/en/thanks). 

I'd like to buy you all (UK based contributors) dinner and, as a token of thanks for your help, work and contributions, give you a special limited edition Mu t-shirt (they're awesome and will, I think, make you smile).

So, if you're mentioned below, please let me know your availability for getting to London sometime in mid-late July by adding a comment to this issue. :+1: 

cc. @carlosperate @ZanderBrown @tjguk @lordmauve @inglesp @alistair-broomhead @MissPhilbin @asb @martinohanlon @stestagg 

Also, if I've missed a contributor off the list who should be there please do shout..!

Finally, I realise not everyone will be able to make it (although I hope you all can). If you can't be there please do email your postal address to ntoll at ntoll.org so I can send you your t-shirt. ;-)

tjguk, 2018-06-21:
That's really nice of you, and I doubt I'm alone in insisting that you don't owe us anything. But, thanks anyway and I can certainly make myself available some time around then. I'm away for most of August (just in case the timelines slip!) but mid-late July is fine.

lordmauve, 2018-06-21:
I'll be at Europython the last week in July, as will many others, I suspect. I'm generally free in July other than Fridays and weekends.

alistair-broomhead, 2018-06-21:
I have to agree with @tjguk, especially given the miniscule contribution I made, but it would be great to meet up, I'll have to check my calendar for weekends, but I'm sure we can find a date that works, fancy doing a [doodle](doodle.com) for it?

ntoll, 2018-06-21:
It's the least I can do. You've all made important contributions!

It'll need to be before the final week in July because I have family commitments in the final week (a holiday -- hence I won't be at EuroPython). 

I'll do a Doodle poll once I know who's likely to be around. This is a classic case of cat herding. ;-)

carlosperate, 2018-06-21:
That is incredibly nice, and it would be awesome to hang out in a Mu-moot, thank you Nicholas!

I'll be going to EuroPython as well, so any time before that should work for me (except the 9th as there is a MicroPython meetup in London, if any of you are around come say hi :).
Also, I very much look forward to wear this special edition Mu t-shirt at EuroPython.

martinohanlon, 2018-06-22:
That is very kind and generous, reading back through the comments it looks like w/c 16th July is probably best ?!

If so - 17th, 18th, 19th, 20th all work for me.

ZanderBrown, 2018-06-22:
18th would be rather complicated, otherwise fairly open to w/c 16th July depending on timings & location

SVG->KGX is around half an hour (depending on train)

ntoll, 2018-06-25:
Hi Folks,

https://doodle.com/poll/8ati9ah3n4gr3fhg

@carlosperate @ZanderBrown @tjguk @lordmauve @inglesp @alistair-broomhead @MissPhilbin @asb @martinohanlon @stestagg -- please let me know your availability (I can't do the Wednesday since I have another engagement that evening).

Would be lovely to see you all. TBH, Tuesday or Thursday are best for me, although Monday evening is likely to be "quiet". ;-)

N.

inglesp, 2018-06-25:
Hi @ntoll,

Thanks for the invitation!  As others have said, you really don't owe us anything.

Unfortunately I'm not going to be able to get to London on any of those dates, but I'll be celebrating from afar.

Have fun!

ntoll, 2018-06-28:
OK folks... Looks like Tuesday 17th July is a go-er. Great stuff!

Let's say 5:30-ish at the Museum Tavern (opposite the British Museum): https://goo.gl/maps/wTJEdYxaUzr There are plenty of places for food around that area. We can decide what takes our fancy on the night.

@ZanderBrown can you send me your email address to ntoll <at> ntoll <dot> org, so I can add you to the calendar invite I'm about to send out.

@carlosperate @ZanderBrown @tjguk @lordmauve @inglesp @alistair-broomhead @MissPhilbin @asb @martinohanlon @stestagg

lordmauve, 2018-06-28:
See you then!

ntoll, 2018-07-16:
Hi folks... I'll close this tomorrow. ;-)

ZanderBrown, 2018-07-17:
Nice seeing you all

Hopefully I'll finally make it to PyConUK this year

ntoll, 2018-07-18:
Mu moot was stonking. Thank you all for coming along.

[image]

ntoll, 2018-07-18:
FYI... t-shirt design assets :-)

https://github.com/mu-editor/mu/tree/master/docs/swag/t-shirts

closed 2018-07-18 by ntoll

tjguk, 2018-07-18:
Great to see you all, and thanks especially to @ntoll for organising, for buying us dinner, for the wonderful T-shirts, and of course for all the work done to make Mu happen :)

tjguk, 2018-07-18:
I've only just realised the (possibly unintentional) pun in the choice of the *Mu*seum Tavern!

ntoll, 2018-07-18:
Very good... ;-)

stestagg, 2018-07-18:
Careful Tim, you may have to have your Mu credentials remuved with puns
like that ;)

tjguk, 2018-07-18:
We are not a*mu*sed :)

carlosperate, 2018-07-18:
> Great to see you all, and thanks especially to @ntoll for organising, for buying us dinner, for the wonderful T-shirts, and of course for all the work done to make Mu happen :)

I coudn't have said it better, thank you Nicholas!

P.S: Now I feel I need like I *Mu*st make a pun as well...

lordmauve, 2018-07-18:
Much-appreciated muster and munch, @ntoll!

martinohanlon, 2018-07-18:
Thanks @ntoll for dinner and thanks everyone for a nice evening :)

tjguk, 2018-07-19:
@ZanderBrown -- I'm afraid I didn't at least. Were you wearing *Mu*ted colours? Or were your socks *Mu*lberry red?

ZanderBrown, 2018-07-19:
Red tie
Green jumper
(navy) Blue trousers

In that order of course (never said it was a *good* refrence)