mu-editor/mu #498: Initial run mode oddness

ZanderBrown, 2018-06-21:
We now detect connected micro:bits and offer to switch to the relevant mode which is great!

However when we are on the initial run this leads to an interesting scenario where the mode selector pops up followed by the "would you like to switch". Interacting with the switch message goes roughly as expected but the mode switcher remains open.

All things considered this seems to be handled quite well but is a bit odd for users

ZanderBrown, 2018-07-06:
Running against master:
[image]

Running against #525:
[image]

Only change introduced is that the mode selector can't disappear behind the main window, in both branches only the mode switched accepts input. (The darkened area is a result of the WM knowing the modal relationship)

Also observed that the switch message can appear on top of the splash screen

carlosperate, 2018-07-06:
> the switch message can appear on top of the splash screen

I was just wondering if I opened a ticket for this one already, I don't think I did.
In my case, with the current Mu, the switch message appears behind the splash screen, but doesn't allow me to click on the spalsh screen to dismiss it, so I have to wait until the time-out removes it.

ntoll, 2018-07-07:
Heh... I keep seeing this too and it's on my "to fix in beta.17" list. I'll be all over this tomorrow.

tmontes, 2019-06-01:
I wanted to test this to verify what the current state of affairs would be, but I just can't seem to find my micro:bit! (why is it so micro? if it was bigger, I'm certain it would be easier to find) :)

Given how old the issue is and the note by @ntoll regarding an upcoming fix, can we say this issue is indeed fixed?

Context: I'm trying to contribute to some "issue triaging and organisation" help understand which of the bazillion open issues actually need attention. :)

carlosperate, 2019-06-01:
> I just can't seem to find my micro:bit!

If you still can't find it you can send me your details and I'll post you one :) (my twitter DMs are open, and/or my email is my github username at gmail.com).

tmontes, 2019-06-01:
Thanks a lot for your generous offer @carlosperate. :) :) :)

I'm positive that if I put a bit of effort into the task, I'll find it. It's just that when I went for it, I couldn't find it in one of the (not so few) obvious places.

dybber, 2020-09-05:
I just checked if I could still replicate this. On my Mac, something else, but just as problematic seems to happen. 

On an initial run, with a micro:bit connected already, the mode selector pop's up, but it doesn't show the "Detected new BBC micro:bit device".

If however, I didn't connect the micro:bit before opening, and then select a mode (e.g. Python mode), then the "Detected new BBC micro:bit device" is displayed when I connect the micro:bit.

dybber, 2020-10-06:
The issue with the splash screen not being hidden on mouse click is probably due to this:

> Since the splash screen is typically displayed before the event loop has started running, it is necessary to periodically call QCoreApplication::processEvents() to receive the mouse clicks.

https://doc.qt.io/qt-5/qsplashscreen.html

dybber, 2020-10-06:
Added a fix for that part, for the rest of this issue I can't replicate it on Mac.

On the initial run, only the mode selector (and splashscreen opens). It doesn't indicate that a device is detected with a pop-up, only in the status bar.

On subsequent runs, it opens on the mode used in the last session, and a pop-up asks if the user wants to change to the appropriate mode.

dybber, 2020-12-05:
@ZanderBrown could you check with newest master whether you're still experiencing this issue?

ZanderBrown, 2020-12-06:
Don't seem to be able to reproduce the original issue anymore - simply never offers to switch

Web detection seems to cause some weirdness when relaunching with an existing session though

Also wow 2018? How'd that happen

carlosperate, 2021-12-27:
We are unable to reproduce anymore, so let's close it as "resolved", thanks everyone!

closed 2021-12-27 by carlosperate