mu-editor/mu #500: CircuitPython feature request: Run mpy-cross on python modules as an option

KM4YRI, 2018-06-22:
On an Adafruit Trinket M0, I'm basically stuck using `mpy-cross` to minify a library that contains a 125-line python class down into a `.mpy` file because it doesn't have the memory to lex and parse the file in real time.  Which is understandable, but frustrating.  Any little changes to the library require me to run `mpy-cross`, copy that file to the M0 USB drive, go back to Mu, hit 'save', and then repeat over and over.

Being that `mpy-cross` is just a regular old executable that could be called in a `subprocess.Popen()` call, it wouldn't be that difficult to have Mu automate this.  Ideally the source python files would need to reside in a folder on the PC, and then 'Save' or maybe a dedicated 'Build' button (which I dearly miss from my Arduino days) would copy the files over to the virtual USB drive.

Would this be considered out of scope for this project?  Assuming that the whole intention of Mu is to make Python programming simpler, this would certainly accomplish this goal.  It would allow newcomers to write more code than they would normally be able to, without having to worry about toolchains and workflows that dip outside of what Mu can do on its own.

I'm familiar enough with Python and PyQt5 that it wouldn't be difficult to do this myself and file a pull request but wanted to open a dialog here.

eighthree, 2019-01-02:
@KM4YRI I've encountered this scenario playing with a Trinket M0 and a library that also needed to be processed by `mpy-cross`, would be a really useful improvement.

carlosperate, 2020-10-26:
The issue with mpy-cross is that it needs to match the exact version of the MicroPython runtime it is targetting.
So that would require Mu to include a mpy-cross version for each possible MicroPython target and released version supported by the editor, which is probably not in the current scope of the project.

dybber, 2021-02-25:
I will move this feature request to our new discussion board here on Github, as that's where we from now on discuss new features.

closed 2021-02-25 by dybber