mu-editor/mu #501: mu microbit gesture "shake" not work

piramid, 2018-06-23:
micro python  accelerometer.is_gesture("shake") not work

ntoll, 2018-06-23:
Please report this to MicroPython with some example code so the maintainers can reproduce the problem: 

https://github.com/bbcmicrobit/micropython

cc/ @dpgeorge

(Mu is just a code editor for writing MicroPython code on the micro:bit... the problem you have is with MicroPython itself.)

Hope this helps. :-)

closed 2018-06-23 by ntoll

dpgeorge, 2018-06-23:
This question has been asked before, and the problem may be the same as discussed in https://github.com/bbcmicrobit/micropython/issues/448

piramid, 2018-06-25:
On MAC OS is work but on Windows OS not work (mu-editor-beta16_win64.exe) . only **accelerometer.is_gesture("shake")** not work
code:
from microbit import *
while True:
    if accelerometer.is_gesture("up"):
        display.show(Image.ARROW_S)
    elif accelerometer.is_gesture("right"):
        display.show(Image.ARROW_E)
    elif accelerometer.is_gesture("down"):
        display.show(Image.ARROW_N)
    elif accelerometer.is_gesture("left"):
        display.show(Image.ARROW_W)
    elif accelerometer.is_gesture("shake"):
        display.show(Image.SAD)
    else:
        display.clear()
    sleep(20)

dpgeorge, 2018-06-25:
@piramid that's very strange that it works on Mac but now Windows.  That doesn't sound right, it should work or not, the OS doesn't matter.  The only thing I can think of is that the versions of the micro:bit MicroPython that come with the different mu's for different OS's are different.  But anyway, as mentioned above, please discuss this here: https://github.com/bbcmicrobit/micropython/issues/448