mu-editor/mu #502: Can not build/run: undefined symbol: PySlice_AdjustIndices

gquintana, 2018-06-23:
I am trying to build and run master branch == v1.0.0.beta.16

I installed a fresh new Python 3.5 virtual env, downloaded requirements (I had to apt install python3-dev)

When I launch `make run` or `make test`, I get:
```
python run.py
Traceback (most recent call last):
  File "run.py", line 2, in <module>
    from mu.app import run
  File "/home/gquintana/Perso/mu/mu/app.py", line 29, in <module>
    from PyQt5.QtCore import QTimer, Qt
ImportError: /home/gquintana/Perso/mu/.venv/lib/python3.5/site-packages/PyQt5/QtCore.so: undefined symbol: PySlice_AdjustIndices
```

The same procedure worked fine on v1.0.0.beta.15

ntoll, 2018-06-23:
What operating system are you using?

How is your version of Qt installed..?

Did you `pip install mu-editor`..?

ntoll, 2018-06-23:
Hi... Mu is Python 3 only and will never support Python 2. Try it in a virtualenv that uses Python3:

```
virtualenv --python=/usr/bin/python3 mu_venv
source mu_venv/bin/activate
pip install mu-editor
```

That should do the trick.

ntoll, 2018-06-23:
Can you let me know if this works..? If it does, I'd love to know how we could improve the pip-based installation instructions here: https://codewith.mu/en/howto/install_with_python  All feedback most welcome! :-)

gquintana, 2018-06-23:
pip install -r requirements.txt fetched 

nudatus==0.0.3
setuptools==39.2.0
pyflakes==2.0.0
pycodestyle==2.4.0
appdirs==1.4.3
qtconsole==4.3.1
PyQt5==5.11
pyserial==3.4

gquintana, 2018-06-23:
I'd like to update french translation so I need to build mu from source

ntoll, 2018-06-23:
Aha... OK... so https://mu.readthedocs.io/en/latest/setup.html is what you need. :-)

It should just work with a clean Python3 virtualenv, git clone the repos (or pull the latest code) and `pip install -r requirements.txt`.

I'm on Artful (17.10) and it works fine for me (although I realise that doesn't help you). Is there something else about your set up that could be impacting things..?

gquintana, 2018-06-23:
I rm -rf the virtual and then I followed the indicated setup. 
Now when I `pip install --editable .` I get a warning:
```
pyqtchart 5.11 has requirement PyQt5>=5.11, but you'll have pyqt5 5.10.1 which is incompatible.
```
And then when I run
```
Traceback (most recent call last):
  File "run.py", line 2, in <module>
    from mu.app import run
  File "/home/gquintana/Perso/mu/mu/app.py", line 34, in <module>
    from mu.interface import Window
  File "/home/gquintana/Perso/mu/mu/interface/__init__.py", line 1, in <module>
    from .main import Window
  File "/home/gquintana/Perso/mu/mu/interface/main.py", line 38, in <module>
    from mu.interface.editor import EditorPane
  File "/home/gquintana/Perso/mu/mu/interface/editor.py", line 25, in <module>
    from PyQt5.Qsci import QsciScintilla, QsciLexerPython, QsciAPIs
ImportError: /home/gquintana/Perso/mu/.venv/lib/python3.5/site-packages/PyQt5/Qt/lib/libQt5Core.so.5: version `Qt_5.11' not found (required by /home/gquintana/Perso/mu/.venv/lib/python3.5/site-packages/PyQt5/Qsci.so)
```
I have in my virtualenv:
PyQtChart-5.11-5.11.1-cp35.cp36.cp37.cp38-abi3-manylinux1_x86_64.whl
PyQt5-5.10.1-5.10.1-cp35.cp36.cp37.cp38-abi3-manylinux1_x86_64.whl
which are probably incompatible

ntoll, 2018-06-23:
OK... something must have happened in the releases of Qt / QScintilla / QCharts since I've just tried to re-create your steps and I'm seeing errors that were not there a couple of days ago when I last did this.

I'm investigating and will let you know how I get on.

ntoll, 2018-06-23:
Good call, I was just about to suggest that. In the meantime I'll try to work out what the latest version of PyQt has broken. :-/

Thanks for the heads up!

Also, please note, the final UI (along with translation strings) will be settled in about a week and a half.

Thanks!

gquintana, 2018-06-23:
Alright. 

We might do some beta testing with real users tomorrow :-P https://mixteen.org/blog/2018/06-mmi.html

ntoll, 2018-06-23:
Awesome! Thank you for the other bug reports too. The more feedback the better..! 

Also, if you're happy for me to do so, do you mind if I blog about your event and the way you use Mu? Perhaps you'd like to guest blog (it'd be awesome if it was in both English and French) with photos and/or a description of how it went. On Monday I'm announcing http://madewith.mu/ (to complement codewith.mu) and hearing and celebrating your story would make a great post.

titimoby, 2018-06-25:
unfortunately, we (mixteen) didn't had enough network/well configured machines to use Mu this time.
But all kids are aware of it, we will push it on our resource page... once it builds again ;)

gquintana, 2018-06-25:
@titimoby We used Mu with Microbit, I had my 2 laptops!

jerryneedell, 2018-07-10:
I am having the sample problem as the original post - same result using the virtual env.
``` 
Traceback (most recent call last):
  File "mu_venv/bin/mu-editor", line 7, in <module>
    from mu.app import run
  File "/home/jerryneedell/mu_venv/lib/python3.5/site-packages/mu/app.py", line 29, in <module>
    from PyQt5.QtCore import QTimer, Qt
ImportError: /home/jerryneedell/mu_venv/lib/python3.5/site-packages/PyQt5/QtCore.so: undefined symbol: PySlice_AdjustIndices
```

I see the reference to a new QT release, but it is not clear id there is recommended workaround.
I am on an Ubuntu 16.04 system
Thanks

askvictor, 2018-09-04:
I've just gotten the same issue trying a `pip install mu-editor` on the Linux system on a Chromebook (i.e. a debian 9 container). The issue disappears if I force the particular dependency versions that @gquintana posted though pip complains that mu-editor requires the later versions of the libraries.