mu-editor/mu #503: File dialog should be modal

gquintana, 2018-06-23:
Mu Editor 1.0b16

1. Click on the "Load" button -> the "Open File" dialog appears
2. Click on the main Mu window behind (or press Alt Tab) --> the "Open File" dialog goes behind the main window
At this point, the main window is in front but doesn't react to anything as if the editor was stuck.

Workaround:
Press click on the "Open file" window in the task bar (or press Alt Tab) to get the "Open File" dialog in foreground and close it

ZanderBrown, 2018-06-23:
I almost had a fix on this a while ago, I'll take another look

ZanderBrown, 2018-06-23:
Hmm we find ourselves in the interesting area where we may be doing something wrong or Qt has a bug

Seems the problem partially stems from the qt native dialog feature but it wouldn't be user friendly to disable that

ZanderBrown, 2018-06-23:
I've confirmed that dialogs are correctly modal when Qt is forced to use its built-in dialog but that isn't a great user experience for the majority of us that are not using KDE

gquintana, 2018-06-25:
The problem doesn't occur on Windows, it's something platform related

ZanderBrown, 2018-06-25:
Yes it's Qt's Gtk3 integration at fault

carlosperate, 2018-07-05:
Using the latest master and PyQT5 5.11 the file dialg acts like a modal in macOS. Has Mu been updated? Or is Qt behaving differently in different platforms?

ZanderBrown, 2018-07-05:
Yeah Qt has multiple FileChooser backends for native integration, as far as I can tell only the Gtk3 (GNOME ext) backend has this bug. KDE definitely doesn't because that's just plain Qt and I'm pretty sure Windows is okay

Tried to work out how to report it but ended up in circles, unfortunately without some hacky user agent sniffing type behaviour there isn't much we can do about this

ntoll, 2018-07-06:
Folks, thanks for all the investigations around this.

As far as I can tell, we'd be creating a hack within Mu because of an upstream problem. The *ideal* course of action is for upstream to fix this. Is there a ticket in Qt we can add to?

ZanderBrown, 2018-07-06:
Not currently (That I could find)

ntoll, 2018-07-07:
OK... well volunteered... :-P Care to link back to this issue..?

(I would do this myself, but you have a better handle on the context and I wouldn't want to submit something that doesn't fully explain the problem.)

ZanderBrown, 2018-07-07:
Not sure I've added enough detail but I think I've jumped through the right number of hoops [QTBUG-69354](https://bugreports.qt.io/browse/QTBUG-69354)

ntoll, 2018-07-07:
OK given this is upstream from us, I'm closing this issue since we're relying on Qt to fix.

@ZanderBrown thanks for the efforts here... and for fixing #525 so other modal dialogs behave.

closed 2018-07-07 by ntoll