mu-editor/mu #504: ImportError: No module named 'mu' when using Debugger

gquintana, 2018-06-23:
Trying to use Debugger in Python 3 mode, I get the error:

1. Open Python 3 script
2. Add a breakpoint
3. Run debugger

```
Traceback (most recent call last):
  File "/home/gquintana/Perso/mu/mu/mu-debug.py", line 2, in <module>
    from mu.app import debug
ImportError: No module named 'mu'
```
Logs
```
2018-06-23 16:25:41,510 - root:111(run) INFO: 

-----------------

Starting Mu 1.0.0.beta.16
2018-06-23 16:25:41,511 - root:112(run) INFO: uname_result(system='Linux', node='XPS-15-GQA', release='4.13.0-45-generic', version='#50~16.04.1-Ubuntu SMP Wed May 30 11:18:27 UTC 2018', machine='x86_64', processor='x86_64')
2018-06-23 16:25:41,511 - root:113(run) INFO: Python path: ['/home/gquintana/Perso/mu', '/home/gquintana/Perso/mu/.venv/lib/python35.zip', '/home/gquintana/Perso/mu/.venv/lib/python3.5', '/home/gquintana/Perso/mu/.venv/lib/python3.5/plat-x86_64-linux-gnu', '/home/gquintana/Perso/mu/.venv/lib/python3.5/lib-dynload', '/usr/lib/python3.5', '/usr/lib/python3.5/plat-x86_64-linux-gnu', '/home/gquintana/Perso/mu/.venv/lib/python3.5/site-packages', '/home/gquintana/Perso/mu/.venv/lib/python3.5/site-packages/IPython/extensions']
2018-06-23 16:25:41,674 - mu.logic:542(__init__) INFO: Setting up editor.
2018-06-23 16:25:41,689 - mu.logic:556(__init__) INFO: Settings path: /home/gquintana/.local/share/mu/settings.json
2018-06-23 16:25:41,689 - mu.logic:557(__init__) INFO: Session path: /home/gquintana/.local/share/mu/session.json
2018-06-23 16:25:41,689 - mu.logic:558(__init__) INFO: Log directory: /home/gquintana/.cache/mu/log
2018-06-23 16:25:41,689 - mu.logic:559(__init__) INFO: Data directory: /home/gquintana/.local/share/mu
2018-06-23 16:25:41,714 - mu.logic:572(setup) INFO: Available modes: adafruit, debugger, microbit, pygamezero, python
2018-06-23 16:25:41,874 - mu.logic:1077(change_mode) INFO: Workspace directory: /home/gquintana/mu_code
2018-06-23 16:25:41,890 - mu.logic:621(restore_session) INFO: Restoring session from: /home/gquintana/.local/share/mu/session.json
2018-06-23 16:25:41,890 - mu.logic:622(restore_session) DEBUG: {'envars': [], 'paths': [], 'minify': False, 'theme': 'day', 'microbit_runtime': '', 'mode': 'python'}
2018-06-23 16:25:41,890 - mu.logic:643(restore_session) INFO: Loaded files.
2018-06-23 16:25:41,890 - mu.logic:647(restore_session) INFO: User defined environment variables: []
2018-06-23 16:25:41,890 - mu.logic:651(restore_session) INFO: Minify scripts on micro:bit? False
2018-06-23 16:25:42,044 - mu.logic:669(restore_session) INFO: Starting with blank file.
2018-06-23 16:25:42,058 - mu.logic:1077(change_mode) INFO: Workspace directory: /home/gquintana/mu_code
2018-06-23 16:29:50,921 - mu.logic:1011(show_admin) INFO: Showing logs from /home/gquintana/.cache/mu/log/mu.log
2018-06-23 16:30:13,398 - mu.logic:1042(select_mode) INFO: Showing available modes: ['adafruit', 'debugger', 'microbit', 'pygamezero', 'python']
2018-06-23 16:31:11,859 - mu.logic:691(new) INFO: Added a new tab.
2018-06-23 16:31:16,641 - mu.modes.python3:229(toggle_repl) INFO: Toggle REPL on.
2018-06-23 16:31:16,643 - mu.modes.python3:58(start_kernel) INFO: ['/home/gquintana/Perso/mu', '/home/gquintana/Perso/mu/.venv/lib/python35.zip', '/home/gquintana/Perso/mu/.venv/lib/python3.5', '/home/gquintana/Perso/mu/.venv/lib/python3.5/plat-x86_64-linux-gnu', '/home/gquintana/Perso/mu/.venv/lib/python3.5/lib-dynload', '/usr/lib/python3.5', '/usr/lib/python3.5/plat-x86_64-linux-gnu', '/home/gquintana/Perso/mu/.venv/lib/python3.5/site-packages', '/home/gquintana/Perso/mu/.venv/lib/python3.5/site-packages/IPython/extensions']
2018-06-23 16:31:16,644 - mu.modes.python3:63(start_kernel) INFO: Starting iPython kernel with user defined envars: {}
2018-06-23 16:31:16,798 - traitlets:257(start_kernel) DEBUG: Starting kernel: ['/home/gquintana/Perso/mu/.venv/bin/python', '-m', 'ipykernel_launcher', '-f', '/tmp/tmpql3ntcr0.json']
2018-06-23 16:31:16,803 - traitlets:542(_create_connected_socket) DEBUG: Connecting to: tcp://127.0.0.1:35867
2018-06-23 16:31:16,808 - asyncio:53(__init__) DEBUG: Using selector: EpollSelector
2018-06-23 16:31:16,809 - asyncio:53(__init__) DEBUG: Using selector: EpollSelector
2018-06-23 16:31:16,809 - traitlets:140(shell_channel) DEBUG: connecting shell channel to tcp://127.0.0.1:38329
2018-06-23 16:31:16,809 - traitlets:542(_create_connected_socket) DEBUG: Connecting to: tcp://127.0.0.1:38329
2018-06-23 16:31:16,810 - traitlets:152(iopub_channel) DEBUG: connecting iopub channel to tcp://127.0.0.1:47095
2018-06-23 16:31:16,810 - traitlets:542(_create_connected_socket) DEBUG: Connecting to: tcp://127.0.0.1:47095
2018-06-23 16:31:16,811 - traitlets:164(stdin_channel) DEBUG: connecting stdin channel to tcp://127.0.0.1:42067
2018-06-23 16:31:16,811 - traitlets:542(_create_connected_socket) DEBUG: Connecting to: tcp://127.0.0.1:42067
2018-06-23 16:31:16,811 - traitlets:176(hb_channel) DEBUG: connecting heartbeat channel to tcp://127.0.0.1:57833
2018-06-23 16:31:17,667 - traitlets:212(_handle_history_reply) DEBUG: Received history reply with 0 entries
2018-06-23 16:31:23,406 - traitlets:545(_handle_inspect_reply) DEBUG: oinfo: {'data': {'text/plain': "\x1b[0;31mDocstring:\x1b[0m\nprint(value, ..., sep=' ', end='\\n', file=sys.stdout, flush=False)\n\nPrints the values to a stream, or to sys.stdout by default.\nOptional keyword arguments:\nfile:  a file-like object (stream); defaults to the current sys.stdout.\nsep:   string inserted between values, default a space.\nend:   string appended after the last value, default a newline.\nflush: whether to forcibly flush the stream.\n\x1b[0;31mType:\x1b[0m      builtin_function_or_method\n"}, 'found': True, 'metadata': {}, 'status': 'ok'}
2018-06-23 16:31:29,334 - traitlets:140(shell_channel) DEBUG: connecting shell channel to tcp://127.0.0.1:38329
2018-06-23 16:31:29,334 - traitlets:542(_create_connected_socket) DEBUG: Connecting to: tcp://127.0.0.1:38329
2018-06-23 16:31:29,347 - traitlets:235(_handle_execute_input) DEBUG: execute_input: {'code': 'print("Hello world")\n', 'execution_count': 1}
2018-06-23 16:31:29,349 - traitlets:566(_handle_stream) DEBUG: stream: {'name': 'stdout', 'text': 'Hello world\n'}
2018-06-23 16:31:29,351 - traitlets:461(_handle_execute_reply) DEBUG: execute: {'execution_count': 1, 'user_expressions': {}, 'status': 'ok', 'payload': []}
2018-06-23 16:31:35,735 - mu.modes.python3:233(toggle_repl) INFO: Toggle REPL off.
2018-06-23 16:31:37,477 - mu.modes.python3:269(toggle_plotter) INFO: Toggle plotter on.
2018-06-23 16:31:37,486 - mu.modes.python3:280(add_plotter) INFO: Started plotter
2018-06-23 16:32:47,178 - mu.modes.python3:272(toggle_plotter) INFO: Toggle plotter off.
2018-06-23 16:32:47,179 - mu.modes.base:146(remove_plotter) DEBUG: Creating directory: /home/gquintana/mu_code/data_capture
2018-06-23 16:32:47,180 - mu.modes.base:156(remove_plotter) INFO: Removing plotter
2018-06-23 16:32:50,254 - mu.logic:903(zoom_in) INFO: Zoom in
2018-06-23 16:32:52,133 - mu.logic:910(zoom_out) INFO: Zoom out
2018-06-23 16:32:53,398 - mu.logic:684(toggle_theme) INFO: Toggle theme to: night
2018-06-23 16:32:54,887 - mu.logic:684(toggle_theme) INFO: Toggle theme to: contrast
2018-06-23 16:32:56,212 - mu.logic:684(toggle_theme) INFO: Toggle theme to: day
[... 685 lines cut ...]
2018-06-23 17:14:30,931 - mu.logic:622(restore_session) DEBUG: {'theme': 'day', 'microbit_runtime': '', 'minify': False, 'mode': 'python', 'paths': ['/home/gquintana/mu_code/multiplication.py'], 'envars': []}
2018-06-23 17:14:30,931 - mu.logic:704(_load) INFO: Loading script from: /home/gquintana/mu_code/multiplication.py
2018-06-23 17:14:30,931 - mu.logic:271(read_and_decode) DEBUG: Trying to decode with utf-8
2018-06-23 17:14:30,931 - mu.logic:274(read_and_decode) INFO: Decoded with utf-8
2018-06-23 17:14:30,932 - mu.logic:287(read_and_decode) DEBUG: Detected newline '\n'
2018-06-23 17:14:30,932 - mu.logic:778(_load) DEBUG: for i in range(1,10):
    for j in range(1,10):
        print("{0}x{1}={2}".format(i,j, i*j))
2018-06-23 17:14:31,014 - mu.logic:643(restore_session) INFO: Loaded files.
2018-06-23 17:14:31,014 - mu.logic:647(restore_session) INFO: User defined environment variables: []
2018-06-23 17:14:31,014 - mu.logic:651(restore_session) INFO: Minify scripts on micro:bit? False
2018-06-23 17:14:31,029 - mu.logic:1077(change_mode) INFO: Workspace directory: /home/gquintana/mu_code
2018-06-23 17:14:39,688 - mu.modes.python3:219(debug) INFO: Starting debug mode.
[... cut ...] gquintana/Perso/mu/.venv/lib/python3.5/site-packages/IPython/extensions']
2018-06-23 17:14:43,744 - mu.logic:1011(show_admin) INFO: Showing logs from /home/gquintana/.cache/mu/log/mu.log
2018-06-23 17:14:49,549 - mu.debugger.client:196(on_fail) ERROR: Echec de la connexion. Es-ce que la machine est lente ou très occupée? Libère des ressources sur la machine, puis essaie à nouveau.
2018-06-23 17:16:39,028 - mu.logic:1011(show_admin) INFO: Showing logs from /home/gquintana/.cache/mu/log/mu.log
```

carlosperate, 2018-07-05:
Did you install Mu by cloning the repo, creating a virtualenv, and running `pip install -r requirements.txt`?
If so you will also need to run `pip install -e .`, more info in the dev docs: https://mu.readthedocs.io/en/latest/setup.html

carlosperate, 2018-09-17:
@gquintana as it has been a while I assume you have been able to resolve the issue by executing `pip install -e`. But if this is still a problem or you've been able to resolve it by doing something else please feel free to reopen the issue.

closed 2018-09-17 by carlosperate