mu-editor/mu #505: REPL, long running functions() and Micro:bit connection

gquintana, 2018-06-23:
1. Plug a Micro:bit
2. Connect REPL on it
3. Start a compass calibration `compass.calibrate()` -> Mu waits for calibration end
4. Unplug Micro:bit (or just lose connection) -> Mu keeps waiting

ZanderBrown, 2018-06-23:
What do you mean by "Mu keeps waiting"?
What are you expecting to happen that isn't?

ntoll, 2018-06-23:
I guess something that says, "Disconnected: the micro:bit appear to have been unplugged." for example...???

ZanderBrown, 2018-06-23:
Such a message in the REPL may reduce any confusion

gquintana, 2018-06-23:
The problem is that the UI/REPL is frozen until the remote function call ends, I see 2 related solutions:
* Execute the remote function call in separate thread  to let the UI have live its own life
* Execute the remote function call with a timeout

ZanderBrown, 2018-06-23:
Oh I see Mu as a whole locks up

I'd vote for separate thread rather than timeout (have we had this discussion before?)

ntoll, 2018-07-10:
I can't recreate this (using Linux) as per the original instructions. Mu remains responsive and, as expected, you can't do anything with the REPL pane, but close it (which is to be expected). Am I missing something here..?

gquintana, 2018-07-11:
I'll test against b17 and see if I can still reproduce.

But as you said, if you can't do anything but close it (and potentially lose what you've done or feel like Mu is completely dead), this is not what I would call "expected". Personally I would
 expect to get back in control after a minute (timeout) with an error message saying something went wrong.

ntoll, 2018-07-11:
When you say "Mu is completely dead" what exactly do you mean? ;-)

When I tried to reproduce, Mu remained responsive. The only thing which may appear "dead" is the REPL, but even then, it still responded to Ctrl-C and Ctrl-D to interrupt and soft reboot. My concern is the problem is on the board, rather than in Mu, and Mu is just (correctly) showing the problematic state the board is in.

Does this make sense? Just trying to ascertain exactly how *Mu* (not the device) is getting into a problematic state (and what that problematic behaviour is precisely). (Sorry).

gquintana, 2018-07-11:
Things seems better with b17: I can come back to the editor

- In Python 3 mode Ctrl+C stops the running while True loop, but not in the Microbit mode running compass.calibrate() (an endless loop is different from a long remote operation)
- Once compass.calibrate() launched (and not progressing), closing and reopening the REPL doesn't help: I have to unplug the microbit to get back in control
- Once compass.calibrate() launched and Microbit unplugged/replugged, I have to close/reopen the REPL several times to be able to reconnect (Cannot connect to device... error message and REPL stuck). I suggest that when connection to device fails, not to open the REPL (which is stuck).

ntoll, 2018-07-12:
@gquintana regarding the untranslatable string... see update here: https://github.com/mu-editor/mu/blob/master/mu/interface/main.py#L353 (thanks for spotting this).

Regarding compass... I guess that's because compass calibration is a blocking operation *on the device* so there's nothing Mu can do about it until the device has finished. Perhaps raise this as an issue in MicroPython? https://github.com/bbcmicrobit/micropython/issues

(Mu is sending data down the serial line, but if the micro:bit is just buffering it while the compass calibration blocks, then there's not much Mu can do I'm afraid. Sorry.)

dybber, 2020-09-29:
I just tried to see where we are at today:

 -  I connected a micro:bit, and opened the REPL in MU
 - In the REPL I entered:
```python
from microbit import compass
compass.calibrate()
```
- I plugged out the microbit, the REPL closed and I received this error message:
[image]
 - I tried again: connecting the micro:bit and calling the calibration routine
 - After a while and after me swinging around the microbit, it was satified calibrated and returned.

I don't think there's much more we can do here, calibration just is an operation that takes time, and even requires user input. We do give an error message, and it's now also displayed in the bottom right corner whether a device is connected or not:

[image]
[image]

I'm thus closing the issue.

closed 2020-09-29 by dybber