mu-editor/mu #508: Materialize code blocks scope

gquintana, 2018-06-25:
While explaining Python to teenagers, I had hard times trying to explaining why they should indent code and where they should indent code.

I would find it very useful to visually represent the scope of a code block: "this if statement starts here and finishes here", "this while statement starts here and ends here". It would allow visually checking whether code is properly indented/aligned.

Yet I have no idea whether Scintilla has such a feature.

gquintana, 2018-06-25:
https://www.scintilla.org/ScintillaDoc.html#TabsAndIndentationGuides

    "Scintilla can also display indentation  guides (vertical lines) to help you to generate code."

ntoll, 2018-06-25:
Mu already does this!
[image]

closed 2018-06-25 by ntoll