mu-editor/mu #509: Twisted library makes building Mu complicated

gquintana, 2018-06-25:
When installing Python requirements to build and run Mu, the "Twisted" library requires a C++ compiler to be present: GCC on Linux, Visual C++ Build Tools 14.0 on Windows.

As a result, setting up a dev environment of Mu is complicated.

This library seems to be drawn from the "scrapy" requirement
Are those library really necessary?

ZanderBrown, 2018-06-25:
Seems it's needed for [adafruit](https://github.com/mu-editor/mu/blob/803153f661097260206ed2b7cc9a1e71b564d7c3/utils/adafruit_api.py) but that seems a little...    hacky

gquintana, 2018-06-25:
I ignore what a wheel is, I am just doing a pip install -r requirements.txt.
Both on Linux and Windows, a C++ compiler is required to run Mu from sources. This is painful. 
Removing scrapy from requirements.txt is good but ugly trick.

ntoll, 2018-06-25:
Sure... please remember that `requirements.txt` is for setting up everything needed for the development environment and, as a result, we expect developers to know what they're doing (so it's not *that* bad to expect developers to know how to install the things they'll need, like a compiler). ;-)