mu-editor/mu #511: Translate API help

gquintana, 2018-06-25:
Why is the API help in mu/modes/api excluded from translation by `make translate`?
It would be super useful to have it in other languages.

ntoll, 2018-07-10:
The API help is auto-generated from the various project's documentation. To be honest, these should be translated upstream. However, this usually isn't the case (for example, Python has minimal French support here: https://docs.python.org/fr/3/).

The task of translating such API docs is *huge*. That's why they're separated away from `make translate`. The difference is `make translate` is for Mu itself, where as `make translate-all` is for the whole universe. ;-)

closed 2018-07-10 by ntoll

gquintana, 2018-07-10:
The strings I mention in this issue are not in Python libraries but in Mu code, it's a juste a matter of removing the exclusion in Makefile and translating some more strings.

ntoll, 2018-07-10:
Sure... but they're automatically generated from upstream libraries (see the scripts in the `utils` directory).

ntoll, 2018-07-10:
I completely agree with you. So my idea regarding partial translation is:

* Translate the Mu UI (which you've done already).
* Update the script that kicks in when you type `make translate-all` to allow you to select desired modules / projects (e.g. just micro:bit and CircuitPython for example).

However, this is a hard problems. For example, the REPL (iPython) actually introspects the code as it goes along... since the docstrings are in English, there's nothing we can do to translate this stuff (unless, of course, you've imported code with French docstrings into the REPL -- sadly Python standard lib is all English).

I wonder if there's a "Translations" working group in the PSF..? They will have lots of experience and suggestions in this regard.

ntoll, 2018-07-10:
Bingo... https://mail.python.org/mailman/listinfo/i18n-sig

tjguk, 2018-07-10:
I'm not sure if I understand where the problem lies here. Am I right in thinking the following?

* Generated API docstrings are gettext-aware
* In principle there's nothing to stop one, some or all of the docstrings being translated into language X

(this is where I'm uncertain)
* We'd rather not have *some* but not *all* strings translated for a given API
* There are some technical barriers (involving poedit or gettext or whatever) which prevent a partial translation of a .py file

ntoll, 2018-07-10:
For me it's simply this is a *huge* undertaking, and Mu probably isn't the place to do it (i.e. it should be upstream in the original projects -- MicroPython, CircuitPython, PyGameZero, Python) etc...

tjguk, 2018-07-10:
Ok. Can't argue with that. I wasn't sure if you were suggesting that there was a local technical issue. (Although I'm not sure how it would work even if the upstream projects *did* translate their docstrings...)

ntoll, 2018-07-10:
@tjguk I've written some scripts which automate the extraction of docstrings from Python and turns them into a format QScintilla understands for the purposes of auto-complete and call tips. The side effect is that they're translatable too. This could be useful for relatively small APIs like CircuitPython or MicroPython on the micro:bit.

ntoll, 2018-07-10:
So, no, there's no technical issue. It's more a problem of organisation / scale and coordination of translation strings.