mu-editor/mu #513: Mu not detecting connected micro:bit

dxf, 2018-06-27:
* What you were trying to do: Open the files dialog box.
* What steps you took to make this happen: Flashed file with "from microbit import *" on the m:b, then clicked files.
* What you expected to happen: files opens
* What actually happened: 
[image]

* Why this difference is problematic (it may not be a bug!): I can't use the app properly...
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. Win10 LTSB 2016, 0.9.13 (happens with latest beta too), exe from codewith.mu

ntoll, 2018-06-29:
Hi @dxf,

So this dialog is shown when Mu cannot detect a serial connection to the micro:bit. To be honest, this relies entirely on the underlying operating system to work properly. If I were you I'd avoid 0.9.13 and update to the latest beta.

Have you checked the following..?

* You waited up to a minute before trying to open the files? This is entirely dependent on how long the underlying OS takes to register there's a serial device available for connection. Sometimes this takes no time at all, sometimes it takes a minute.

* Do you have the latest firmware on your micro:bit..? Please check / read this article to be sure: https://support.microbit.org/support/solutions/articles/19000019131-how-to-upgrade-the-firmware-on-the-micro-bit

* Is your system configured in some unusual way? We test Mu extensively on Windows 10 and don't encounter this problem except for when we have an old micro:bit (update the firmware!) or the OS is being slow (as described in the first point).

closed 2018-06-29 by ntoll

dxf, 2018-07-16:
Hi there,

Sorry I didn't see this, github didn't ping me. I'll try them in September when I go back to school where the m:bs are

jnank1, 2021-06-23:
I got the same message with a Micro:bit V2 on Windows 8.1.  I had upgraded Mu to the lastest version (64-bit).  I also tried on Windows 10 and just used the older version of Mu because my Windows 10 was 32-bit.  I did not get the message on the older Mu.  On Windows 8.1, which is 64-bit, with the new version on Mu, I not only got the "Could not find an attached BBC Micro:bit" and there was a red "X" in the lower right to indicate this.  However, Windows 10 was recognizing the Micro:bit on usb, and I was in fact, able to copy a .hex file to the Micro:bit usb drive and it worked ran on the Micro.bit.  I am not sure what gives with the new 64-bit Mu and recognizing the Micro:bit on usb.  Just another note, the web version on micropython works fine on the same Microbit V2.

carlosperate, 2021-06-23:
Thanks for the report @jnank1, as this is an old issue I've created https://github.com/mu-editor/mu/issues/1659 to look into this.