mu-editor/mu #515: macOS, annoying messages after upload to micro:bit

sstaub, 2018-06-28:
v1.0.0b16

Using micro:bit with macOS 10.13.5

After each upload to the micro:bit you get the same annoying message,
that you unmount the device not correctly. (see german screenshot)
That is a macOS problem which can solved by uploading the code
via CMSIS-DAP instead using MSD, I think.
I have had the same problem with other mbed devices using PlatformIO,
and it was solved with this method.
Hope you can do it also by this way.

[image]

jaustin, 2018-07-12:
@sstaub this behaviour should be fixed (well, more accurately worked around) by the new flashing flow that's been integrated by ntoll in beta17. Could you try that out and confirm that on your machine the dialog is gone (possibly you will still see this for the *first* flash as we have to get the latest version of MicroPython on your board, but after that I hope subsequent flashes would be without the message) - Download is here https://codewith.mu/en/download

sstaub, 2018-07-12:
Yeah, it is working now.

closed 2018-07-12 by sstaub