mu-editor/mu #518: Feature request: language selector

aroberge, 2018-06-29:
I just downloaded Mu for the first time and am I very impressed by it. 

My OS locale is French. While Mu's interface was displayed in English but Mu somehow knew that my preferred language is French as it brought me to the non-existant https://codewith.mu/fr/help/1.0 page.

I think it would be useful if Mu had a language selector instead as a menu item instead of relying on the locale of the OS (as I am guessing it does) and then defaulting to English if it is missing.  The preferred language of the user (especially if they use Mu in a computer lab) may be a different choice altogether.

gitterman, 2019-01-21:
I am located in Germany and my regional settings on MS windows 7x64 are of course German to have the correct date-time-currency settings.
My system is US english though which is also my default locale as well as my keyboard layout.
Thus deriving the interface language from regional settings is not what I need.
Auto-detecting the language is a nice idea but especially in a programmers environment it is often preferred to use international English instead of local language. 
Please add a setting which allows the user to easily change the GUI language.
I found that I can override the language in the "__init__.py" but this is not what I would regard as ease of use.

carlosperate, 2020-09-29:
Thank you for the feature request @aroberge and @gitterman, as issue https://github.com/mu-editor/mu/issues/692 contains an advanced discussion I'll close this ticket to keep the conversation in a single place.

closed 2020-09-29 by carlosperate