mu-editor/mu #519: mu 0.9.13:  TypeError: connect() failed between (int,int,Qt::KeyboardModifiers) and unislot()

hroncok, 2018-06-29:
Hi, I'm trying to rebuild the mu RPM package 0.9.13 in Fedora 29 with Python 3.7, qscintilla 2.10.5, pyqt5 5.10.2, qt 5.11.1.

I get `TypeError: connect() failed between (int,int,Qt::KeyboardModifiers) and unislot()`

<details>
<summary>Full pytest FAILURES</summary>

```
=================================== FAILURES ===================================
__________________________ test_EditorPane_configure ___________________________
    def test_EditorPane_configure():
        """
        Check the expected configuration takes place. NOTE - this is checking the
        expected attributes are configured, not what the actual configuration
        values may be. I.e. we're checking that, say, setIndentationWidth is
        called.
        """
        api = ['api help text', ]
>       ep = mu.interface.EditorPane('/foo/bar.py', 'baz', api)
tests/test_interface.py:126: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
mu/interface.py:226: in __init__
    self.configure()
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
self = <mu.interface.EditorPane object at 0x7fffa7703ca8>
    def configure(self):
        """
            Set up the editor component.
            """
        # Font information
        font = Font().load()
        self.setFont(font)
        # Generic editor settings
        self.setUtf8(True)
        self.setAutoIndent(True)
        self.setIndentationsUseTabs(False)
        self.setIndentationWidth(4)
        self.setTabWidth(4)
        self.setEdgeColumn(79)
        self.setMarginLineNumbers(0, True)
        self.setMarginWidth(0, 50)
        self.setBraceMatching(QsciScintilla.SloppyBraceMatch)
        self.SendScintilla(QsciScintilla.SCI_SETHSCROLLBAR, 0)
        self.set_theme()
        # Markers and indicators
        self.markerDefine(self.RightArrow, self.MARKER_NUMBER)
        self.setMarginSensitivity(1, True)
        self.setIndicatorDrawUnder(True)
        for type_ in self.check_indicators:
            self.indicatorDefine(
                self.SquiggleIndicator, self.check_indicators[type_]['id'])
        for type_ in self.search_indicators:
            self.indicatorDefine(
                self.StraightBoxIndicator, self.search_indicators[type_]['id'])
        self.setAnnotationDisplay(self.AnnotationBoxed)
>       self.marginClicked.connect(self.on_marker_clicked)
E       TypeError: connect() failed between (int,int,Qt::KeyboardModifiers) and unislot()
mu/interface.py:258: TypeError
__________________________ test_EditorPane_set_theme ___________________________
    def test_EditorPane_set_theme():
        """
        Check all the expected configuration calls are made to ensure the widget's
        theme is updated.
        """
        api = ['api help text', ]
>       ep = mu.interface.EditorPane('/foo/bar.py', 'baz', api)
tests/test_interface.py:184: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
mu/interface.py:226: in __init__
    self.configure()
[... 645 lines cut ...]
        # Markers and indicators
        self.markerDefine(self.RightArrow, self.MARKER_NUMBER)
        self.setMarginSensitivity(1, True)
        self.setIndicatorDrawUnder(True)
        for type_ in self.check_indicators:
            self.indicatorDefine(
                self.SquiggleIndicator, self.check_indicators[type_]['id'])
        for type_ in self.search_indicators:
            self.indicatorDefine(
                self.StraightBoxIndicator, self.search_indicators[type_]['id'])
        self.setAnnotationDisplay(self.AnnotationBoxed)
>       self.marginClicked.connect(self.on_marker_clicked)
E       TypeError: connect() failed between (int,int,Qt::KeyboardModifiers) and unislot()
mu/interface.py:258: TypeError
==================== 15 failed, 152 passed in 6.12 seconds =====================
```
</details>

I wonder where does `unislot()` come from. Any ideas what's wrong?

(I am not able to update to 1.0.0 just yet (couldn't find PyQt5.QtChart in Fedora).)

Full build log: https://kojipkgs.fedoraproject.org//work/tasks/4867/27944867/build.log

Full root log (package versions, etc): https://kojipkgs.fedoraproject.org//work/tasks/4867/27944867/root.log

ntoll, 2018-06-29:
Hi @hroncok,

My advice is to not use 0.9.13. It's old, full of bugs and 1.0.beta.16 (soon to be final) is far better. Perhaps package that instead.

I've `unislot` problems one time before, and that's been due to version mis-matches between the various Qt libraries / components and Python builds. Perhaps 3.7 isn't yet fully supported..?

Please do let me know how you get on. Alternatively, if there's anything I can do to kill off 0.9.13 and update Fedora to 1.0.final (when it arrives in about 3 weeks) then please let me know.

hroncok, 2018-06-30:
I'm working on getting PyQt5.QtChart into Fedora.

ntoll, 2018-06-30:
@hroncok oooh..... that's wonderful. If there's anything I can do to help, please don't hesitate to give me a kick. cc/@kushaldas (Kushal originally packaged Mu and may have insight / opinions / help to offer in this respect). Thank you! :-)

ntoll, 2018-06-30:
BTW... beta.17 will be out next week. A release candidate the week after and then hopefully (assuming no show-stoppers) 1.0.final soon after that. If you're going to package Mu in Fedora (and, please do!) make sure it's 1.0.final that eventually gets into the package.

hroncok, 2018-06-30:
mu in Fedora is 0.9.13. I will update it, but if I hit this problem, I'm screwed anyway.

hroncok, 2018-07-02:
1.0.0 beta.16 with Python 3.6 (not even 3.7 yet):

```
tests/interface/test_main.py:1495: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
mu/interface/main.py:263: in add_tab
    new_tab.connect_margin(self.breakpoint_toggle)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <mu.interface.editor.EditorPane object at 0x7f17150e40d8>
func = <MagicMock id='139737108954864'>

    def connect_margin(self, func):
        """
            Connect clicking the margin to the passed in handler function.
            """
>       self.marginClicked.connect(func)
E       TypeError: connect() failed between (int,int,Qt::KeyboardModifiers) and unislot()

mu/interface/editor.py:170: TypeError
```

hroncok, 2018-07-02:
I guess I need to try a minimal reproducer and check when this started happening. Maybe it's a bug in Fedora's PyQt5 or QScintilla build.

ZanderBrown, 2018-07-02:
How odd, works fine for me

Fedora 28, (other versions to follow...)

hroncok, 2018-07-02:
This is using the RPM packages. Are you testing with pip installed or RPM installed?

ZanderBrown, 2018-07-02:
RPM i believe... (checking)

ZanderBrown, 2018-07-02:
Sorry for the delay (temperature is becoming an issue round here!)

Okay looks like pip was overriding RPM

However 99% sure i'm now running against: (Plotter has dissapeared as expected)
```
python3-qt5-5.10-3.fc28.x86_64
python3-qscintilla-qt5-2.10.2-3.fc28.x86_64
python3-3.6.5-1.fc28.x86_64
```

hroncok, 2018-07-02:
Find a shorter reproducer, reported in fedora: https://bugzilla.redhat.com/show_bug.cgi?id=1597386

ntoll, 2018-07-02:
@hroncok thanks for all your work with this issue! :-)

hroncok, 2018-08-14:
The underlying issue is gone :tada: 

Working towards 1.0.0  (currently we don't have gpiozero, guizero and pigpio).

closed 2018-08-14 by hroncok