mu-editor/mu #520: Make "install for everyone" the default option in the Windows installer.

ntoll, 2018-06-29:
Update the pynsist configuration file to reflect this.

tjguk, 2018-06-29:
What's the thinking behind this change @ntoll?

ntoll, 2018-06-29:
See the conversation on Gitter (https://gitter.im/mu-editor/general). A user is having problems installing on a suite of computers.

tjguk, 2018-07-01:
I haven't yet started to dive into the world of NSIS/pynsist. But I can confirm that this invocation installs silently for all users on my Win10 machine:

  mu-editor_beta16_win64.exe /S /AllUsers

It installs to %ProgramFiles%\Mu as you would expect, and no amount of "/D c:\whatever..." at the end of the command-line will make any difference.

I don't know if this is enough to get the OP up and running. But I will start to dig into the NSIS/pynsist  tooling.

ntoll, 2018-07-01:
@tjguk ooh... That incantation is *exactly* what we need.

Please don't investigate any further since I believe we have a working solution that just needs to be in the docs / website (I'll do that right now). Also, IIRC you mentioned that installing for the current user is the current convention for default. Ergo, we have a work-around and we shouldn't change what we're doing.

As always, THANK YOU for finding this. It would have taken me *ages* to figure this out. :-)

I'll close this issue once the website / docs are updated.

ntoll, 2018-07-01:
Well, the Python team at MS is pretty authoritative for me... ;-)

tjguk, 2018-07-01:
Ok; Steve's confirmed that a per-user default is the right way, principally because the average user shouldn't have admin privs enough to install system-wide:

https://twitter.com/zooba/status/1013445957290061824

So I think we're good to go with the docs change.

ntoll, 2018-07-01:
OK... docs have been changed [here](https://github.com/mu-editor/mu-editor.github.io/commit/e631138cdd63e35ddcc2f54ea822dcb12c98f52b) (for the main website), and [here]() (in the developer docs).

See: https://codewith.mu/en/howto/install_windows#admins for the short-cut on the main website and https://mu.readthedocs.io/en/latest/packaging.html#windows-installer in the developer docs.

closed 2018-07-01 by ntoll