mu-editor/mu #523: On Windows, screen reader reads lines out of turn

ntoll, 2018-07-04:
This has been demonstrated on more than one screen reader. This doesn't appear to happen when the file that's being edited has been loaded from the file system. I.e. it only works with tabs that were originally created from new.

Steps to reproduce:

* Open new tab
* Type the following:

```
1
2
3
4
5
6
7
8
9
10
```

* Move between lines using down and up arrow keys. 
* Screen reader will say things like "1, 2" instead of "1".

tjguk, 2018-07-04:
Line-endings?

ZanderBrown, 2018-07-04:
@tjguk  that seems so simple and obvious it may just be the problem

dumbledad, 2018-07-23:
I had a quick try with SciTE to see if it was a Scintilla problem but I am not seeing the same error there.

tjguk, 2018-07-23:
Nice idea. Unfortunately the QScintilla control we're using is a complete re-implementation of the Scintilla control, not merely a wrapper!

dumbledad, 2018-07-23:
Ah. back to the drawing board. I'd like to find out the minimal way of deciding if it's a bug with mu or QScintilla (or NVDA)

tjguk, 2018-07-23:
If you haven't: could you try saving the file, please, and then trying again? Or: can you confirm whether the behaviour is at all different with:

* A new and unsaved file
* A new and saved file
* An existing file which has been opened

The reason for these questions is that we have some slightly fiddly line-ending code going on when saving or opening and since this looks a bit like a line-ending issue I'd like to know if there's any likely interaction.

dumbledad, 2018-07-23:
It's not you.

If I make a simple QScintilla editor in python using the example code [here](https://qscintilla.com/an-editor-in-a-gui/) I get exactly the same bug. I'll work out where to submit that and post a link here.

tjguk, 2018-07-23:
Ah. Good catch, then. Hopefully it'll be something Nice and Easy [tm] for the PyQt guys and they'll be able to post an updated wheel Real Soon [tm] for us to update to.

Thanks for working this through for us.

dumbledad, 2018-07-24:
I'm not entirely sure where to post QScintilla bug reports, so I've put it on their mailing list here: https://riverbankcomputing.com/pipermail/qscintilla/2018-July/001318.html

dumbledad, 2018-08-03:
I heard back from Phil Thompson at Riverbank Computing. He asked if I could try tonight's QScintilla snapshot. I don't think I am going to get time to attend to this soon, so if anyone has a working Mu build, NVDA, and access to the QScintilla snapshots please take a look. He suggested that the problem isrelatedto CR-LF line endings. and that if that is the case Mu could set the EOL mode to EOLUnix on all platforms.

devdanzin, 2020-12-26:
Cannot reproduce on current master and fresh installed NVDA. Also tried [a simple QScintilla editor example](https://qscintilla.com/#editor), no mistakes. But if I press the right arrow at the end of a line, NVDA will read "carriage return" or "line feed".

Not sure whether QScintilla changed, NVDA special-cases the previous issue, or my environment isn't representative of the ones where the problem shows up.