mu-editor/mu #524: Following the developer setup instructions results in Qt lib clashes

carlosperate, 2018-07-05:
Based on the instructions in https://mu.readthedocs.io/en/latest/setup.html

First we install the latest versions of all the dependencies by using a requirements.txt file without versions.

Then we `pip install --editable .` which reinstalls some of the packages already included in the requirements.txt file. Because some of these versions are locked with `==` and others are set to a minimun `>=` we ended up with an incompatible version of PyQtCharts:
```
PyQt5==5.10.1
PyQtChart==5.11.2
```

<details>
<summary>Ouput for pip freeze:</summary>
<pre>
alabaster==0.7.11
appdirs==1.4.3
appnope==0.1.0
asn1crypto==0.24.0
atomicwrites==1.1.5
attrs==18.1.0
Automat==0.7.0
Babel==2.6.0
backcall==0.1.0
beautifulsoup4==4.6.0
certifi==2018.4.16
cffi==1.11.5
chardet==3.0.4
constantly==15.1.0
coverage==4.5.1
cryptography==2.2.2
cssselect==1.0.3
cycler==0.10.0
decorator==4.3.0
docutils==0.14
gpiozero==1.4.1
guizero==0.5.2
hyperlink==18.0.0
idna==2.7
imagesize==1.0.0
incremental==17.5.0
ipykernel==4.8.2
ipython==6.4.0
ipython-genutils==0.2.0
jedi==0.12.1
Jinja2==2.10
jupyter-client==5.2.3
jupyter-core==4.4.0
kiwisolver==1.0.1
lxml==4.2.3
MarkupSafe==1.0
matplotlib==2.1.2
more-itertools==4.2.0
-e git+http://github.com/mu-editor/mu.git@150787bb22bcf47a5170a859adbd18e4741b0fcc#egg=mu_editor
nudatus==0.0.3
numpy==1.14.5
packaging==17.1
parsel==1.5.0
parso==0.3.0
pexpect==4.6.0
pgzero==1.2
pickleshare==0.7.4
pigpio==1.40.post1
Pillow==5.2.0
pkginfo==1.4.2
pluggy==0.6.0
prompt-toolkit==1.0.15
ptyprocess==0.6.0
py==1.5.4
pyasn1==0.4.3
pyasn1-modules==0.2.2
pycodestyle==2.3.1
pycparser==2.18
PyDispatcher==2.0.5
pyflakes==1.6.0
pygame==1.9.3
Pygments==2.2.0
pynsist==2.1
pyOpenSSL==18.0.0
pyparsing==2.2.0
PyQt5==5.10.1
PyQt5-sip==4.19.11
PyQtChart==5.11.2
pyserial==3.4
pytest==3.6.3
pytest-cov==2.5.1
pytest-random-order==0.8.0
python-dateutil==2.7.3
pytz==2018.5
pyzmq==17.0.0
QScintilla==2.10.7
qtconsole==4.3.1
queuelib==1.5.0
requests==2.19.1
requests-download==0.1.2
requests-toolbelt==0.8.0
Scrapy==1.5.0
service-identity==17.0.0
simplegeneric==0.8.1
sip==4.19.8
six==1.11.0
snowballstemmer==1.2.1
Sphinx==1.7.5
sphinxcontrib-websupport==1.1.0
tornado==5.0.2
tqdm==4.23.4
traitlets==4.3.2
twine==1.11.0
Twisted==18.4.0
urllib3==1.23
w3lib==1.19.0
wcwidth==0.1.7
win-cli-launchers==0.1
yarg==0.1.9
zope.interface==4.5.0
</pre>
</details>

And running Mu would cause the following output (here in case people search for it in the issue tracker):
```
$ python run.py 
Traceback (most recent call last):
  File "run.py", line 2, in <module>
    from mu.app import run
  File "/Users/microbit-carlos/workspace/mine/mu-test/mu/app.py", line 34, in <module>
    from mu.interface import Window
  File "/Users/microbit-carlos/workspace/mine/mu-test/mu/interface/__init__.py", line 1, in <module>
    from .main import Window
  File "/Users/microbit-carlos/workspace/mine/mu-test/mu/interface/main.py", line 38, in <module>
    from mu.interface.editor import EditorPane
  File "/Users/microbit-carlos/workspace/mine/mu-test/mu/interface/editor.py", line 25, in <module>
    from PyQt5.Qsci import QsciScintilla, QsciLexerPython, QsciAPIs
ImportError: dlopen(/Users/microbit-carlos/.local/share/virtualenvs/mu-test-vZhRV17F/lib/python3.6/site-packages/PyQt5/Qsci.so, 2): Library not loaded: @rpath/QtPrintSupport.framework/Versions/5/QtPrintSupport
  Referenced from: /Users/microbit-carlos/.local/share/virtualenvs/mu-test-vZhRV17F/lib/python3.6/site-packages/PyQt5/Qsci.so
  Reason: Incompatible library version: Qsci.so requires version 5.11.0 or later, but QtPrintSupport provides version 5.10.0
```

This can be fixed locally by either upgrading PyQt5 or downgrading PyQtCharts. Although with an upgraded PyQt5 it will complain about having the wrong version if running `mu-editor` instead of `python run.py`.

The fix in the repo would be to either lock the PyQtCharts version or change how we are setting the versions in requirements.txt vs setup.py (there was a discussion about this in the deb file issue as well).

tmontes, 2019-06-01:
I suppose that this is no longer true and the issue may be closed. Unless there's any detail that needs addressing: either in the packaging / dev setup tooling or associated docs. Anything missing?

carlosperate, 2019-06-01:
All Qt packages are locked and in a single place now, so we can close this one.

closed 2019-06-01 by carlosperate