mu-editor/mu #526: Closing panels by switching mode doesn't alter internal "on/off" state

carlosperate, 2018-07-06:
Steps to replicate:

* Open Mu
* Go to Python mode
* Open REPL
* Change mode (for example to micro:bit mode)
    * At this point REPL is closed
* Go back to Python mode
* Click on REPL button
    * Nothing happens as the internal state of Mu thinks the REPL pane is still open and now it tried to close it
* Click on REPL button
    * Now it opens

I have chosen the steps above to demostrate the issue because it doesn't require any device attached to the computer, but this is more obvious on modes like the micro:bit where you can have the Files pane open, change mode, change back, and at that point it won't let you open the REPL and instead show you an alert with the message: "REPL and file system cannot work at the same time."

Large gif incoming:

[image]

ntoll, 2018-07-07:
Just looked into this, it's because the interface layer is changing without it going through the logic layer (which keeps track of state AND the interface layer). I guess this was missed in some refactoring somewhere. In any case, it should be simple to fix (coming in a moment).

ntoll, 2018-07-07:
Please check with the latest commit. I've created #530 to remind me to refactor/simplify the whole made-changing mechanism in 1.1.

closed 2018-07-07 by ntoll

carlosperate, 2018-07-08:
Thanks Nicholas!

commit 6c3a4fac (2018-07-07): Ensure the old mode is reset as expected upon change of mode. Fixes #526.