mu-editor/mu #528: Update to Python 3.7

ntoll, 2018-07-06:
For this to be successful we need to ensure there are binary wheels for all the required compiled packages upon which Mu depends (for example, PyGame).

SpotlightKid, 2018-08-23:
Please note that there is currently [a bug](https://bugs.python.org/issue31014) in the `webbrowser` module for Python 3.7, which makes Mu crash when the help button is used and the user has the `BROWSER` environment variable set to a custom program.

In the AUR package for mu-editor, I apply a patch with this workaround:

```
--- mu/logic.py.orig	2018-08-14 22:14:04.663531208 +0200
+++ mu/logic.py	2018-08-14 22:14:25.616864697 +0200
@@ -988,6 +988,10 @@
         major_version = '.'.join(__version__.split('.')[:2])
         url = 'https://codewith.mu/{}/help/{}'.format(language_code,
                                                       major_version)
+        # XXX: hacky workaround for https://bugs.python.org/issue31014
+        if 'BROWSER' in os.environ:
+            del os.environ['BROWSER']
+
         webbrowser.open_new(url)
 
     def quit(self, *args, **kwargs):
```

tjguk, 2018-08-28:
@SpotlightKid thanks for the heads-up. I see that the [fix has been applied upstream](https://github.com/python/cpython/commit/a410f9f614b62cd7df220186d081ffd73786be91) so hopefully by the time we do come to build from 3.7, we should be picking up the corrected version. 

I'd be reluctant to apply your workaround as is since it silently ignores the user's choice of browser. (And I can't think of easy ways to detect whether or not it would be needed).

SpotlightKid, 2018-08-28:
Well, the alternative is that Mu crashes if somebody runs it under Python 3.7.0, has `BROWSER` set and uses the help function. That version may still be present on many systems even after a fixed version is released. A version check could be inserted so that `BROWSER` is only ignored with Python 3.7.0. Unfortunately the `webbrowser` module does not expose a version number of its own.

carlosperate, 2018-09-17:
@SpotlightKid Does Mu crash because of an unhandled exception? Or because Python crashes?

SpotlightKid, 2018-09-17:
Unhanled exception. Just set the `BROWSER` environment variable to e.g. `xdg-open` before starting Mu to test it.

carlosperate, 2018-09-17:
I think that rather than adding a very specific fix specific to Python 3.7 we could catch the exception and show the URL to the user for them to open in whatever browser they chose.

SpotlightKid, 2018-09-17:
Here's a better workaround:

```
import os
import webbrowser

env_browser = os.environ.get('BROWSER')
if env_browser:
    del os.environ['BROWSER']

webbrowser.register_standard_browsers()
if env_browser not in webbrowser._browsers:
    webbrowser.register(env_browser, None, webbrowser.GenericBrowser(env_browser), preferred=True)

# optional
if env_browser:
    os.environ['BROWSER'] = env_browser

webbrowser.open('http://example.com/')
```

This works, because the actual issue is triggered when the browser set in the `BROWSER` env variable is already found during `webbrowser.register_standard_browsers()`, which gets called by `webbrowser.open()` when first used.

This workaround uses an internal implementation detail of `webbrowser`, though, the private module global `_browsers`, so this fix may fail in future versions of `webbrowser`, so it should probably still be guarded by a version check.

SpotlightKid, 2018-09-28:
The issue is fixed in Python 3.7.1rc1: https://docs.python.org/3.7/whatsnew/changelog.html#python-3-7-1-release-candidate-1 (but installations on Python 3.7.0 will remain susceptible).

dybber, 2020-09-29:
I guess this was fixed a long time ago, at least Travis now builds on 3.7 and I'm successfully running Mu on 3.8 as well. Closing.

closed 2020-09-29 by dybber