mu-editor/mu #529: Feature suggestion: Block comment in/out on right click menu

carlosperate, 2018-07-06:
We've recently had the handy new feature to block comment in/out code (https://github.com/mu-editor/mu/issues/516 thanks Nicholas!).

Would it be possible to add an entry on the editor right click menu to do the same? It would make it a lot more discoverable.

ntoll, 2018-07-07:
This can be a 1.1 feature if more people ask for it. (Several people asked for the commenting toggle as a hot-key, which is why I added it.)

ZanderBrown, 2018-07-07:
Not sure a menu item would be good, perhaps it makes block comments 'too magic'?

ntoll, 2018-07-07:
TBH, on reflection, I think (like find/replace) this is an "advanced" feature that should only be accessed via a hot-key.

carlosperate, 2018-07-08:
> Not sure a menu item would be good, perhaps it makes block comments 'too magic'?

Not sure I understand what you mean. How would an entry on the right click drop down menu make the feature more magical?

The search/replace is a fairly common shortcut that most people use as muscle memory, on the other hand a lot of editors use a different shortcut for toggling comments. Because of this, at least to me, this feature feels very hidden. I would probably try `Ctrl + /`, very unlikely I would `Ctrl + K`, but I don't use VS all that much these days.

However, I do agree that perhaps the feature should only be added if more users request it.

dybber, 2020-09-29:
I agree with @ntoll. Some features don't need to be discoverable, and for me it's okay that this is one of these things you discover later on, rather than to put too many things into right-click menus. 

It's not that I find it very advanced, but everytime you add something like this, it can add a tiny bit of complexity, which might limit the amount of cognitive capacity that the students can put into the project they are working on. It can cause distraction, questions, etc.

carlosperate, 2020-09-29:
I still think the feature is incredibly opaque if you don't happen to know the shortcut, and as the assigned keys is not a standard between editors, knowing it will  depend on what editor you have used in the past (and we assume this is the first editor for many of our users) .
Nevertheless it's been 3 years and there hasn't been any other requests, so we can close this ticket :)

closed 2020-09-29 by carlosperate