mu-editor/mu #530: Refactor mode chaging in 1.1 so mode instances present a common interface for cleaning up their state

ntoll, 2018-07-07:
It's a potential mess in logic.py.

dybber, 2020-11-23:
Just to make sure I understand this issue. What you suggest is that all modes have a sort of "tear_down" function, which when called will e.g. close REPL's, plotter's etc. which the mode has opened? That does make sense.

ntoll, 2020-12-06:
Yes.... Sorry. @dybber you've understood correctly. My "mess" comment cam about because logic.py, while starting out simply enough, has "evolved" into something more complicated than it was at first imagined (welcome to software engineering!) and may, as a result, cause problems.

In which case, as @tjguk has done with the `venv` related work, please feel free to refactor, abstract and reorganise the code as you see fit so it's simpler, makes clear sense, and is easier to use and/or change. :+1:

dybber, 2020-12-06:
Okay, thanks, so this change is not to support any specific new functionality, but to make it more modular. I guess the end goal is still to make modes installable as plugins?

dybber, 2020-12-07:
I think a refactor is a good idea, and I would like to help with this, as that's something I believe I'm good at :-)  However, I think we should wait doing large rewrites such as this until the work on virtual environments have been merged #1072.

carlosperate, 2020-12-07:
In that case maybe we should avoid major refactors until 1.1 is ready? Otherwise we'll never get it out

dybber, 2020-12-07:
I agree on the move to the 1.2 milestone, to keep down the complexity of getting to a 1.1 release.