mu-editor/mu #531: Debugger: Exception in the debugger stop execution

carlosperate, 2018-07-08:
When the debugger is running and a exception is thrown in the user code, it is captured in the log, but the debugger stops and the UI doesn't show any indication of what happened.

Heavy gif below:

[image]

<details>
<summary>Log</summary>
<pre>
-----------------

Starting Mu 1.0.0.beta.16
2018-07-08 17:22:48,141 - root:112(run) INFO: uname_result(system='Darwin', node='Carlos-MBP.local', release='17.6.0', version='Darwin Kernel Version 17.6.0: Tue May  8 15:22:16 PDT 2018; root:xnu-4570.61.1~1/RELEASE_X86_64', machine='x86_64', processor='i386')
2018-07-08 17:22:48,141 - root:113(run) INFO: Python path: ['/Users/microbit-carlos/workspace/mine/mu-test', '/Users/microbit-carlos/.local/share/virtualenvs/mu-test-vZhRV17F/lib/python36.zip', '/Users/microbit-carlos/.local/share/virtualenvs/mu-test-vZhRV17F/lib/python3.6', '/Users/microbit-carlos/.local/share/virtualenvs/mu-test-vZhRV17F/lib/python3.6/lib-dynload', '/usr/local/Cellar/python3/3.6.4_2/Frameworks/Python.framework/Versions/3.6/lib/python3.6', '/Users/microbit-carlos/.local/share/virtualenvs/mu-test-vZhRV17F/lib/python3.6/site-packages', '/Users/microbit-carlos/workspace/mine/mu-test', '/Users/microbit-carlos/.local/share/virtualenvs/mu-test-vZhRV17F/lib/python3.6/site-packages/IPython/extensions']
2018-07-08 17:22:48,437 - mu.logic:543(__init__) INFO: Setting up editor.
2018-07-08 17:22:48,438 - mu.logic:560(__init__) INFO: Settings path: /Users/microbit-carlos/Library/Application Support/mu/settings.json
2018-07-08 17:22:48,438 - mu.logic:561(__init__) INFO: Session path: /Users/microbit-carlos/Library/Application Support/mu/session.json
2018-07-08 17:22:48,438 - mu.logic:562(__init__) INFO: Log directory: /Users/microbit-carlos/Library/Logs/mu
2018-07-08 17:22:48,438 - mu.logic:563(__init__) INFO: Data directory: /Users/microbit-carlos/Library/Application Support/mu
2018-07-08 17:22:48,462 - mu.logic:576(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2018-07-08 17:22:48,588 - mu.logic:1116(change_mode) INFO: Workspace directory: /Users/microbit-carlos/mu_code
2018-07-08 17:22:48,590 - mu.logic:630(restore_session) INFO: Restoring session from: /Users/microbit-carlos/Library/Application Support/mu/session.json
2018-07-08 17:22:48,590 - mu.logic:631(restore_session) DEBUG: {'theme': 'day', 'mode': 'python', 'paths': ['/Users/microbit-carlos/mu_code/microbit_example.py', '/Users/microbit-carlos/mu_code/debug_test.py'], 'envars': [], 'minify': False, 'microbit_runtime': ''}
2018-07-08 17:22:48,590 - mu.logic:713(_load) INFO: Loading script from: /Users/microbit-carlos/mu_code/microbit_example.py
2018-07-08 17:22:48,592 - mu.logic:272(read_and_decode) DEBUG: Trying to decode with utf-8
2018-07-08 17:22:48,594 - mu.logic:275(read_and_decode) INFO: Decoded with utf-8
2018-07-08 17:22:48,594 - mu.logic:288(read_and_decode) DEBUG: Detected newline '\n'
2018-07-08 17:22:48,594 - mu.logic:787(_load) DEBUG: from microbit import display
display.scroll('BB')
2018-07-08 17:22:48,667 - mu.logic:713(_load) INFO: Loading script from: /Users/microbit-carlos/mu_code/debug_test.py
2018-07-08 17:22:48,668 - mu.logic:272(read_and_decode) DEBUG: Trying to decode with utf-8
2018-07-08 17:22:48,668 - mu.logic:275(read_and_decode) INFO: Decoded with utf-8
2018-07-08 17:22:48,668 - mu.logic:288(read_and_decode) DEBUG: Detected newline '\n'
2018-07-08 17:22:48,668 - mu.logic:787(_load) DEBUG: print('test')
a = 1/0
2018-07-08 17:22:48,682 - mu.logic:652(restore_session) INFO: Loaded files.
2018-07-08 17:22:48,682 - mu.logic:656(restore_session) INFO: User defined environment variables: []
2018-07-08 17:22:48,682 - mu.logic:660(restore_session) INFO: Minify scripts on micro:bit? False
2018-07-08 17:22:48,695 - mu.logic:1116(change_mode) INFO: Workspace directory: /Users/microbit-carlos/mu_code
2018-07-08 17:22:58,711 - mu.logic:1138(autosave) INFO: Autosave has detected changes.
2018-07-08 17:22:58,711 - mu.logic:847(save_tab_to_file) INFO: Saving script to: /Users/microbit-carlos/mu_code/debug_test.py
2018-07-08 17:22:58,711 - mu.logic:848(save_tab_to_file) DEBUG: print('test')
a = 1/0
print('end')

2018-07-08 17:23:00,997 - mu.logic:847(save_tab_to_file) INFO: Saving script to: /Users/microbit-carlos/mu_code/debug_test.py
2018-07-08 17:23:01,000 - mu.logic:848(save_tab_to_file) DEBUG: print('test')
a = 1/0
print('end')

2018-07-08 17:23:46,304 - mu.modes.python3:219(debug) INFO: Starting debug mode.
2018-07-08 17:23:46,330 - mu.logic:1116(change_mode) INFO: Workspace directory: /Users/microbit-carlos/mu_code
2018-07-08 17:23:46,337 - mu.modes.debugger:113(start) DEBUG: print('test')
a = 1/0
print('end')

2018-07-08 17:23:46,344 - mu.interface.panes:662(start_process) INFO: Running script: /Users/microbit-carlos/mu_code/debug_test.py
2018-07-08 17:23:46,344 - mu.interface.panes:667(start_process) INFO: Command args: []
2018-07-08 17:23:46,345 - mu.interface.panes:685(start_process) INFO: Working directory: /Users/microbit-carlos/mu_code
2018-07-08 17:23:46,345 - mu.interface.panes:690(start_process) INFO: Python path: ['/Users/microbit-carlos/workspace/mine/mu-test', '/Users/microbit-carlos/.local/share/virtualenvs/mu-test-vZhRV17F/lib/python36.zip', '/Users/microbit-carlos/.local/share/virtualenvs/mu-test-vZhRV17F/lib/python3.6', '/Users/microbit-carlos/.local/share/virtualenvs/mu-test-vZhRV17F/lib/python3.6/lib-dynload', '/usr/local/Cellar/python3/3.6.4_2/Frameworks/Python.framework/Versions/3.6/lib/python3.6', '/Users/microbit-carlos/.local/share/virtualenvs/mu-test-vZhRV17F/lib/python3.6/site-packages', '/Users/microbit-carlos/workspace/mine/mu-test', '/Users/microbit-carlos/.local/share/virtualenvs/mu-test-vZhRV17F/lib/python3.6/site-packages/IPython/extensions']
2018-07-08 17:23:47,606 - mu.debugger.client:147(worker) DEBUG: ["bootstrap", {"breakpoints": []}]
2018-07-08 17:23:47,607 - mu.debugger.client:147(worker) DEBUG: ["stack", {"stack": []}]
2018-07-08 17:23:47,609 - mu.debugger.client:147(worker) DEBUG: ["call", {"args": null}]
2018-07-08 17:23:47,610 - mu.debugger.client:147(worker) DEBUG: ["stack", {"stack": []}]
2018-07-08 17:23:47,611 - mu.debugger.client:147(worker) DEBUG: ["line", {"filename": "/Users/microbit-carlos/mu_code/debug_test.py", "line": 1}]
2018-07-08 17:23:47,618 - mu.debugger.client:147(worker) DEBUG: ["stack", {"stack": []}]
2018-07-08 17:23:49,127 - mu.debugger.client:147(worker) DEBUG: ["line", {"filename": "/Users/microbit-carlos/mu_code/debug_test.py", "line": 2}]
2018-07-08 17:23:49,127 - mu.debugger.client:147(worker) DEBUG: ["stack", {"stack": []}]
2018-07-08 17:23:50,266 - mu.debugger.client:147(worker) DEBUG: ["exception", {"name": "ZeroDivisionError", "value": "ZeroDivisionError('division by zero',)"}]
2018-07-08 17:23:50,267 - mu.debugger.client:147(worker) DEBUG: ["stack", {"stack": []}]
2018-07-08 17:23:55,853 - mu.logic:1044(show_admin) INFO: Showing logs from /Users/microbit-carlos/Library/Logs/mu/mu.log
</pre>
</details>

ntoll, 2018-07-08:
OK... the exception will actually show if you keep pressing the "step over" button, which suggests to me a bunch of messages are stuck in a buffer somewhere. This'll need to be flushed so the exception message gets through without having to keep kicking whatever it is that's reading the message queue (my guess). I'll investigate and fix now. :-)

Good catch @carlosperate!

ntoll, 2018-07-08:
It turns out I never implemented anything to handle user-generated exceptions. :-D

In that situation, the debugger would just sit there and keep reacting to button presses until the script finally finished and the error message was emitted to stdout.

My solution in this situation is to log the error message and make the script continue so the error message and other helpful output is emitted to stdout as a helpful message. The debugger will thus be in a valid end state for the user to stop it and update their code.

closed 2018-07-08 by ntoll

carlosperate, 2018-07-09:
Thanks Nicholas!