mu-editor/mu #536: pgzero window does not get focus and can't receive keyboard events.

nanoflite, 2018-07-10:
I'm trying to receive keyboard events, but the pgzero window does not get focus. Clicking on the window also does not give it focus, so no keyboard events are received.

I'm currently using 1.0.0.beta.17. The same bug is present in beta.16 as well.
I'm on OSX.

```
2018-07-10 18:56:58,517 - mu.modes.pygamezero:134(run_game) DEBUG: def update():
    if keyboard[keys.SPACE]:
        print("SPACE")
2018-07-10 18:56:58,517 - mu.modes.base:71(get_default_workspace) ERROR: Workspace value in the settings file is not a validdirectory: /Volumes/CIRCUITPY
2018-07-10 18:56:58,521 - mu.interface.panes:662(start_process) INFO: Running script: /Users/johan/Projects/pygame_shooter/keyboard.py
2018-07-10 18:56:58,521 - mu.interface.panes:667(start_process) INFO: Command args: []
2018-07-10 18:56:58,521 - mu.interface.panes:685(start_process) INFO: Working directory: /Users/johan/mu_code
2018-07-10 18:56:58,521 - mu.interface.panes:690(start_process) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2018-07-10 18:58:54,968 - mu.logic:1045(show_admin) INFO: Showing logs from /Users/johan/Library/Logs/mu/mu.log
```

ntoll, 2018-07-10:
Thank you for reporting this.

Are you running Mu within a virtualenv or with Conda..? If so, I believe you may have run into this problem: https://github.com/pygame/pygame/issues/203

I've just checked on my Mac with the version of Mu installed via the "official" download (see: https://codewith.mu/en/howto/install_macos) and I can focus the PyGame window and it receives mouse events. However, I notice that the stdin/stdout pane has keyboard focus so those sorts of event are not going to the right place (you've found a bug!).

I'll make sure this is fixed as a matter of urgency.

ntoll, 2018-07-10:
Perhaps @lordmauve can throw some light on this if it's the PyGame focus bug..?

(Note, the official install of Mu for OSX comes with its own version of Python 3.6 embedded within.)

lordmauve, 2018-07-10:
This is to do with "Framework" Vs non Framework Python. I don't know what that means exactly, but we used to have code in Pygame Zero that re-execs the process using a Framework Python if it find that it is a non-Framework Python. However, I think we make this fail silently if it couldn't find the correct path, which may be happening.

ntoll, 2018-07-10:
OK... I've been doing some investigations and tests... it looks like you've run into this problem in PyGame: https://github.com/pygame/pygame/issues/203

As you'll see, the problem is fixed in the upcoming 1.9.4 release of PyGame, but I'm not sure if this will happen in time for next week's 1.0 release. As it happens, @illume (PyGame's maintainer) is a buddy of mine and I've emailed him to try to get a sense of his (volunteer led) time-lines. We may get lucky and see a fixed version of PyGame released before Mu, but my expectation is we'll just have to wait and I'll update Mu's documentation with (I hope) suggestions for work-arounds (there are a few non-technical suggestions for getting the PyGame Window into focus in the PyGame GitHub issue).

Let's see what happens.

ntoll, 2018-07-10:
@lordmauve yeah... that's my conclusion too (hard to tell what's happening). As @illume says in the related PyGame issue (see my answer to @nanoflite), it should be fixed in 1.9.4.

carlosperate, 2018-07-11:
I can confirm I see the same issue in macOS running from a normal python installation and virtualenv AND from the official dmg release.
Is there any info from my system that I can provide to check if I am running anything odd? (I've never used conda, but don't know if that and venv are the only situations where this happens).

ntoll, 2018-07-11:
@nanoflite I've just heard form Rene (PyGame maintainer)... a new release with the fix for this problem will be out by the weekend. :-)

nanoflite, 2018-07-11:
That's good news!

carlosperate, 2018-07-16:
In case v1.9.4 doesn't make it in time, is it worth trying 1.9.4.dev0?

Since pygame zero takes any version >= 1.9.2 (https://github.com/lordmauve/pgzero/blob/master/setup.py#L11) if we add it as a dependency in Mu directly it should be able to use that version.

ntoll, 2018-07-16:
I'm just about to test this *right now*. @illume is moving mountains to get this out in time. I hope it all lines up in time. If not, for *just* the packaged Mac version we should use the latest dev version for 1.0 so the focus problem is no longer an issue.

ntoll, 2018-07-19:
FYI PyGame 1.9.4 is out and will make it into the final release of Mu. I tested 1.9.4 on OSX and the focus issue no longer occurs.

closed 2018-07-19 by ntoll