mu-editor/mu #537: Flash doesn't work in 1.0.0-beta.17

bosnivan, 2018-07-10:
Hi!

Firstly, I must thank you for this nice editor. :)

After package upgrade to 1.0.0-beta.17 I can't flash hex file to micro:bit. Before upgrade everything worked as expected. I'm using WinPython-64bit-3.6.2.0Qt5. Here's log file:
2018-07-10 21:49:42,315 - root:111(run) INFO: 

-----------------

Starting Mu 1.0.0.beta.17
2018-07-10 21:49:42,316 - root:112(run) INFO: uname_result(system='Windows', node='WMAR1785', release='10', version='10.0.14393', machine='AMD64', processor='Intel64 Family 6 Model 78 Stepping 3, GenuineIntel')
2018-07-10 21:49:42,316 - root:113(run) INFO: Python path: ['C:\\Moje\\Programiranje\\WinPython-64bit-3.6.2.0Qt5\\python-3.6.2.amd64\\Scripts\\mu-editor.exe', 'c:\\moje\\programiranje\\winpython-64bit-3.6.2.0qt5\\python-3.6.2.amd64\\python36.zip', 'c:\\moje\\programiranje\\winpython-64bit-3.6.2.0qt5\\python-3.6.2.amd64\\DLLs', 'c:\\moje\\programiranje\\winpython-64bit-3.6.2.0qt5\\python-3.6.2.amd64\\lib', 'c:\\moje\\programiranje\\winpython-64bit-3.6.2.0qt5\\python-3.6.2.amd64', 'c:\\moje\\programiranje\\winpython-64bit-3.6.2.0qt5\\python-3.6.2.amd64\\lib\\site-packages', 'c:\\moje\\programiranje\\winpython-64bit-3.6.2.0qt5\\python-3.6.2.amd64\\lib\\site-packages\\win32', 'c:\\moje\\programiranje\\winpython-64bit-3.6.2.0qt5\\python-3.6.2.amd64\\lib\\site-packages\\win32\\lib', 'c:\\moje\\programiranje\\winpython-64bit-3.6.2.0qt5\\python-3.6.2.amd64\\lib\\site-packages\\Pythonwin', 'c:\\moje\\programiranje\\winpython-64bit-3.6.2.0qt5\\python-3.6.2.amd64\\lib\\site-packages\\IPython\\extensions']
2018-07-10 21:49:42,457 - mu.logic:543(__init__) INFO: Setting up editor.
2018-07-10 21:49:42,457 - mu.logic:561(__init__) INFO: Settings path: C:\Users\ibosnic\AppData\Local\python\mu\settings.json
2018-07-10 21:49:42,458 - mu.logic:562(__init__) INFO: Session path: C:\Users\ibosnic\AppData\Local\python\mu\session.json
2018-07-10 21:49:42,458 - mu.logic:563(__init__) INFO: Log directory: C:\Users\ibosnic\AppData\Local\python\mu\Logs
2018-07-10 21:49:42,458 - mu.logic:564(__init__) INFO: Data directory: C:\Users\ibosnic\AppData\Local\python\mu
2018-07-10 21:49:42,599 - mu.logic:577(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2018-07-10 21:49:43,089 - mu.logic:1119(change_mode) INFO: Workspace directory: C:\Users\ibosnic\mu_code
2018-07-10 21:49:43,089 - mu.logic:631(restore_session) INFO: Restoring session from: C:\Users\ibosnic\AppData\Local\python\mu\session.json
2018-07-10 21:49:43,089 - mu.logic:632(restore_session) DEBUG: {'theme': 'day', 'mode': 'microbit', 'paths': ['C:\\Moje\\Radovi\\mu_code\\TEST.py'], 'envars': [], 'minify': True, 'microbit_runtime': ''}
2018-07-10 21:49:43,090 - mu.logic:714(_load) INFO: Loading script from: C:\Moje\Radovi\mu_code\TEST.py
2018-07-10 21:49:43,091 - mu.logic:272(read_and_decode) DEBUG: Trying to decode with utf-8
2018-07-10 21:49:43,091 - mu.logic:275(read_and_decode) INFO: Decoded with utf-8
2018-07-10 21:49:43,092 - mu.logic:288(read_and_decode) DEBUG: Detected newline '\r\n'
2018-07-10 21:49:43,092 - mu.logic:788(_load) DEBUG: from microbit import *
2018-07-10 21:49:43,491 - mu.logic:653(restore_session) INFO: Loaded files.
2018-07-10 21:49:43,492 - mu.logic:657(restore_session) INFO: User defined environment variables: []
2018-07-10 21:49:43,492 - mu.logic:661(restore_session) INFO: Minify scripts on micro:bit? True
2018-07-10 21:49:43,516 - mu.logic:1119(change_mode) INFO: Workspace directory: C:\Users\ibosnic\mu_code
2018-07-10 21:49:49,362 - mu.modes.microbit:237(flash) INFO: Preparing to flash script.
2018-07-10 21:49:49,362 - mu.modes.microbit:247(flash) DEBUG: Python script:
2018-07-10 21:49:49,362 - mu.modes.microbit:248(flash) DEBUG: b'from microbit import *'
2018-07-10 21:49:49,368 - mu.modes.microbit:296(flash) INFO: Path to micro:bit: D:\
2018-07-10 21:49:49,368 - mu.modes.microbit:297(flash) INFO: Serial port: COM4
2018-07-10 21:49:49,369 - mu.modes.microbit:298(flash) INFO: Device serial number: 6
2018-07-10 21:49:49,369 - mu.modes.microbit:317(flash) INFO: Checking target device.
2018-07-10 21:49:49,878 - mu.modes.microbit:321(flash) INFO: {'sysname': 'microbit', 'nodename': 'microbit', 'release': '1.0.0-beta.1', 'version': 'micro:bit v1.0.0-beta.1+8f503af on 2018-7-9; MicroPython v1.9.2-34-gd64154c73 on 2017-09-01', 'machine': 'micro:bit with nRF51822'}
2018-07-10 21:49:49,878 - mu.modes.microbit:340(flash) INFO: Board MicroPython: 1.0.0-beta.1
2018-07-10 21:49:49,878 - mu.modes.microbit:342(flash) INFO: Mu MicroPython: 1.0.0-beta.1
2018-07-10 21:49:49,879 - mu.modes.microbit:450(copy_main) INFO: Copying main.py onto device
2018-07-10 21:49:49,879 - mu.modes.microbit:460(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', "f(b'from microbit import *')", 'fd.close()']
2018-07-10 21:49:50,221 - mu.modes.microbit:463(copy_main) INFO: (b'', b'')
2018-07-10 21:49:54,534 - mu.logic:1045(show_admin) INFO: Showing logs from C:\Users\ibosnic\AppData\Local\python\mu\Logs\mu.log
2018-07-10 21:50:19,575 - mu.modes.microbit:237(flash) INFO: Preparing to flash script.
2018-07-10 21:50:19,575 - mu.modes.microbit:247(flash) DEBUG: Python script:
2018-07-10 21:50:19,575 - mu.modes.microbit:248(flash) DEBUG: b'from microbit import *'
2018-07-10 21:50:19,581 - mu.modes.microbit:296(flash) INFO: Path to micro:bit: D:\
2018-07-10 21:50:19,581 - mu.modes.microbit:297(flash) INFO: Serial port: COM4
2018-07-10 21:50:19,581 - mu.modes.microbit:298(flash) INFO: Device serial number: 6
2018-07-10 21:50:19,581 - mu.modes.microbit:317(flash) INFO: Checking target device.
2018-07-10 21:50:20,087 - mu.modes.microbit:321(flash) INFO: {'sysname': 'microbit', 'nodename': 'microbit', 'release': '1.0.0-beta.1', 'version': 'micro:bit v1.0.0-beta.1+8f503af on 2018-7-9; MicroPython v1.9.2-34-gd64154c73 on 2017-09-01', 'machine': 'micro:bit with nRF51822'}
2018-07-10 21:50:20,088 - mu.modes.microbit:340(flash) INFO: Board MicroPython: 1.0.0-beta.1
2018-07-10 21:50:20,088 - mu.modes.microbit:342(flash) INFO: Mu MicroPython: 1.0.0-beta.1
2018-07-10 21:50:20,088 - mu.modes.microbit:450(copy_main) INFO: Copying main.py onto device
2018-07-10 21:50:20,089 - mu.modes.microbit:460(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', "f(b'from microbit import *')", 'fd.close()']
2018-07-10 21:50:20,438 - mu.modes.microbit:463(copy_main) INFO: (b'', b'')
2018-07-10 21:50:24,955 - mu.logic:1045(show_admin) INFO: Showing logs from C:\Users\ibosnic\AppData\Local\python\mu\Logs\mu.log

ntoll, 2018-07-10:
Thank you for your feedback and kind words.

Hmm... what exactly is the problem? I mean, why do you think it has failed. 

By the looks of the log, your code is:

```
from microbit import *
```

Which will show no side-effects.

The log also shows that flashing succeeded (the penultimate line of the log, which shows `INFO: (b'', b'')` tells me that there was no output from the device (the first, empty `b''`), nor error message either (the second, empty `b''`).

Perhaps try flashing something like this:

```
from microbit import *
display.scroll("It works")
```

Please note, the new flashing method will usually complete in less than a second. If you're used to the older method (which could take 10 seconds) the new method may be over seemingly too quickly.

Please do let me know how you get on and I'm more than happy to answer any questions you may have.

bosnivan, 2018-07-10:
Hi!

Sorry for bad report. I expected longer flashing. :)
I found out that problem was display.off() in first hex I flashed. Here's better example:
1) First hex prints ABC, second hex prints XYZ. It works as expected.
2) First hex has display.off(), second hex prints XYZ. When I flash second hex it shows nothing. Only After I reset micro:bit, XYZ is shown. Maybe that's normal behaviour?
Log file:
2018-07-10 22:38:28,715 - root:111(run) INFO: 

-----------------

Starting Mu 1.0.0.beta.17
2018-07-10 22:38:28,715 - root:112(run) INFO: uname_result(system='Windows', node='WMAR1785', release='10', version='10.0.14393', machine='AMD64', processor='Intel64 Family 6 Model 78 Stepping 3, GenuineIntel')
2018-07-10 22:38:28,715 - root:113(run) INFO: Python path: ['C:\\Moje\\Programiranje\\WinPython-64bit-3.6.2.0Qt5\\python-3.6.2.amd64\\Scripts\\mu-editor.exe', 'c:\\moje\\programiranje\\winpython-64bit-3.6.2.0qt5\\python-3.6.2.amd64\\python36.zip', 'c:\\moje\\programiranje\\winpython-64bit-3.6.2.0qt5\\python-3.6.2.amd64\\DLLs', 'c:\\moje\\programiranje\\winpython-64bit-3.6.2.0qt5\\python-3.6.2.amd64\\lib', 'c:\\moje\\programiranje\\winpython-64bit-3.6.2.0qt5\\python-3.6.2.amd64', 'c:\\moje\\programiranje\\winpython-64bit-3.6.2.0qt5\\python-3.6.2.amd64\\lib\\site-packages', 'c:\\moje\\programiranje\\winpython-64bit-3.6.2.0qt5\\python-3.6.2.amd64\\lib\\site-packages\\win32', 'c:\\moje\\programiranje\\winpython-64bit-3.6.2.0qt5\\python-3.6.2.amd64\\lib\\site-packages\\win32\\lib', 'c:\\moje\\programiranje\\winpython-64bit-3.6.2.0qt5\\python-3.6.2.amd64\\lib\\site-packages\\Pythonwin', 'c:\\moje\\programiranje\\winpython-64bit-3.6.2.0qt5\\python-3.6.2.amd64\\lib\\site-packages\\IPython\\extensions']
2018-07-10 22:38:28,835 - mu.logic:543(__init__) INFO: Setting up editor.
2018-07-10 22:38:28,836 - mu.logic:561(__init__) INFO: Settings path: C:\Users\ibosnic\AppData\Local\python\mu\settings.json
2018-07-10 22:38:28,836 - mu.logic:562(__init__) INFO: Session path: C:\Users\ibosnic\AppData\Local\python\mu\session.json
2018-07-10 22:38:28,836 - mu.logic:563(__init__) INFO: Log directory: C:\Users\ibosnic\AppData\Local\python\mu\Logs
2018-07-10 22:38:28,836 - mu.logic:564(__init__) INFO: Data directory: C:\Users\ibosnic\AppData\Local\python\mu
2018-07-10 22:38:28,966 - mu.logic:577(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2018-07-10 22:38:29,474 - mu.logic:1119(change_mode) INFO: Workspace directory: C:\Users\ibosnic\mu_code
2018-07-10 22:38:29,475 - mu.logic:631(restore_session) INFO: Restoring session from: C:\Users\ibosnic\AppData\Local\python\mu\session.json
2018-07-10 22:38:29,475 - mu.logic:632(restore_session) DEBUG: {'theme': 'day', 'mode': 'microbit', 'paths': ['C:\\Moje\\Radovi\\mu_code\\TEST.py', 'C:\\Moje\\Radovi\\mu_code\\REPL.py'], 'envars': [], 'minify': True, 'microbit_runtime': ''}
2018-07-10 22:38:29,475 - mu.logic:714(_load) INFO: Loading script from: C:\Moje\Radovi\mu_code\TEST.py
2018-07-10 22:38:29,476 - mu.logic:272(read_and_decode) DEBUG: Trying to decode with utf-8
2018-07-10 22:38:29,477 - mu.logic:275(read_and_decode) INFO: Decoded with utf-8
2018-07-10 22:38:29,478 - mu.logic:288(read_and_decode) DEBUG: Detected newline '\r\n'
2018-07-10 22:38:29,478 - mu.logic:788(_load) DEBUG: from microbit import *

display.scroll("XYZ")
2018-07-10 22:38:29,850 - mu.logic:714(_load) INFO: Loading script from: C:\Moje\Radovi\mu_code\REPL.py
2018-07-10 22:38:29,851 - mu.logic:272(read_and_decode) DEBUG: Trying to decode with utf-8
2018-07-10 22:38:29,851 - mu.logic:275(read_and_decode) INFO: Decoded with utf-8
2018-07-10 22:38:29,851 - mu.logic:288(read_and_decode) DEBUG: Detected newline '\r\n'
2018-07-10 22:38:29,851 - mu.logic:788(_load) DEBUG: from microbit import *

display.off()
2018-07-10 22:38:29,861 - mu.logic:653(restore_session) INFO: Loaded files.
2018-07-10 22:38:29,862 - mu.logic:657(restore_session) INFO: User defined environment variables: []
2018-07-10 22:38:29,862 - mu.logic:661(restore_session) INFO: Minify scripts on micro:bit? True
2018-07-10 22:38:29,884 - mu.logic:1119(change_mode) INFO: Workspace directory: C:\Users\ibosnic\mu_code
2018-07-10 22:38:39,886 - mu.logic:848(save_tab_to_file) INFO: Saving script to: C:\Moje\Radovi\mu_code\REPL.py
2018-07-10 22:38:39,887 - mu.logic:849(save_tab_to_file) DEBUG: from microbit import *

display.scroll("X")
2018-07-10 22:38:39,909 - mu.logic:1145(autosave) INFO: Autosave detected and saved changes in C:\Moje\Radovi\mu_code\REPL.py.
2018-07-10 22:38:42,512 - mu.logic:848(save_tab_to_file) INFO: Saving script to: C:\Moje\Radovi\mu_code\REPL.py
2018-07-10 22:38:42,513 - mu.logic:849(save_tab_to_file) DEBUG: from microbit import *

display.scroll("ABC")
2018-07-10 22:38:43,869 - mu.modes.microbit:237(flash) INFO: Preparing to flash script.
2018-07-10 22:38:43,869 - mu.modes.microbit:247(flash) DEBUG: Python script:
2018-07-10 22:38:43,869 - mu.modes.microbit:248(flash) DEBUG: b'from microbit import *\n\ndisplay.scroll("ABC")'
2018-07-10 22:38:43,876 - mu.modes.microbit:296(flash) INFO: Path to micro:bit: D:\
2018-07-10 22:38:43,876 - mu.modes.microbit:297(flash) INFO: Serial port: COM4
2018-07-10 22:38:43,876 - mu.modes.microbit:298(flash) INFO: Device serial number: 6
2018-07-10 22:38:43,877 - mu.modes.microbit:317(flash) INFO: Checking target device.
[... cut ...]
2018-07-10 22:39:02,275 - mu.modes.microbit:342(flash) INFO: Mu MicroPython: 1.0.0-beta.1
2018-07-10 22:39:02,275 - mu.modes.microbit:450(copy_main) INFO: Copying main.py onto device
2018-07-10 22:39:02,275 - mu.modes.microbit:460(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', 'f(b\'from microbit import *\\n\\ndisplay.scroll("XYZ")\')', 'fd.close()']
2018-07-10 22:39:02,635 - mu.modes.microbit:463(copy_main) INFO: (b'', b'')
2018-07-10 22:39:08,359 - mu.logic:1045(show_admin) INFO: Showing logs from C:\Users\ibosnic\AppData\Local\python\mu\Logs\mu.log

ntoll, 2018-07-10:
OK... I've just checked and the `display.off()` state isn't reset between soft-restarts. This is a bug in MicroPython, and I've reported it here: https://github.com/bbcmicrobit/micropython/issues/527

ZanderBrown, 2018-07-11:
Haven't found a pattern but we do seem to have some REPL v Flash weirdness going on