mu-editor/mu #538: Betat 17 - QT error on Linix

jerryneedell, 2018-07-10:
I am having the same problem as in a closed issue  with Beta17
It installs fine via pip3 but gives the error below
same result using the virtual env.

``` 
Traceback (most recent call last):
  File "mu_venv/bin/mu-editor", line 7, in <module>
    from mu.app import run
  File "/home/jerryneedell/mu_venv/lib/python3.5/site-packages/mu/app.py", line 29, in <module>
    from PyQt5.QtCore import QTimer, Qt
ImportError: /home/jerryneedell/mu_venv/lib/python3.5/site-packages/PyQt5/QtCore.so: undefined symbol: PySlice_AdjustIndices
```

I see the reference to a new QT release, but it is not clear id there is recommended workaround.
I am on an Ubuntu 16.04 system
Thanks

ntoll, 2018-07-10:
OK... when on Linux, you should follow the advice in the instructions (https://codewith.mu/en/howto/install_with_python) and use a virtualenv (until such time as we manage to get Mu accepted as a Debian package -- which is work that's ongoing).

This sort of error happens when different versions of Qt are found on the same computer. A virtualenv isolates your Python apps.

The simple instructions are:

* Ensure virtualenv and Python 3 are installed.
* In a new folder somewhere safe (e.g. in `~/virtualenvs`) type: `virtualenv --python=/usr/bin/python3 muvenv` (this creates a new virtualenv called `muenv` and uses Python3.
* Activate the new virtualenv like this: `source muvenv/bin/activate` (you'll see the name of the virtualenv prepended to your shell prompt, indicating the virtualenv is active)
* With the virtualenv active, type: `pip install mu-editor`.
* Once installed, and still with the virtualenv active, start mu: `mu-editor`

It should just work :tm:. ;-)

Please do let me know how you get on.

ntoll, 2018-07-10:
OK... I'm on Ubuntu 17.10 and, following my instructions above, Mu installed fine. I'm not sure what more to suggest. :-/ Please do keep me posted.

jerryneedell, 2018-07-11:
tried on Ubuntu 16.04
```
jerryneedell@Ubuntu-Macmini:~/mu_venv$ virtualenv --python=/usr/bin/python3 muvenv
Already using interpreter /usr/bin/python3
Using base prefix '/usr'
New python executable in /home/jerryneedell/mu_venv/muvenv/bin/python3
Also creating executable in /home/jerryneedell/mu_venv/muvenv/bin/python
Installing setuptools, pkg_resources, pip, wheel...done.
jerryneedell@Ubuntu-Macmini:~/mu_venv$ source muvenv/bin/activate
(muvenv) jerryneedell@Ubuntu-Macmini:~/mu_venv$ pip install mu-editor
Collecting mu-editor
  Using cached https://files.pythonhosted.org/packages/99/70/d63467c6d94cace750085e9891bc348642d6f07a12f6e3849c11d8d40f7d/mu_editor-1.0.0b17-py3-none-any.whl
Collecting pyqt5==5.11.2 (from mu-editor)
  Using cached https://files.pythonhosted.org/packages/3a/c7/4a9bec78c864051051b41b4cc76672ecc232e6dc7dbb91a5f8ff6f20ff64/PyQt5-5.11.2-5.11.1-cp35.cp36.cp37.cp38-abi3-manylinux1_x86_64.whl
Collecting qtconsole==4.3.1 (from mu-editor)
  Using cached https://files.pythonhosted.org/packages/90/ff/047e0dca2627b162866920e7aa93f04523c0ae81e5c67060eec85701992d/qtconsole-4.3.1-py2.py3-none-any.whl
Collecting pyserial==3.4 (from mu-editor)
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Collecting gpiozero>=1.4.1 (from mu-editor)
  Using cached https://files.pythonhosted.org/packages/8f/df/2f22cea78f4539032e091579d6f32d9ebe360c04534838d151d37a667f2d/gpiozero-1.4.1-py2.py3-none-any.whl
Collecting pyflakes==2.0.0 (from mu-editor)
  Using cached https://files.pythonhosted.org/packages/44/98/af7a72c9a543b1487d92813c648cb9b9adfbc96faef5455d60f4439aa99b/pyflakes-2.0.0-py2.py3-none-any.whl
Collecting pigpio>=1.40.post1 (from mu-editor)
  Using cached https://files.pythonhosted.org/packages/76/38/f8cc5307ce65e79fab59385a56338716ab721fb27e4c116fdfc4226b000e/pigpio-1.40.post1-py2.py3-none-any.whl
Collecting pycodestyle==2.4.0 (from mu-editor)
  Using cached https://files.pythonhosted.org/packages/e5/c6/ce130213489969aa58610042dff1d908c25c731c9575af6935c2dfad03aa/pycodestyle-2.4.0-py2.py3-none-any.whl
Collecting matplotlib==2.2.2 (from mu-editor)
  Using cached https://files.pythonhosted.org/packages/81/31/4e261379e0cd4e9bbacfc96b124ebac0706b44374bd1d34ef899796f741b/matplotlib-2.2.2-cp35-cp35m-manylinux1_x86_64.whl
Collecting appdirs>=1.4.3 (from mu-editor)
  Using cached https://files.pythonhosted.org/packages/56/eb/810e700ed1349edde4cbdc1b2a21e28cdf115f9faf263f6bbf8447c1abf3/appdirs-1.4.3-py2.py3-none-any.whl
Collecting qscintilla==2.10.7 (from mu-editor)
  Using cached https://files.pythonhosted.org/packages/a6/d2/c13ab35d1747d7836d9ebdaad3bec1e7a6ce7681461f6db4765d691e2474/QScintilla-2.10.7-5.11.1-cp35.cp36.cp37.cp38-abi3-manylinux1_x86_64.whl
Collecting guizero>=0.5.2 (from mu-editor)
  Using cached https://files.pythonhosted.org/packages/87/19/e8d1831fedc1328c39dacf9a9c5342ceed0315cf301aa2284cd09fc91e9e/guizero-0.5.2-py3-none-any.whl
Collecting pgzero==1.2 (from mu-editor)
  Using cached https://files.pythonhosted.org/packages/48/e5/e5f14292373cb5fc7539aa01307b184c1e3c954d68945d8c44778669dd82/pgzero-1.2-py3-none-any.whl
Collecting PyQtChart==5.11.2 (from mu-editor)
  Using cached https://files.pythonhosted.org/packages/77/b6/ed090be2ecef15fe8a2767a42ca66b884c2ef5777e081f0ac2e7b6e4eeed/PyQtChart-5.11.2-5.11.1-cp35.cp36.cp37.cp38-abi3-manylinux1_x86_64.whl
Collecting semver>=2.8.0 (from mu-editor)
  Using cached https://files.pythonhosted.org/packages/21/18/a0de8cda637ba3efee1b3617ded00601507ce15bd70a39399740e0fd415f/semver-2.8.1-py2.py3-none-any.whl
Collecting requests>=2.19.1 (from mu-editor)
  Using cached https://files.pythonhosted.org/packages/65/47/7e02164a2a3db50ed6d8a6ab1d6d60b69c4c3fdf57a284257925dfc12bda/requests-2.19.1-py2.py3-none-any.whl
Collecting Pillow>=5.2.0 (from mu-editor)
  Using cached https://files.pythonhosted.org/packages/3d/9e/b8a10bec874d2f6f2668b0d71ddf0d0a69ed9307d2ef111607adf81fdc1c/Pillow-5.2.0-cp35-cp35m-manylinux1_x86_64.whl
Collecting PyQt5_sip<4.20,>=4.19.11 (from pyqt5==5.11.2->mu-editor)
  Using cached https://files.pythonhosted.org/packages/1f/73/a097e063ea84f9c0d5641bd45fd0147d7e88ad0e92c86638fa7ada14f436/PyQt5_sip-4.19.12-cp35-cp35m-manylinux1_x86_64.whl
Collecting pygments (from qtconsole==4.3.1->mu-editor)
  Using cached https://files.pythonhosted.org/packages/02/ee/b6e02dc6529e82b75bb06823ff7d005b141037cb1416b10c6f00fc419dca/Pygments-2.2.0-py2.py3-none-any.whl
Collecting jupyter-client>=4.1 (from qtconsole==4.3.1->mu-editor)
  Using cached https://files.pythonhosted.org/packages/94/dd/fe6c4d683b09eb05342bd2816b7779663f71762b4fa9c2d5203d35d17354/jupyter_client-5.2.3-py2.py3-none-any.whl
Collecting jupyter-core (from qtconsole==4.3.1->mu-editor)
  Using cached https://files.pythonhosted.org/packages/1d/44/065d2d7bae7bebc06f1dd70d23c36da8c50c0f08b4236716743d706762a8/jupyter_core-4.4.0-py2.py3-none-any.whl
Collecting ipykernel>=4.1 (from qtconsole==4.3.1->mu-editor)
  Using cached https://files.pythonhosted.org/packages/ab/3f/cd624c835aa3336a9110d0a99e15070f343b881b7d651ab1375ef226a3ac/ipykernel-4.8.2-py3-none-any.whl
Collecting traitlets (from qtconsole==4.3.1->mu-editor)
  Using cached https://files.pythonhosted.org/packages/93/d6/abcb22de61d78e2fc3959c964628a5771e47e7cc60d53e9342e21ed6cc9a/traitlets-4.3.2-py2.py3-none-any.whl
Collecting ipython-genutils (from qtconsole==4.3.1->mu-editor)
[... cut ...]
(muvenv) jerryneedell@Ubuntu-Macmini:~/mu_venv$ mu-editor 
Traceback (most recent call last):
  File "/home/jerryneedell/mu_venv/muvenv/bin/mu-editor", line 7, in <module>
    from mu.app import run
  File "/home/jerryneedell/mu_venv/muvenv/lib/python3.5/site-packages/mu/app.py", line 29, in <module>
    from PyQt5.QtCore import QTimer, Qt
ImportError: /home/jerryneedell/mu_venv/muvenv/lib/python3.5/site-packages/PyQt5/QtCore.so: undefined symbol: PySlice_AdjustIndices
(muvenv) jerryneedell@Ubuntu-Macmini:~/mu_venv$ 
```

jerryneedell, 2018-07-11:
tried on  Ubuntu 18.04
```
jerryneedell@Ubuntu18:~/mu_venv$ virtualenv --python=/usr/bin/python3 muvenv
Already using interpreter /usr/bin/python3
Using base prefix '/usr'
New python executable in /home/jerryneedell/mu_venv/muvenv/bin/python3
Also creating executable in /home/jerryneedell/mu_venv/muvenv/bin/python
Installing setuptools, pkg_resources, pip, wheel...done.
jerryneedell@Ubuntu18:~/mu_venv$ pip install mu-editor
Collecting mu-editor
  Downloading https://files.pythonhosted.org/packages/c4/88/8ac7af11af5cb3d8825eb3a660087801af333e80b70eb8fb923bd3871f54/mu-editor-1.0.0b17.tar.gz (1.6MB)
    100% |████████████████████████████████| 1.6MB 769kB/s 
Collecting Pillow>=5.2.0 (from mu-editor)
  Downloading https://files.pythonhosted.org/packages/6e/27/709a8493071ec649a56d5a3194f648ec7cd792189e994bbd2ef5d285670d/Pillow-5.2.0-cp27-cp27mu-manylinux1_x86_64.whl (2.0MB)
    100% |████████████████████████████████| 2.0MB 536kB/s 
Collecting PyQtChart==5.11.2 (from mu-editor)
  Could not find a version that satisfies the requirement PyQtChart==5.11.2 (from mu-editor) (from versions: )
No matching distribution found for PyQtChart==5.11.2 (from mu-editor)
jerryneedell@Ubuntu18:~/mu_venv$ 
```

tjguk, 2018-07-11:
Chipping in rather late: the error message in the OP refers to a Python API that was only added in 3.6.1. Since the venv is using Python 3.5 it appears that Qt have offered a wheel fo 3.5 which uses code only available in 3.6

ntoll, 2018-07-11:
Aha... so Mu will now only work with Python 3.6+ as a result, right..?

ZanderBrown, 2018-07-11:
Sounds like a Qt bug rather than anything else

jerryneedell, 2018-07-22:
@ntoll FYI -- I was finally able to get mu running on Ubuntu 16.04 after installing python3.6 then using "sudo pip3.6 install mu-editor"
Congratulations on the 1.0.0 release!

ntoll, 2018-07-27:
Good to hear. Thank you.