mu-editor/mu #541: Style suggestion: Differentiate more the active tab

carlosperate, 2018-07-11:
I don't know if this happens to anybody else, but every time I have two tabs open I keep wrongly thinking that the active tab is the one with the darker shade.

Of course, a darker tab usually means it is not the selected tab, but I think my brain gets confused because the selected tab has the same colour as the rest of the window, and so the one that is more visible to me is the other one.

The other themes highlight the tab with a lighter shade, this problem doesn't happen to me when using those, only with the default theme:

[image]

ZanderBrown, 2018-07-11:
@ntoll we can easily change the colour but what would @carlosperate be looking for?

carlosperate, 2018-07-11:
Thanks Zander! I'm not too bother as long as its different than the background colour and lighter than an inactive tab. Would a lighter shade of grey from the background colour be distinguishable enough?
You are pretty good with these things, so I trust your judgement here.

ZanderBrown, 2018-07-11:
Thanks for your support!

Not quite sure where to go with this one, the shades are pretty close already.

Looking at the other themes and my comments I intended to use the border colour:
[image]
but that would darken not lighten

The pure white approach might go nicely with the white text area:
[image]

Or a lighter form of the focus colour:
[image]

I'm largely in favour of the 3rd

ntoll, 2018-07-11:
I'm with you @ZanderBrown on number 3.

I dunno about you, but this feels like choosing wallpaper... ;-)

carlosperate, 2018-07-11:
Number 3 sounds like a winner, thanks Zander!

closed 2018-07-11 by ntoll

carlosperate, 2018-07-12:
Thanks Zander and Nicholas! It's subtle, but it makes a big difference :)

commit 1c347c14 (2018-07-11): Fix #541
commit ed4df45d (2018-07-11): Merge pull request #546 from ZanderBrown/base/current-tab

Fix #541