mu-editor/mu #543: Mu application name in different OS

carlosperate, 2018-07-11:
Right now we have Mu named differently on different platforms:

* Windows: Named `Mu`, as it installs in a folder with that name, and the executable has that name.
* macOS: Name `mu-editor`, which doesn't follow the OS convention as it should be `Mu Editor`.
* Linux/pip-installed: Invoked with the `mu-editor` command
* Debian Packaging: Not sure about this one, since `mu` is taken, is it going to be `mu-editor` as well?
* RPM packaging: ?

Maybe the application should adopt 1 name that works in all platforms?

carlosperate, 2018-07-11:
Okay, so it will be `mu-editor` in debian, rpm and any other packaging system, right?

I'm far from a mac expert (assuming you've included it Unix-y bag), but all the applications I've got on my system have their app bundle names capitalised and spaces used when separating word. It looks like this might be a well established convention, and Mu looks a bit out of place with `mu-editor`. Will the name be updated to follow it or do you prefer to stick to the current one?

ntoll, 2018-07-11:
`mu-editor` for all Linux, yes.

As for OSX, I'll see if I can change it.

ZanderBrown, 2018-07-11:
Local RPM user: [we are mu provided by the package mu](https://apps.fedoraproject.org/packages/mu/contents/)

carlosperate, 2018-08-24:
Ok, so just as a general summary for Linux:
* Debian packages are named `mu-editor` (this PR also to change the name: https://github.com/mu-editor/mu/pull/620)
* Looks like the AUR package is already `mu-editor`: https://aur.archlinux.org/packages/mu-editor/
* However, Fedora package is `mu`.
    * @ntoll should the RPM package be renamed from `mu` to `mu-editor`? Is Kushal Das the maintainer?

ZanderBrown, 2018-08-24:
They [are](https://apps.fedoraproject.org/packages/mu/overview) and the 1.0 builds do provide [`mu-editor` now](https://koji.fedoraproject.org/koji/rpminfo?rpmID=15171517) but are still packaged as mu