mu-editor/mu #544: Debugger: Line highlighting selects more than it should

carlosperate, 2018-07-11:
When running the debugger in beta 17 the debugger line highlighter covers the first character of the line below as well.

GIF:
[image]

This is running on Windows, can this be replicated in the other platforms?

ntoll, 2018-07-11:
Which version of Windows... It works fine for me on Windows 10.

ntoll, 2018-07-11:
Might this be a 64/32 bit difference, or maybe a newline problem (i.e. `len(line)` counts `\r\n` instead of `\n`.).

tjguk, 2018-07-11:
Does the same thing happen if you do / don't save the file? (That seems to be where line-ending differences kick in)

ntoll, 2018-07-11:
Please try again, in light of f03182744690426c03c70eda49e4c588e03aae01 (which should mitigate `\r\n` based ObOE).

carlosperate, 2018-07-12:
> Which version of Windows... It works fine for me on Windows 10.

Windows 10 64-bit running Mu 64-bit.

> Does the same thing happen if you do / don't save the file? (That seems to be where line-ending differences kick in)

Debugger forces you to save the file before debugging, so it was already saved by the point the debugger started running.

> Please try again, in light of f031827 (which should mitigate \r\n based ObOE).

Just tried this in the latest master and got excited when it worked correctly there, but then I thought I should git checkout 97d0307065ba1f915f04c0f5787fcda42996c05d (b17 commit), and running from my virtualenv I cannot replicate there either, only with the installed version :/

I'll see if I can pick up the build from appveyor. 
@ntoll for releases do you ship the installers from the CI? Or rebuild in a local machine?

ntoll, 2018-07-12:
@carlosperate I ship from CI.

carlosperate, 2018-07-12:
Tried the latest CI .exe installer and looked like the issue is gone, however, I went back to my original environment (installed b17), I cannot replicate again, argggh! 😓

carlosperate, 2018-07-12:
Okay, so @tjguk was onto something when he was thinking about the file state before saving.

Managed to replicate in my virtualenv installation with a git checkout of beta17.
The problem only happened when you have just created a new file, wrote some code, and then clicked debug (and saved, as it forces you to).

Whenever the file is loaded into Mu (like when Mu is closed and reopened) the debugger works fine.

And I can finally confirm @ntoll's update fixed the isssue 🍾

closed 2018-07-12 by carlosperate