mu-editor/mu #545: Request: clear the logs in the dialog window

CleoQc, 2018-07-11:
I would like to be able to clear the logs in the window that pops up when clicking on the gear icon in the lower right corner.
This would not necessarily delete the log file itself but would limit what is shown in the dialog window (it would only show the log file since the last "clear log window" command. 

I spent a while trying to figure out what was going wrong, and that log info became quite big due to various test attempts on my part.

ZanderBrown, 2018-07-11:
KISS approach would probably be to just delete the file but then we would lose the version info header (but that should be doable)

Thoughts @ntoll ?

tmontes, 2018-07-17:
Chiming in with one possible approach:
* Instead of deleting the current log, call `<logging-handler>.doRollover()`.
* Given that the logging handler is a `logging.handlers.TimedRotatingFileHandler`, it will do its "log file rotation" thing, and the current log will be empty.
* One important detail is that the rotation may overwrite the previously rotated log file. Two possible improvements are:
  * Use a log handler derived from `l.h.TimedRotatingFileHandler` with a customised `rotation_filename(self, default_name)` method.
  * More lazily, but also effective, given it's a public API, set `.suffix` on the handler object to something like `'%Y-%m-%d_%H%M%S'` which will have the rotated log files include the date and time with per-hour/minute/second resolution (the current `.suffix` being `'%Y-%m-%d_%H'` due to the fact that the handler is created to rotate files on a daily basis).
* More, to avoid losing version/environment information from newly created (post-rotation) log files, that info can/should be explicitly logged again after the `.doRollover()`.

My 2c.

A few related ideas that may deserve issues on their own (not that I feel the need for them), if a "clear log" button ever gets into the UI:
* Would it be valuable to have a button to "copy the current log file to the clipboard"?
* And/or, for more advanced users, have a button that opens up the OS file navigator in the directory containing the log files.
* How about setting logging detail? A simplified UI could be used, instead of "debug/info/warning/error/critical", a "disabled/enabled/detailed" could match "critical/warning/debug".

ZanderBrown, 2018-07-17:
Adding some of these things to the context menu seems reasonable (little harder to hit by accident than buttons)

CleoQc, 2018-07-17:
The context menu is what I had in mind when I opened up this issue. This feature is not worth taking real estate for a button and is best hidden away until you actually need it.