mu-editor/mu #547: Feedback to the user while flashing messages and a progressbar

StefanL38, 2018-07-11:
Hi I'm using Mu 1.0.0.beta.17 with the MicroBit.
It works as far as I have tested Mu. IMHO when flashing code into the MB there should be displayed a message "start flashing" 
a progressbar that shows how flashing goes on and a message 
"code <Filename> successfully flashed"

If you miss the flash-button while clicking or if flashing was successful but the code does not work the user will get NO feedback at all. That's too less. So please add messages as decribed above.
best regards  

Stefan

ZanderBrown, 2018-07-11:
Messages are shown in the status bar ([which is a discussion in itself](http://www.panopticoncentral.net/2004/04/26/status-bar-almost-entirely-superfluous/)) because the feeling was dialogs are little too intrusive

Not sure a progress bar would help as it would be tricky for this to present any useful information

StefanL38, 2018-07-12:
I downloaded the beta.17 yesterday (11.07.2018). In this version NO message is shown in the statusbar when flashing to the MicroBit
If I save the file there appears a message.

About the progressbar:  I guess there is some handshaking between the PC and the MB while uploading.
So at least start and end of uploading should be detectable. I can determine this from the following test:

I clicked Flash and immediately clicked Files. If flashing is still active there is a delay before the file-windows appear.

So a messagebox or a small animation of a line that show something like "running lights" simple as a dashline of some pixel thick can be done.

What do you exactly mean by "I've updated master so there's guaranteed to be a confirmation message in the status bar"  which version do I have to download where again to get this "updated master"

best regards  

Stefan 
About a window beeing too intrusive: The MicroBit is for BEGINNERS. So a messagebox will HELP beginners to see what is going on.  If Mu gives NO feedback IMHO that is too hiding.

ntoll, 2018-07-12:
@StefanL38 

I hear ya...

1) All Mu can detect is when flashing starts and finishes. On both occasions a message is displayed in the status bar along the lines of "Flashing micro:bit" and "Finished flashing". This is the current behaviour.
2) I first thought like you about having a message box when flashing commenced, and this was the original behaviour. However, we got so many complaints about it, and so many "thank got it's been removed" messages on Twitter when we got rid of it, that were we to introduce it again, we'd have rioting in the streets, IYSWIM. ;-)
3) There *was* (until I fixed it last night) a route through the logic for flashing that wouldn't display any messages in the status bar. This has been fixed (and that's what I meant by "updated in master") and will be in the next official release (the release candidate arriving on Friday or Monday).

Hope this helps.