mu-editor/mu #548: Mu Python Mode Using System Python not Virtual Environment on Linux

amos1969, 2018-07-12:
I've managed to get Mu working on Mint (18.2) with Python 3.7, in a virtual environment, but when I switch to standard Python Mode and open up the REPL it shows Python 3.5.2 as the version number (the default Python 3 interpreter for the version of Mint I'm on).

If I want to do Pygame stuff directly in the Python editor, that'll mean that I need to ensure Pygame is installed on 3.5.2 outside the virtual environment.

Dave

ntoll, 2018-07-12:
Hi Dave, :-)

That's weird behaviour. Here's something to try... just have a script in PyGame or Python mode with the following content:

```
import sys
print(sys.version)
```

Then run it.

In the pane that opens, you should see the version of Python which Mu is using to run your script. I'm pretty confident this will be the version of Python used to run Mu. (When Mu starts a script it asks Python to give it a path to Python, and Python will give it a path to itself... so if you're using Python 3.7 it'll return a path to *that* version of Python.)

It's odd that the REPL is using system Python. I believe it's perhaps because the implementation is using a Jupyter notebook and Jupyter (the thing that runs the REPL) is basically being asked to start a Python thing, asks the OS for Python and the OS gives it back the path for system Python rather than the version of Python included in Mu. This is likely a bug in Jupyter, or perhaps I've failed to configure it properly to use a specific version of Python. I'll investigate further.

Thanks for letting me know!

N.

amos1969, 2018-07-12:
Hi Nicholas

I've done that, the window which opens appears to be running it with 3.7.0 but the REPL appears to be using 3.5.2:

[image]

Cheers
Dave

ntoll, 2018-07-15:
@amos1969 Hi Dave, have you managed to run the code I pasted above? Would be interested in the results. Thanks. N.

ntoll, 2018-07-27:
OK... some comments from Thomas:

```
I started writing out some convoluted options, but while looking at them I realised there's a simpler override that should work. Just before this line:
https://github.com/mu-editor/mu/blob/487c8116b5f916b06d6739ba46289547fff8ca32/mu/modes/python3.py#L74

Set self.repl_kernel_manager.kernel_cmd = [sys.executable, ...] (you'll need to track down the arguments that go in ... to launch ipykernel, but these should be basically static - you can find them in kernelspecs, or there's a function somewhere called make_ipkernel_cmd).

That activates this override in jupyter_client:
https://github.com/jupyter/jupyter_client/blob/86a347d77b4fb817c636154bb68f48342e0eee67/jupyter_client/manager.py#L156-L157

I'll leave the notes I started writing in below the line, so I don't need to rewrite them if the above turns out not to work. But feel free to ignore them otherwise! :-)

Best wishes,
Thomas

------

I think there are two options - neither of them entirely straightforward, unfortunately:

1. Create a kernelspec with the Python you want to use. There's a special case in jupyter_client that if argv[0] is just 'python', that means the same Python as the parent process. Put it at the head of the search path by setting the JUPYTER_PATH environment variable (http://jupyter.readthedocs.io/en/latest/projects/jupyter-directories.html#envvar-JUPYTER_PATH ). Note that the directory you give in JUPYTER_PATH should have a 'kernels' subdirectory, which contains the kernelspec directories, which contain kernel.json. So you'll have something like this:

JUPYTER_PATH=.../mu/jupyter_data
.../mu/jupyter_data/kernels/python3/kernel.json  <- this actually defines your kernel.

2. Alternatively, subclass QtKernelManager (which you use directly in Mu to launch the kernel) and 
```

(Yes, they're truncated).

dybber, 2021-01-26:
Isn't this being solved by the venv-work?

carlosperate, 2021-03-16:
@amos1969 hopefully this has been fixed in the latest 1.1.0-beta.2 release:
- https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.2 
- https://pypi.org/project/mu-editor/1.1.0b2/

Could you try installing this version in a virtualenv and see if it works correctly?

carlosperate, 2021-04-26:
Hi @amos1969, just a friendly ping to see if you'd be able to check if this is still an issue in the latest beta release?

amos1969, 2021-04-27:
Hi @carlosperate 

I installed Mu (1.1.0.beta.3) in a virtual environment. The version of Python that runs when I run code in Python3 mode matches the version from the virtual environment but doesn't include any packages that were installed into that virtual environment. It also doesn't pick up anything from the system Python. So it looks like it's fixed.

carlosperate, 2021-04-27:
Awesome 🎉 
Thank you for the report and all the testing!

closed 2021-04-27 by carlosperate