mu-editor/mu #549: crash: flashing on Windows using CI build for ac011b7

jaustin, 2018-07-12:
Using the build from https://s3-eu-west-2.amazonaws.com/mu-builds/windows/mu_2018-07-12_09_39_master_ac011b7_64bit.exe

Installed 'for just this user' not for whole system, using the exe installer

* What you were trying to do:

flash the micro:bit

* What steps you took to make this happen:

Windows 10
Install Mu from exe installer
Select 'just for this user' when prompted
Launch Mu from start menu
Switch to micro:bit mode
Write some code
click 'flash'

* What you expected to happen:

The micro:bit would be flashed

* What actually happened,

Mu crashed

* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

There's a micro:bit plugged in and is present as MICROBIT (E:) however I don't see a device in Device Manager. That's unusual for this machine, I use the COM ports for this often.... 

Using drag-and-drop flash does continue to work if I do it manually, so would make a viable fallback for Mu in this situation.

Log extract

(in this log I also clicked 'repl' additionally to the instructions above, got an error that the MicroPython couldn't be found)

> 
> Starting Mu 1.0.0.beta.17
> 2018-07-12 14:18:50,702 - root:113(run) INFO: uname_result(system='Windows', node='DESKTOP-38P63UE', release='10', version='10.0.17134', machine='AMD64', processor='Intel64 Family 6 Model 142 Stepping 10, GenuineIntel')
> 2018-07-12 14:18:50,702 - root:114(run) INFO: Python path: ['C:\\Users\\jmaus\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\jmaus\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\jmaus\\AppData\\Local\\Mu\\Python', 'C:\\Users\\jmaus\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\jmaus\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
> 2018-07-12 14:18:50,796 - mu.logic:543(__init__) INFO: Setting up editor.
> 2018-07-12 14:18:50,797 - mu.logic:561(__init__) INFO: Settings path: C:\Users\jmaus\AppData\Local\python\mu\settings.json
> 2018-07-12 14:18:50,797 - mu.logic:562(__init__) INFO: Session path: C:\Users\jmaus\AppData\Local\python\mu\session.json
> 2018-07-12 14:18:50,797 - mu.logic:563(__init__) INFO: Log directory: C:\Users\jmaus\AppData\Local\python\mu\Logs
> 2018-07-12 14:18:50,797 - mu.logic:564(__init__) INFO: Data directory: C:\Users\jmaus\AppData\Local\python\mu
> 2018-07-12 14:18:50,817 - mu.logic:577(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
> 2018-07-12 14:18:51,321 - mu.logic:1118(change_mode) INFO: Workspace directory: C:\Users\jmaus\mu_code
> 2018-07-12 14:18:51,323 - mu.logic:631(restore_session) INFO: Restoring session from: C:\Users\jmaus\AppData\Local\python\mu\session.json
> 2018-07-12 14:18:51,323 - mu.logic:632(restore_session) DEBUG: {}
> 2018-07-12 14:18:51,323 - mu.logic:1075(select_mode) INFO: Showing available modes: ['python', 'adafruit', 'microbit', 'debugger', 'pygamezero']
> 2018-07-12 14:18:55,329 - mu.logic:1080(select_mode) INFO: New mode selected: microbit
> 2018-07-12 14:18:55,349 - mu.logic:1118(change_mode) INFO: Workspace directory: C:\Users\jmaus\mu_code
> 2018-07-12 14:18:55,778 - mu.logic:679(restore_session) INFO: Starting with blank file.
> 2018-07-12 14:18:55,791 - mu.logic:1118(change_mode) INFO: Workspace directory: C:\Users\jmaus\mu_code
> 2018-07-12 14:18:59,744 - mu.modes.base:210(find_device) WARNING: Could not find device.
> 2018-07-12 14:18:59,744 - mu.modes.base:211(find_device) DEBUG: Available ports:
> 2018-07-12 14:18:59,745 - mu.modes.base:215(find_device) DEBUG: ['PID:0 VID:0 PORT:COM12', 'PID:0 VID:0 PORT:COM13']
> 2018-07-12 14:18:59,773 - mu.interface.main:699(show_message) DEBUG: Could not find an attached device.
> 2018-07-12 14:18:59,773 - mu.interface.main:700(show_message) DEBUG: Please make sure the device is plugged into this computer.
> 
> It must have a version of MicroPython (or CircuitPython) flashed onto it before the REPL will work.
> 
> Finally, press the device's reset button and wait a few seconds before trying again.
> 2018-07-12 14:19:02,932 - mu.modes.base:238(toggle_repl) INFO: Toggle REPL on.
> 2018-07-12 14:19:07,166 - mu.logic:1045(show_admin) INFO: Showing logs from C:\Users\jmaus\AppData\Local\python\mu\Logs\mu.log
> 2018-07-12 14:19:41,183 - mu.modes.microbit:239(flash) INFO: Preparing to flash script.
> 2018-07-12 14:19:41,183 - mu.modes.microbit:249(flash) DEBUG: Python script:
> 2018-07-12 14:19:41,183 - mu.modes.microbit:250(flash) DEBUG: b'# Write your code here :-)\r\nimport microbit\r\nmicrobit.display.scroll("hello")'
> 2018-07-12 14:19:41,188 - mu.modes.microbit:298(flash) INFO: Path to micro:bit: E:\
> 2018-07-12 14:19:41,188 - mu.modes.microbit:299(flash) INFO: Serial port: None
> 2018-07-12 14:19:41,188 - mu.modes.microbit:300(flash) INFO: Device serial number: None
> 2018-07-12 14:19:41,188 - mu.modes.microbit:318(flash) INFO: Checking target device.
> 2018-07-12 14:19:41,191 - mu.modes.microbit:354(flash) WARNING: Could not detect version of MicroPython.
> 2018-07-12 14:19:41,191 - mu.modes.microbit:371(flash) INFO: Flashing new MicroPython runtime onto device
> 2018-07-12 14:19:41,192 - root:95(excepthook) ERROR: Unrecoverable error
> Traceback (most recent call last):
>   File "C:\Users\jmaus\AppData\Local\Mu\pkgs\mu\modes\microbit.py", line 389, in flash
>     model_serial_number = int(serial_number[:4])
> TypeError: 'NoneType' object is not subscriptable

jaustin, 2018-07-12:
Looking more into the 'no COM port' issue I think this is the behaviour for micro:bit devices where
both:
 - the firmware version on the micro:bit is 0234 (which is the case for all UK school-drop micro:bits)
 - the mbed driver is not installed 

For newer micro:bits, (0241 or higher) on Windows the driver is not required for the device to enumerate.

So 'drive visible but no COM port' is a reasonable thing to have in UK schools, although these people will also miss out on the REPL.

ntoll, 2018-07-13:
OK... so, my bad for not catching "no COM port". I'll put it in a try/except and if it fails fall back to forced flashing with the script appended to the hex file. Does that sound like a plan?

ntoll, 2018-07-14:
Please test.

closed 2018-07-14 by ntoll

carlosperate, 2018-07-14:
Jonny is away for a week, so I can test it in the meantime.
I gave this a go in macOS by step debugging and plugging in/out the micro:bit at the different steps that are checking for a serial port and seems to be working fine, thanks Nicholas!

If we wanted to tripple check we could go on Windows and uninstall the serial driver from the device manager, but I probably won't be able to test that until Monday.

jaustin, 2018-07-15:
I'm just downloading CI builds before getting on the plane so should be able to test and get back to you at some point today.

jaustin, 2018-07-15:
FWIW, anyone that needs to test can use this 0234 firmware here
https://www.mbed.com/en/platform/hardware/prototyping-production/daplink/daplink-on-kl26z/

ntoll, 2018-07-15:
I have a Windows machine for testing purposes... I'll have a play now. But (big "but") I'm probably not the right person to do this since I wrote the code, so will have blind spots and expectations for usage. Far better for someone else to check this.

@carlosperate if you can test on Monday *first thing* and report back, then I can do the RC in the afternoon.

Right now I'm going down a bunch of issues to fix before tomorrow and then the check-list for making a release immediately before the release tomorrow: https://mu.readthedocs.io/en/latest/release.html

tjguk, 2018-07-15:
I'm sitting next to two W10 machines right now. One certainly doesn't have the mbed driver installed (as I've only just installed W10). I'm happy to test. IIUC, given that I don't have the serial driver. My micro:bit is using relatively old firmware.

IIUC:

* On a version of Mu prior to https://github.com/mu-editor/mu/commit/f535bcfe92d8d4692fbe1d6da9b28e16d9d7a7ed with older firmware, attempting flash the mb will raise an unhandled exception

* With a Mu built since @ntoll commit, it should notice the lack of serial port and fall back to filesystem

* With a micro:bit with newer firmware, both should work

Have I got that right?

ZanderBrown, 2018-07-15:
I believe so, be careful assuming a new W10 install doesn't have the driver because Windows Update does that automatically now.

ntoll, 2018-07-15:
When you say "fall back to file-system" it's actually just doing an "old skool" flash (i.e. copying the hex file with the Python code appended to the end onto the device).

The "new style" method of flashing is to use the faux-file system on the micro:bit and copy just the code onto the device as `main.py`. IYSWIM. However, to copy such code onto the faux-file system needs a serial connection. Hence this problem. :-)

tjguk, 2018-07-15:
Sorry -- I'm clearly hazy [*] on the implementation. Regardless of my unnecessary / incorrect description of why it fails, is the following expectation correct for testing?

* Older Mu / older firmware -- flash fails

* Newer Mu / older firmware -- flash succeeds

* Newer firmware -- flash succeeds either way

[*] Where hazy means "I understand as much as you would if you'd been half-listening to an in-depth conversation from across a crowded room!"

carlosperate, 2018-07-15:
As Zander was saying, the reason the micro:bit needs the old firmware for testing is because newer versions of DAPLink on the micro:bit should trigger Windows 10 to automatically install the serial driver.

So the main thing to be able to test this in the same circumstance as the original issue would be to first make sure you have a micro:bit with the old firmware before plugging it into the Windows 10 computer.
Then plug it into the Win10 PC and check that the micro:bit (with the old firmware) shows up as MSD, but doesn't show as a serial device.

That's the reason why I was suggesting to manually uninstall the device driver from the device manager, as that should recreate the same situation.

carlosperate, 2018-07-15:
To complete a bit the checklist:

* Older Mu / older firmware -- flash fails
    * micro:bit appears as a MSD, but doesn't show as a serial device
    * Mu crashes when trying to access the serial

* Newer Mu / older firmware -- flash succeeds
    * micro:bit appears as MSD, but doesn't show as a serial device
    * Mu flashes the micro:bit using the old method of embedding the code into the hex and copying the file to the MSD

* Newer firmware -- flash succeeds either way
    * micro:bit appears as MSD, and Windows automatically installs the serial driver, so it also appears  as a serial device
    * Mu flashing will depend on the micro:bit state (does it have the latest MicroPython already running?) and will be able to flash via serial

tjguk, 2018-07-15:
Fortunately I seem to have the right combination of things to test this. So far...

micro: bit with older firmware: no serial port detected

* Older Mu (pre-https://github.com/mu-editor/mu/commit/f535bcfe92d8d4692fbe1d6da9b28e16d9d7a7ed) / older firmware -- Mu raises an TypeError

* Newer Mu / older firmware -- Succeeds

Just about to try reflashing the firmware per https://www.mbed.com/en/platform/hardware/prototyping-production/daplink/daplink-on-kl26z/

tjguk, 2018-07-15:
I've flashed to https://mbed-media.mbed.com/cms_page_media/3291/0234_microbit_if.hex (and rebooted) but still not showing as a serial device. Windows update not picking anything up either. Have I used the right firmware?

carlosperate, 2018-07-15:
> I've flashed to https://mbed-media.mbed.com/cms_page_media/3291/0234_microbit_if.hex (and rebooted) but still not showing as a serial device. Windows update not picking anything up either. Have I used the right firmware?

It would be the other one, 0243.

> micro: bit with older firmware: no serial port detected
>
> * Older Mu (pre-f535bcf) / older firmware -- Mu raises an TypeError
> * Newer Mu / older firmware -- Succeeds

That's great, thanks Tim!

tjguk, 2018-07-15:
I've flashed with 0243, but now the flashing appears to have no effect. The flash appears to succeed and gives no error but the m:b shows the results of the previous program. The (snipped) logs:

2018-07-15 17:59:19,082 - mu.modes.microbit:297(flash) INFO: Path to micro:bit: E:\
2018-07-15 17:59:19,092 - mu.modes.base:217(find_device) INFO: Found device on port: COM3
2018-07-15 17:59:19,092 - mu.modes.base:218(find_device) INFO: Serial number: 9900000031864E450020100B000000200000000097969901
2018-07-15 17:59:19,092 - mu.modes.microbit:302(flash) INFO: Serial port: COM3
2018-07-15 17:59:19,092 - mu.modes.microbit:303(flash) INFO: Device serial number: 9900000031864E450020100B000000200000000097969901
2018-07-15 17:59:19,092 - mu.modes.microbit:328(flash) INFO: Checking target device.
2018-07-15 17:59:19,742 - mu.modes.microbit:332(flash) INFO: {'sysname': 'microbit', 'nodename': 'microbit', 'release': '1.0.0-beta.1', 'version': 'micro:bit v1.0.0-beta.1+8f503af on 2018-7-9; MicroPython v1.9.2-34-gd64154c73 on 2017-09-01', 'machine': 'micro:bit with nRF51822'}
2018-07-15 17:59:19,742 - mu.modes.microbit:351(flash) INFO: Board MicroPython: 1.0.0-beta.1
2018-07-15 17:59:19,742 - mu.modes.microbit:353(flash) INFO: Mu MicroPython: 1.0.0-beta.1
2018-07-15 17:59:19,742 - mu.modes.microbit:484(copy_main) INFO: Copying main.py onto device
2018-07-15 17:59:19,742 - mu.modes.microbit:494(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', "f(b'# Write your code here :-)\\nfrom microbit import *\\ndisplay.scroll')", 'f(b\'("abc")\')', 'fd.close()']
2018-07-15 17:59:20,262 - mu.modes.microbit:497(copy_main) INFO: (b'', b'')
2018-07-15 18:00:21,742 - mu.logic:1039(show_admin) INFO: Showing logs from C:\Users\tim\AppData\Local\python\mu\Logs\mu.log

tjguk, 2018-07-15:
This last behaviour happens for the unfixed and the fixed versions of Mu

carlosperate, 2018-07-15:
I think it's possible you are seeing the "old" behaviour in MicroPython changed in https://github.com/bbcmicrobit/micropython/pull/526 (after the MicroPython released in beta 17 of Mu).

Basically if you have used Mu to flash the micro:bit without a serial driver it will have embedded the code into the hex file and copied the file via MSD.
Then you have updated the firmware and now Mu sees the serial device and therefore copies the code as a main.py file.
In the version of MicroPython included in Mu (v1.0.0-beta.1) the embedded code takes execution precedence, and so it always executes even if there is a main.py.

If you flahs an empty interpreter (in latest Mu create a new empty script and flash that), then it should copy the MicroPython.hex file without any embedded code, and any new code flashed should be always sent as a main.py file via micro:bit filesystem.

carlosperate, 2018-07-15:
Thank you for testing Tim!

jaustin, 2018-07-15:
Thanks Tim for looking into this and testing! I can confirm (unsurprisingly) the same behaviour as Tim sees. That is, flashing is chosen to be via MSD for 0234 but if I then upgrade my interface firmware I get serial comms and Mu tries the new 'fast flashing' straight off. We expect the next MicroPython to behave properly in this respect, so no worries there.

I trued this to test different flashing logic too:
* Using 0244 firmware
* Use an old version of Mu to put an old MicroPython hex on (using MSD, so embedded script)
* Use new version of Mu, click 'flash' --> micro:bit gets updated with new script via MSD (it's slow)
* Use new version of Mu and flash a different script --> just script gets updated (it's fast)

This works perfectly.
So, that's great news. Thanks @ntoll 

(The only slightly odd thing was that the 'flash' button became greyed out during this time and I had to restart the REPL to get it back. That's a different issue though. I'll try to reproduce and file an issue)

commit f535bcfe (2018-07-14): Fixes #549. If no serial connection to the micro:bit is available, revert to the old style flashing of the hex file with code appended to the end.