mu-editor/mu #550: Micro:bit misidentified as 'unsupported micro:bit' on Windows

jaustin, 2018-07-12:
If you are reporting a bug, we would like to know:

Using the build from https://s3-eu-west-2.amazonaws.com/mu-builds/windows/mu_2018-07-12_09_39_master_ac011b7_64bit.exe
(CI build for ac011b7 )

Installed 'for just this user' not for whole system, using the exe installer
* What you were trying to do,

Flash a micro:bit

* What steps you took to make this happen,

Be on Windows 10
Insert a micro:bit with firmware version >0241
(don't have the mbed dirver installed)
Be in the 'micro:bit' mode
Click flash

* What you expected to happen,

micro:bit was flashed

* What actually happened,

A dialogue saying 'this micro:bit is not supported'
[image]

This micro:bit is supported!

* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

The log below is not lying - pyserial is returning '8' as the serial number
```
>>> import serial
>>> from serial.tools.list_ports import comports as list_serial_ports
>>> a =  list_serial_ports()
>>> a[2].usb_info()
'USB VID:PID=0D28:0204 SER=8 LOCATION=1-1.1.2:x.1'
```

I suspect this is related to https://github.com/pyserial/pyserial/issues/283

Here's the details.txt from my board

> # DAPLink Firmware - see https://mbed.com/daplink
> Unique ID: 9900000031864e450008100f0000003a0000000097969901
> HIC ID: 97969901
> Auto Reset: 1
> Automation allowed: 0
> Overflow detection: 0
> Daplink Mode: Interface
> Interface Version: 0244
> Git SHA: c72963241000edb2e913ea806e62e98c2cd11552
> Local Mods: 0
> USB Interfaces: MSD, CDC, HID
> Interface CRC: 0xdf00d62f
> Remount count: 0

Here's the log

> Starting Mu 1.0.0.beta.17
> 2018-07-12 14:34:16,745 - root:113(run) INFO: uname_result(system='Windows', node='DESKTOP-38P63UE', release='10', version='10.0.17134', machine='AMD64', processor='Intel64 Family 6 Model 142 Stepping 10, GenuineIntel')
> 2018-07-12 14:34:16,745 - root:114(run) INFO: Python path: ['C:\\Users\\jmaus\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\jmaus\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\jmaus\\AppData\\Local\\Mu\\Python', 'C:\\Users\\jmaus\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\jmaus\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
> 2018-07-12 14:34:16,843 - mu.logic:543(__init__) INFO: Setting up editor.
> 2018-07-12 14:34:16,843 - mu.logic:561(__init__) INFO: Settings path: C:\Users\jmaus\AppData\Local\python\mu\settings.json
> 2018-07-12 14:34:16,843 - mu.logic:562(__init__) INFO: Session path: C:\Users\jmaus\AppData\Local\python\mu\session.json
> 2018-07-12 14:34:16,843 - mu.logic:563(__init__) INFO: Log directory: C:\Users\jmaus\AppData\Local\python\mu\Logs
> 2018-07-12 14:34:16,843 - mu.logic:564(__init__) INFO: Data directory: C:\Users\jmaus\AppData\Local\python\mu
> 2018-07-12 14:34:16,864 - mu.logic:577(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
> 2018-07-12 14:34:17,421 - mu.logic:1118(change_mode) INFO: Workspace directory: C:\Users\jmaus\mu_code
> 2018-07-12 14:34:17,422 - mu.logic:631(restore_session) INFO: Restoring session from: C:\Users\jmaus\AppData\Local\python\mu\session.json
> 2018-07-12 14:34:17,422 - mu.logic:632(restore_session) DEBUG: {}
> 2018-07-12 14:34:17,422 - mu.logic:1075(select_mode) INFO: Showing available modes: ['python', 'adafruit', 'microbit', 'debugger', 'pygamezero']
> 2018-07-12 14:34:20,998 - mu.logic:1080(select_mode) INFO: New mode selected: microbit
> 2018-07-12 14:34:21,017 - mu.logic:1118(change_mode) INFO: Workspace directory: C:\Users\jmaus\mu_code
> 2018-07-12 14:34:21,450 - mu.logic:679(restore_session) INFO: Starting with blank file.
> 2018-07-12 14:34:21,473 - mu.logic:1118(change_mode) INFO: Workspace directory: C:\Users\jmaus\mu_code
> 2018-07-12 14:34:41,070 - mu.modes.microbit:239(flash) INFO: Preparing to flash script.
> 2018-07-12 14:34:41,070 - mu.modes.microbit:249(flash) DEBUG: Python script:
> 2018-07-12 14:34:41,070 - mu.modes.microbit:250(flash) DEBUG: b'# Write your code here :-)\r\nimport microbit\r\nmicrobit.display.scroll("hi ntoll")'
> 2018-07-12 14:34:41,075 - mu.modes.microbit:298(flash) INFO: Path to micro:bit: None
> 2018-07-12 14:34:41,075 - mu.modes.microbit:299(flash) INFO: Serial port: None
> 2018-07-12 14:34:41,075 - mu.modes.microbit:300(flash) INFO: Device serial number: None
> 2018-07-12 14:34:47,957 - mu.interface.main:255(get_microbit_path) DEBUG: Getting micro:bit path: 
> 2018-07-12 14:34:47,957 - mu.modes.microbit:308(flash) DEBUG: User defined path to micro:bit: 
> 2018-07-12 14:34:47,966 - mu.interface.main:699(show_message) DEBUG: Could not find an attached BBC micro:bit.
> 2018-07-12 14:34:47,966 - mu.interface.main:700(show_message) DEBUG: Please ensure you leave enough time for the BBC micro:bit to be attached and configured correctly by your computer. This may take several seconds. Alternatively, try removing and re-attaching the device or saving your work and restarting Mu if the device remains unfound.
> 2018-07-12 14:35:16,053 - mu.modes.microbit:239(flash) INFO: Preparing to flash script.
> 2018-07-12 14:35:16,053 - mu.modes.microbit:249(flash) DEBUG: Python script:
> 2018-07-12 14:35:16,053 - mu.modes.microbit:250(flash) DEBUG: b'# Write your code here :-)\r\nimport microbit\r\nmicrobit.display.scroll("hi ntoll")'
> 2018-07-12 14:35:16,063 - mu.modes.microbit:298(flash) INFO: Path to micro:bit: F:\
> 2018-07-12 14:35:16,063 - mu.modes.microbit:299(flash) INFO: Serial port: COM5
> 2018-07-12 14:35:16,063 - mu.modes.microbit:300(flash) INFO: Device serial number: 8
> 2018-07-12 14:35:16,063 - mu.modes.microbit:318(flash) INFO: Checking target device.
> 2018-07-12 14:35:17,207 - mu.modes.microbit:354(flash) WARNING: Could not detect version of MicroPython.
> 2018-07-12 14:35:17,212 - mu.modes.microbit:371(flash) INFO: Flashing new MicroPython runtime onto device
> 2018-07-12 14:35:17,215 - mu.interface.main:699(show_message) DEBUG: Unsupported BBC micro:bit.
> 2018-07-12 14:35:17,215 - mu.interface.main:700(show_message) DEBUG: Your device is newer than this version of Mu. Please update Mu to the latest version to support this device.
> 
> https://codewith.mu/
> 2018-07-12 14:36:35,101 - mu.logic:1045(show_admin) INFO: Showing logs from C:\Users\jmaus\AppData\Local\python\mu\Logs\mu.log

carlosperate, 2018-07-12:
Can confirm the same Pyserial behaviour on Windows:

```Python
>>> from serial.tools.list_ports import comports
>>> mb_port = comports()[0]
>>> mb_port.hwid
'USB VID:PID=0D28:0204 SER=8 LOCATION=1-7.1.1:x.1'
>>> mb_port.usb_info()
'USB VID:PID=0D28:0204 SER=8 LOCATION=1-7.1.1:x.1'
>>> mb_port.serial_number
'8'
```

ntoll, 2018-07-13:
OK... so this is a spanner in the works. I'll try to work something out.

ntoll, 2018-07-14:
By the looks of the issue in PySerial, it's not fixed and I can't see any progress on this. I'll look into `QSerialPortInfo.availablePorts()` and use that instead.

ntoll, 2018-07-15:
@carlosperate @jaustin it works for me (using PySerial) on Windows 10. :-/

ntoll, 2018-07-15:
This is with an old (pre-release) testing board:

```
>>> from serial.tools.list_ports import comports
>>> x = comports()
>>> x[0].serial_number
'990002343.....'
```

Ummm... before proceeding I'd like to understand what's different between my set-up of Windows 10 and yours!

ntoll, 2018-07-15:
Sorry... my mistake... I need a *newer* version of the firmware... I misread... please ignore (trying to recreate again now)

ntoll, 2018-07-15:
Huh..???? It works for me with a device containing the following firmware details:

```
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 9900000031864e45003c10070000006e0000000097969901
HIC ID: 97969901
Auto Reset: 1
Automation allowed: 0
Overflow detection: 0
Daplink Mode: Interface
Interface Version: 0243
Git SHA: b403a07e3696cee1e116d44cbdd64446e056ce38
Local Mods: 0
USB Interfaces: MSD, CDC, HID
Interface CRC: 0x14256f44
Remount count: 0
```

jaustin, 2018-07-15:
Do you know if you've ever installed the mbed serial driver on that
machine? A good way to tell is to see if you get a REPL on an 0234  board
(IE original firmware)

J

ntoll, 2018-07-15:
@carlosperate can you provide me with the *exact* instructions to recreate this. Trying and failing to see the same thing as you. :-(

ntoll, 2018-07-15:
@jaustin ahh... so I need to uninstall the driver..? (runs off excitedly to check)

Thanks Jonny, that may have been the vital bit of info I was missing.

ntoll, 2018-07-15:
OK... driver uninstalled... Now Windows simply doesn't see the micro:bit.

E.g.

```
>>> from serial.tools.list_ports import comports
>>> x = comports()
>>> x
[]
```

This isn't the behaviour you describe.

What am I doing wrong...?

ntoll, 2018-07-15:
OK... fixed it... sorry for the noisy comments on this issue. It's taken some time to work out what the context of the problem is. I now see the same behaviour a you report (so I can at lest test potential solutions).

ntoll, 2018-07-15:
Hurrah... I can confirm that Qt's serial implementation does NOT have this problem. Switching to that for grabbing the serial number. Fix coming very soon..!

ntoll, 2018-07-15:
Please check the commit with the fix. Thanks.

closed 2018-07-15 by ntoll

jaustin, 2018-07-15:
Using the CI build mu_2018-07-15_15_57_master_8c5554b_64bit.exe

* Confirm that using pyserial I still get told the serial number is '8'
* Flash my micro:bit using Mu: it works!

So I confirm this bug is squashed for me too. Hooray for ntoll and Qt's serial implementation!

carlosperate, 2018-07-16:
Sorry, I thought I replied to this yesterday as well.
I can confirm the same, on my environment PySerial still reports serial number as 8, but the latest Mu flashes fine and reports in its logs the correct serial number. 

Out of curiosity @ntoll, how did you change your environment to get PySerial from reporting the right serial number to the wrong one? Did you have the Mbed driver installed via manual installer?

ntoll, 2018-07-16:
@carlosperate Yeah... it was the mbed driver... I'd missed that vital bit of information when trying to test. ;-) Very much a case of my working more haste less speed (IYSWIM).

commit 73ab6dea (2018-07-15): Fixes #550. Use Qt's serial classes to get the device's serial number.