mu-editor/mu #551: Unsuccesful flashing Win10 using 1.0.0.beta.17/mu-editor_beta17_win64.exe

microbit-mark, 2018-07-12:
from micro:bit-support:6553

"we have installed the latest version of Mu selected microbit mode but when we go to flash the microbit it tells us there was a problem and the log file isn't helpful (have attached the log) if we try the same code in the older version of MU it works fine but causes our computers to crash intermittently when running any code. is there a fix for the new version so it will flash properly, we get the flashing light. "

* What you were trying to do
Flash a hex file to the micro:bit in micro:bit mode

* What steps you took to make this happen
- Download Mu from  https://github.com/mu-editor/mu/releases/download/v1.0.0.beta.17/mu-editor_beta17_win64.exe
-  Open in micro:bit mode
- Attempt to flash code

[image]

* What you expected to happen
Flash completes successfully and program runs

* What actually happened
Flash appears to work (LED flashes), but error is reported  "There was a problem flashing the micro:bit"

* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 
https://github.com/mu-editor/mu/releases/download/v1.0.0.beta.17/mu-editor_beta17_win64.exe
Win10 1803
Installed for /allusers of the machine
Device shows up in Device Manager and as drive E: on the machine

* Log
2018-07-12 11:23:18,936 - root:111(run) INFO: 

-----------------

Starting Mu 1.0.0.beta.17
2018-07-12 11:23:18,937 - root:112(run) INFO: uname_result(system='Windows', node='Dan', release='10', version='10.0.17134', machine='AMD64', processor='Intel64 Family 6 Model 63 Stepping 2, GenuineIntel')
2018-07-12 11:23:18,938 - root:113(run) INFO: Python path: ['C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\Python\\python36.zip', 'C:\\Program Files\\Mu\\Python', 'C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']
2018-07-12 11:23:19,200 - mu.logic:543(__init__) INFO: Setting up editor.
2018-07-12 11:23:19,201 - mu.logic:561(__init__) INFO: Settings path: C:\Users\dan.SPS\AppData\Local\python\mu\settings.json
2018-07-12 11:23:19,201 - mu.logic:562(__init__) INFO: Session path: C:\Users\dan.SPS\AppData\Local\python\mu\session.json
2018-07-12 11:23:19,201 - mu.logic:563(__init__) INFO: Log directory: C:\Users\dan.SPS\AppData\Local\python\mu\Logs
2018-07-12 11:23:19,201 - mu.logic:564(__init__) INFO: Data directory: C:\Users\dan.SPS\AppData\Local\python\mu
2018-07-12 11:23:19,225 - mu.logic:577(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2018-07-12 11:23:19,812 - mu.logic:1119(change_mode) INFO: Workspace directory: C:\Users\dan.SPS\mu_code
2018-07-12 11:23:19,816 - mu.logic:631(restore_session) INFO: Restoring session from: C:\Users\dan.SPS\AppData\Local\python\mu\session.json
2018-07-12 11:23:19,816 - mu.logic:632(restore_session) DEBUG: {'theme': 'day', 'mode': 'microbit', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': ''}
2018-07-12 11:23:19,816 - mu.logic:653(restore_session) INFO: Loaded files.
2018-07-12 11:23:19,817 - mu.logic:657(restore_session) INFO: User defined environment variables: []
2018-07-12 11:23:19,817 - mu.logic:661(restore_session) INFO: Minify scripts on micro:bit? False
2018-07-12 11:23:20,820 - mu.logic:679(restore_session) INFO: Starting with blank file.
2018-07-12 11:23:20,837 - mu.logic:1119(change_mode) INFO: Workspace directory: C:\Users\dan.SPS\mu_code
2018-07-12 11:24:26,338 - mu.modes.microbit:237(flash) INFO: Preparing to flash script.
2018-07-12 11:24:26,339 - mu.modes.microbit:247(flash) DEBUG: Python script:
2018-07-12 11:24:26,339 - mu.modes.microbit:248(flash) DEBUG: b'from microbit import *\r\ndisplay.show("D")\r\n'
2018-07-12 11:24:26,426 - mu.modes.microbit:296(flash) INFO: Path to micro:bit: M:\
2018-07-12 11:24:26,426 - mu.modes.microbit:297(flash) INFO: Serial port: None
2018-07-12 11:24:26,426 - mu.modes.microbit:298(flash) INFO: Device serial number: None
2018-07-12 11:24:26,427 - mu.modes.microbit:317(flash) INFO: Checking target device.
2018-07-12 11:24:26,509 - mu.modes.microbit:353(flash) WARNING: Could not detect version of MicroPython.
2018-07-12 11:24:26,509 - mu.modes.microbit:370(flash) INFO: Flashing new MicroPython runtime onto device
2018-07-12 11:24:35,710 - mu.modes.microbit:450(copy_main) INFO: Copying main.py onto device
2018-07-12 11:24:35,710 - mu.modes.microbit:460(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', 'f(b\'from microbit import *\\r\\ndisplay.show("D")\\r\\n\')', 'fd.close()']
2018-07-12 11:24:35,792 - mu.modes.microbit:475(flash_failed) ERROR: Could not find micro:bit.
2018-07-12 11:24:35,852 - mu.interface.main:700(show_message) DEBUG: There was a problem flashing the micro:bit.
2018-07-12 11:24:35,853 - mu.interface.main:701(show_message) DEBUG: Please do not disconnect the device until flashing has completed. Please check the logs for more information.
2018-07-12 11:36:12,580 - mu.interface.main:243(get_save_path) DEBUG: Getting save path: F:/Desktop/mutest
2018-07-12 11:36:12,580 - mu.logic:882(check_for_shadow_module) INFO: Checking path "F:/Desktop/mutest" for shadow module.
2018-07-12 11:36:12,580 - mu.logic:848(save_tab_to_file) INFO: Saving script to: F:/Desktop/mutest.py
2018-07-12 11:36:12,580 - mu.logic:849(save_tab_to_file) DEBUG: from microbit import *

display.show("D")

[... 164 lines cut ...]
2018-07-12 12:21:03,670 - mu.modes.microbit:317(flash) INFO: Checking target device.
2018-07-12 12:21:03,752 - mu.modes.microbit:353(flash) WARNING: Could not detect version of MicroPython.
2018-07-12 12:21:03,752 - mu.modes.microbit:370(flash) INFO: Flashing new MicroPython runtime onto device
2018-07-12 12:21:12,946 - mu.modes.microbit:450(copy_main) INFO: Copying main.py onto device
2018-07-12 12:21:12,946 - mu.modes.microbit:460(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', 'f(b\'from microbit import *\\ndisplay.show("D")\\r\\n\\r\\n\\r\\n\')', 'fd.close()']
2018-07-12 12:21:13,029 - mu.modes.microbit:475(flash_failed) ERROR: Could not find micro:bit.
2018-07-12 12:21:13,031 - mu.interface.main:700(show_message) DEBUG: There was a problem flashing the micro:bit.
2018-07-12 12:21:13,032 - mu.interface.main:701(show_message) DEBUG: Please do not disconnect the device until flashing has completed. Please check the logs for more information.
2018-07-12 12:21:24,581 - mu.logic:1076(select_mode) INFO: Showing available modes: ['python', 'adafruit', 'microbit', 'debugger', 'pygamezero']
2018-07-12 12:21:47,795 - mu.logic:1045(show_admin) INFO: Showing logs from C:\Users\dan.SPS\AppData\Local\python\mu\Logs\mu.log
2018-07-12 12:25:50,570 - mu.modes.microbit:237(flash) INFO: Preparing to flash script.
2018-07-12 12:25:50,570 - mu.modes.microbit:247(flash) DEBUG: Python script:
2018-07-12 12:25:50,570 - mu.modes.microbit:248(flash) DEBUG: b'from microbit import *\ndisplay.show("D")\r\n\r\n\r\n'
2018-07-12 12:25:50,658 - mu.modes.microbit:296(flash) INFO: Path to micro:bit: M:\
2018-07-12 12:25:50,658 - mu.modes.microbit:297(flash) INFO: Serial port: None
2018-07-12 12:25:50,658 - mu.modes.microbit:298(flash) INFO: Device serial number: None
2018-07-12 12:25:50,659 - mu.modes.microbit:317(flash) INFO: Checking target device.
2018-07-12 12:25:50,740 - mu.modes.microbit:353(flash) WARNING: Could not detect version of MicroPython.
2018-07-12 12:25:50,740 - mu.modes.microbit:370(flash) INFO: Flashing new MicroPython runtime onto device
2018-07-12 12:25:59,934 - mu.modes.microbit:450(copy_main) INFO: Copying main.py onto device
2018-07-12 12:25:59,935 - mu.modes.microbit:460(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', 'f(b\'from microbit import *\\ndisplay.show("D")\\r\\n\\r\\n\\r\\n\')', 'fd.close()']
2018-07-12 12:26:00,016 - mu.modes.microbit:475(flash_failed) ERROR: Could not find micro:bit.
2018-07-12 12:26:00,019 - mu.interface.main:700(show_message) DEBUG: There was a problem flashing the micro:bit.
2018-07-12 12:26:00,019 - mu.interface.main:701(show_message) DEBUG: Please do not disconnect the device until flashing has completed. Please check the logs for more information.
2018-07-12 12:26:29,183 - mu.logic:1045(show_admin) INFO: Showing logs from C:\Users\dan.SPS\AppData\Local\python\mu\Logs\mu.log

carlosperate, 2018-07-12:
Looks like maybe there isn't enough time from the point the deviced is flashed using the OS FS, to the point where Mu tries to access the micro:bit FS via REPL?

microbit-mark, 2018-07-13:
User reports it fixed with updated firmware

closed 2018-07-13 by microbit-mark

ntoll, 2018-07-13:
OK... so can we always say, when it comes to the new Mu flashing functionality, that the user should make sure they have the latest firmware..? ;-)

carlosperate, 2018-07-13:
Is it a correct statement that when micro:bit is flashed via "copying MicroPython hex file" it will always try to then copy main.py via serial? Instead of attaching the code to the hex?

What does Mu currently do between "start copying hex" and "try to transfer main.py via serial"? How does it determine that the copy is done and serial is available? Is it a predefined time delay before I tries to connect?
If that's the case it's possible different computers and operating systems could take longer to go through that process and then fail to connect?

ntoll, 2018-07-13:
Yes... I'm currently investigating this on different set-ups. Fix/patch coming later this afternoon.

tjguk, 2018-07-13:
@ntoll when I was looking at this area of code the other day, there seemed to be a certain confusion over the user-defined microbit path (eg around https://github.com/mu-editor/mu/blob/master/mu/modes/microbit.py#L374 where we're acting on the basis of a user-defined path but then not using it as such). If you're refactoring that area in general, I'll leave it up to you. If you'd rather I raised a separate issue I'm happy to do that. In either case I'll wait until you've finished work on that area of code

ntoll, 2018-07-13:
@tjguk the first thing I'm going to do post 1.0 is refactor this method... it's too long, confusing and contains all sorts of moving parts. ;-)

jaustin, 2018-07-15:
We see extremely low incidence of people updating their micro:bit firmware without being asked to, and for a class teacher to do 30 devices is a good chunk of time they might well want to spend on something else. Therefore if we can avoid requiring firmware update for the basic functionality by using mass storage as fallback, that's really helpful. I think it's fine to say 'you get this added ability if you have the latest firmware' (like, for example, no need to install the driver if you're on Windows 8.1+, which is a massive win!) - but we'll need to help people find the firmware (it's on https://www.mbed.com/en/platform/hardware/prototyping-production/daplink/daplink-on-kl26z/ )