mu-editor/mu #552: Dark Themes no more readable on macOS Mojave beta

sstaub, 2018-07-12:
b17, MacBook Pro15'' 10.14 Beta (18A326h) Mojave
The dark themes are no more readable on macOS Mojave Beta.
This happens on the both modes of Mojave.
[image]
[image]

ZanderBrown, 2018-07-12:
Ha! Saw this earlier when someone tried it out, didn't really get a chance to investigate.

Especially when Mojave is in beta it's hard to say if this is a bug with macOS, Qt or ourselves.

@carlosperate how do they look on your mac?

sstaub, 2018-07-12:
It seems it is an macOS Mojave problem, the new dark mode is more alpha than beta.

ZanderBrown, 2018-07-13:
> My macbook crashes enough as it is (probably a hardware issue though), so I wasn't too eager to install a beta OS. HS has been fine.

(Sorry I did mean non-Beta)

sstaub, 2018-07-13:
The mu normal mode is ok, only the dark mu themes are the problem, independently of the mode of mojave.

ZanderBrown, 2018-07-13:
So just to be clear: This happens with the Mojave light theme?

sstaub, 2018-07-13:
Yes, with light and dark theme.

stestagg, 2018-07-17:
I'm seeing this too, in Mojave with both OSX light and dark theme.

For some reason, it's caused by loading/using the font from a QFontDatabase.

I've got Source Code pro installed in my system, and running mu 1.0rc unmodified shows:
[image]
[image]

Note, all text is black, just the background color changes.

But commenting out line 67 of themes.py:
```cls._DATABASE.addApplicationFontFromData(font_data)```
causes mu to use the system installed (identical) version of Source code pro, and the text colours immediately return:

[image]
[image]

I've dug into various things but haven't been able to get to the bottom of why using QFontDatabase loaded fonts causes this

ntoll, 2018-07-18:
OK... @stestagg, what do you suggest as a work around..? Some sort of bodge to check the platform and version..?

ntoll, 2018-07-18:
@stestagg is this something you can do/test/check? I'm currently up to my eyes with Windows signing "fun" and don't have access to a Mac.

stestagg, 2018-07-18:
Yeah. I’m out tonight but can take a look tomorrow.

Steve

ntoll, 2018-07-18:
OK... no pressure, but just so I can manage my day tomorrow, when might this be done? My aim is to have the final builds done tomorrow so I can test all the things ready for an "event free" Friday release. ;-)

stestagg, 2018-07-18:
I’ll see what I can do tonight, but I have to do a long day at work
tomorrow so that probably won’t help you much :)

I don’t think it can be that hard to do, and I e got access to OS X 11 and
12 so tonight should be reasonable

ntoll, 2018-07-19:
@stestagg you're a star! Thank you. I check / merge now.

ntoll, 2018-07-19:
Fixed by #567

closed 2018-07-19 by ntoll